zephyrproject-rtos/zephyr #80938: Safety: Fault Detection

simhein, 2024-11-05:
This issue ongoing discussions around the topic of fault detection and  fault detection layer to Zephyr development.

Status (05 November 2024):

In the Safety WG a proposal was started to implement/integrate a fault detection layer, where hardware/software fault detection can be connect to zephyr.
A working document was set up under following link: https://docs.google.com/document/d/17JCHUi_Xan6r0MhaG_8jP3IVqkH4UYoMq-PbXGmR1YQ/edit?tab=t.0

The first goal is to bring the fault detection proposal to an RFC state to present it to the project.
The Long term goal is to implement such fault detection layer into zephyr.

simhein, 2024-11-11:
@NelsonFerragut For your information if you didn't see it already.

simhein, 2024-12-13:
Status 03 December 2024: 

No objections from the working group regarding the RFC draft in the document.
Decided next steps are:

- Nelson will gather requirements regarding the concept of the fault detection to make it more clear
- Simon will help providing a problem description to the RFC
- The RFC will be published as an RFC in an issue and will come out of the draft document to make it more public to the community

closed 2025-02-26 by github-actions[bot]

NelsonFerragut, 2025-02-27:
Safety WG decided not to pursue this RFC in favor of using the [Task Watchdog](https://docs.zephyrproject.org/latest/services/task_wdt/index.html) feature of Zephyr.