zephyrproject-rtos/zephyr #80949: llext: arc: fix compiler-linker mismatch on qemu_arc_hs5x

laurenmurphyx64, 2024-11-05:
(Refer to llext ARC support introducing [PR](https://github.com/zephyrproject-rtos/zephyr/pull/80078))

**Describe the bug**
Two tests fail due to an apparent (???) mismatch between compiler and linker

**To Reproduce**
```
west build -p -b qemu_arc/qemu_arc_hs5x tests/subsys/llext/simple -T tests/subsys/llext/simple/llext.simple.writable_relocatable -t run
west build -p -b qemu_arc/qemu_arc_hs5x tests/subsys/llext/simple -T llext.simple.writable_relocatable_slid_linking -t run
```

**Expected behavior**
Passing tests.

**Impact**
Annoying.

**Logs and console output**

```
$ west build -p -b qemu_arc/qemu_arc_hs5x tests/subsys/llext/simple -T llext.simple.writable_relocatable_slid_linking -t run
-- west build: making build dir /home/laurenmu/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/laurenmu/zephyrproject/zephyr/tests/subsys/llext/simple
-- CMake version: 3.28.3
-- Found Python3: /home/laurenmu/zephyrproject/.venv/bin/python3 (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/laurenmu/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/laurenmu/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: qemu_arc, qualifiers: qemu_arc_hs5x
-- Found host-tools: zephyr 0.17.0 (/home/laurenmu/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/laurenmu/zephyr-sdk-0.17.0)
-- Found Dtc: /home/laurenmu/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/laurenmu/zephyrproject/zephyr/boards/qemu/arc/qemu_arc_qemu_arc_hs5x.dts
-- Generated zephyr.dts: /home/laurenmu/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/laurenmu/zephyrproject/zephyr/build/zephyr/edt.pickle
-- Generated zephyr.dts: /home/laurenmu/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/laurenmu/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/laurenmu/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/laurenmu/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/laurenmu/zephyrproject/zephyr/boards/qemu/arc/qemu_arc_qemu_arc_hs5x_defconfig'
Merged configuration '/home/laurenmu/zephyrproject/zephyr/tests/subsys/llext/simple/prj.conf'
Merged configuration '/home/laurenmu/zephyrproject/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/laurenmu/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/laurenmu/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/arc64-zephyr-elf/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/arc64-zephyr-elf-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done (4.6s)
-- Generating done (0.1s)
-- Build files have been written to: /home/laurenmu/zephyrproject/zephyr/build
-- west build: running target run
[1/180] Preparing syscall dependency handling

[2/180] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (/home/laurenmu/zephyrproject/zephyr), build: v3.7.0-4121-ga3d061346664
[16/180] Linking C executable threads_kernel_objects_ext_llext_lib.obj
FAILED: threads_kernel_objects_ext_llext_lib.obj 
: && ccache /home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/arc64-zephyr-elf-gcc  -gdwarf-4 -r CMakeFiles/threads_kernel_objects_ext_llext_lib.dir/src/threads_kernel_objects_ext.c.obj -o threads_kernel_objects_ext_llext_lib.obj   && :
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: CMakeFiles/threads_kernel_objects_ext_llext_lib.dir/src/threads_kernel_objects_ext.c.obj: ABI is incompatible with the selected emulation:
  target emulation 'elf32-littlearc64' does not match 'elf64-littlearc64'
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: failed to merge target specific data of file CMakeFiles/threads_kernel_objects_ext_llext_lib.dir/src/threads_kernel_objects_ext.c.obj
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: attempt to do relocatable link with elf32-littlearc64 input and elf64-littlearc64 output
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: CMakeFiles/threads_kernel_objects_ext_llext_lib.dir/src/threads_kernel_objects_ext.c.obj: file class ELFCLASS32 incompatible with ELFCLASS64
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
[... 46 lines cut ...]
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: attempt to do relocatable link with elf32-littlearc64 input and elf64-littlearc64 output
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: CMakeFiles/syscalls_ext_llext_lib.dir/src/syscalls_ext.c.obj: file class ELFCLASS32 incompatible with ELFCLASS64
/home/laurenmu/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/../lib/gcc/arc64-zephyr-elf/12.2.0/../../../../arc64-zephyr-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/laurenmu/zephyrproject/zephyr/build --target run
```

```
$ ~/zephyr-sdk-0.17.0/arc64-zephyr-elf/bin/arc64-zephyr-elf-objdump -x build/CMakeFiles/syscalls_ext_llext_lib.dir/src/syscalls_ext
.c.obj
build/CMakeFiles/syscalls_ext_llext_lib.dir/src/syscalls_ext.c.obj:     file format elf32-littlearc64
build/CMakeFiles/syscalls_ext_llext_lib.dir/src/syscalls_ext.c.obj
architecture: arc64:32, flags 0x00000011:
HAS_RELOC, HAS_SYMS
start address 0x00000000
private flags = 0x400:-mcpu=unknown (ABI:v4)
...
```

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain: Zephyr SDK 17.0
 - [PR, soon to be merged hopefully, this bug is preemptive](https://github.com/zephyrproject-rtos/zephyr/pull/80078)

pillo79, 2025-01-23:
@laurenmurphyx64 I did some experiments on ARC (`hs5x`). 
It appears to be using the 64-bit toolchain to output a 32 bit binary (🤷‍) by specifying `-mcpu=hs5x` on both compiler and linker commands. LLEXT inherits the flag on compiles, but not on linker calls: it gets added via the magic `toolchain_ld_link_elf` CMake macro, which is used for final Zephyr linking only. Without that flag the compiler defaults to 64 bit... and rightfully complains.

I think the ARC toolchain should be improved so that the `-mcpu` flag is added to the common linker options, but I'm not an expert there so I don't know if and how to do this properly.

laurenmurphyx64, 2025-01-24:
> [@laurenmurphyx64](https://github.com/laurenmurphyx64) I did some experiments on ARC (`hs5x`). It appears to be using the 64-bit toolchain to output a 32 bit binary (🤷‍) by specifying `-mcpu=hs5x` [...]

@stephanosio Thoughts on this?

PR #84631 (2025-02-05): llext: enforce bitness match
Enforces bitness match of ARC compilers and linkers invoked building LLEXT apps by inheriting -mcpu flag from Zephyr. The ARC toolchain becomes confused without the -mcpu flag. On qemu_hs5x we see GCC select elf32-littlearc64 for linking an object previously compiled as elf64-littlearc64.

Fixes #80949
commit f0eb7ebc (2025-02-05): llext: arc: enforce bitness match

Enforces bitness match of ARC compilers and linkers invoked
building LLEXT apps by inheriting -mcpu flag from Zephyr.
The ARC toolchain becomes confused without the -mcpu flag.
On qemu_hs5x we see GCC select elf32-littlearc64 for
linking an object previously compiled as elf64-littlearc64.

Fixes #80949
closed 2025-02-05 by kartben