zephyrproject-rtos/zephyr #80978: build_system: Allow to build without adding sources to the `app` target

clamattia, 2024-11-06:
# Problem
When building an application, which extends zephyr via `zephyr/module.yml` and `zephyr_library()` to add modules and the `SYS_INIT` makro to start them, then it is possible to build without a `main` function.

Currently an empty file has to be given to the cmake `app` target. Otherwise, cmake throws the following error ` No SOURCES given to target: app`.

# Solution
Allow to build an app consisting only of `prj.conf`, `dt_overlay` and `CMakeLists.txt`:
```
cmake_minimum_required(...)
find_package(...)
project(...)
```
The rest is done via extension in `zephyr/module.yml`. It is not necessary to add an empty source file to the `app` target.

# Alternatives
If devs are doing this approach, then they need to add an empty file to target `app`.

clamattia, 2024-11-12:
> , but just using modules with no project code?

Why not?

Different apps provide different `prj.conf`s and different `devicetree overlay`s.
Rest is implemented in the modular way (like zephyr itself).

clamattia, 2025-03-12:
If you think this is not worth, let me know. I will close the issue.

The fix would be pretty easy. When creating the `app` target, just add add an empty file to it. Maybe there is some cleaner cmake way of doing it. But that would work at least.

pdgendt, 2025-03-12:
There's already an empty file that you can easily use 🤷 

```cmake
target_sources(app PRIVATE ${ZEPHYR_BASE}/misc/empty_file.c)
```

clamattia, 2025-03-12:
> There's already an empty file that you can easily use 🤷
> 
> target_sources(app PRIVATE ${ZEPHYR_BASE}/misc/empty_file.c)

Would zephyr be willing to add this to their code?

clamattia, 2025-03-12:
I would do a PR, but I can not find, where the `app` target is created.

Alternatively, if you decide this is the applications job to add the empty file with `target_sources(app PRIVATE ${ZEPHYR_BASE}/misc/empty_file.c)` I can close this issue.

Thank you.

pdgendt, 2025-03-12:
> Alternatively, if you decide this is the applications job to add the empty file with `target_sources(app PRIVATE ${ZEPHYR_BASE}/misc/empty_file.c)` I can close this issue.

This is what I would suggest.

nashif, 2025-03-13:
> > Alternatively, if you decide this is the applications job to add the empty file with `target_sources(app PRIVATE ${ZEPHYR_BASE}/misc/empty_file.c)` I can close this issue.
> 
> This is what I would suggest.

why not just create an empy file in the app directory instead of relying on something that is considered internal to Zephyr?

pdgendt, 2025-03-13:
> why not just create an empy file in the app directory instead of relying on something that is considered internal to Zephyr?

Sure, my point is that this is trivial, and I think having the error message `No SOURCES given to target: app` is more helpful than harmful.

clamattia, 2025-03-13:
Decision was made, that zephyr will not add this.
Thank you

closed 2025-03-13 by clamattia