zephyrproject-rtos/zephyr #80989: RFC: New Testing Approach for Power Management Subsystem.

gbarkadiusz, 2024-11-06:
## Introduction

This proposal suggests a new method for testing the Power Management (PM) subsystem in Zephyr. The approach enhances the testing environment by adding an external power monitor to verify that the system reaches expected power states accurately. This improvement aims to make power management tests more reliable and to provide precise measurements for validating power state transitions.

### Problem description
The Zephyr Power Management subsystem provides multiple sleep modes for efficient power usage. However, reliably confirming that a device enters each specific power state is challenging. For example, the `stm32l562e_dk` board supports multiple power states (`k_cpu_idle`, `stop_0`, `stop_1`, `stop_2`), but current test methods are limited in confirming that each state is reached as expected.

### Proposed change
This proposal involves integrating a `Power Monitor` to track current consumption during testing. Using the `stm32l562e_dk` board’s onboard `Power Monitor`, the test setup can measure current in each power state. Pytest will then calculate the Root Mean Square (RMS) value of each detected power state, making it easier to verify whether each state has been achieved. This allows for accurate state detection and validation based on measured current consumption.

## Detailed RFC
An external `Power Monitor` will be added to measure the current of the target device during tests. The `stm32l562e_dk` board, which has an onboard `Power Monitor`, will be used for testing. A script named `pwsh_stm32l562.py` will facilitate communication with the board, handling the data retrieval and current measurement.

### Proposed change (Detailed)
The `Power_Monitor` abstract class serves as a base for implementing this functionality. It requires the following methods to be implemented:

`init_power_monitor(device_id: str) -> bool:`
 Initializes the power monitor for a specific device ID.
`measure_current(duration: int):`
 Measures current over a given duration.
`get_data(duration: int) -> List[float]:`
 Retrieves an array of current measurements over the specified duration, returning values in amperes.

**Setup and Configuration**
[image]
**Documentation:**
STM32L562E-DK Evaluation Board: [STMicroelectronics STM32L562E-DK](https://www.st.com/en/evaluation-tools/stm32l562e-dk.html)
STM32L562E-DK User Manual: [User Manual DM0063555](https://www.st.com/resource/en/user_manual/dm00635554-discovery-kit-with-stm32l562qe-mcu-stmicroelectronics.pdf)
Power Monitor Firmware Documentation: [Getting Started with PowerShield Firmware](https://www.st.com/resource/en/user_manual/um2269-getting-started-with-powershield-firmware-stmicroelectronics.pdf)
**Configuration:** Based on the documentation provided, the `stm32l562e_dk` board should be configured as follows:
[image]

Three tests demonstrate this approach, each designed to validate different aspects of power management:

**_pm.states:_** Verifies the transition to all available power states on `stm32l562e_dk`.
**_pm.residency_time:_** Tests the residency time of each power state.
**_pm.wakeup_timer:_** Tests the RTC alarm’s ability to wake the CPU from the lowest power state.
Each test calculates RMS current values for each detected state and compares them to expected values, which were manually set based on experimental data.

**Test: pm.states**
This test is designed to verify all available power states on the `stm32l562e_dk` board. The results from the `measure_current` function can be found in the `pm_states.csv` file.
[image]
The chart above illustrates the measured current consumption during the execution of the `pm_states` test. Key elements of the chart are highlighted as follows:

**_Startup:_** The current peak observed at the beginning of the test.
**_Boot Delay:_** The delay associated with the additional configuration set to `CONFIG_BOOT_DELAY=500.`
**_k_cpu_idle:_** The low power state achieved using `k_msleep(Tms)`, where `Tms` is shorter than the minimum residency time for the `stop_0` power state defined in the device tree source `(.dts)` file.
**_Wake Up and Print Log:_** The target board wakes up between each power state and logs the corresponding messages.
**_stop_0, stop_1, stop_2, active_state:_** The measured current consumption for each respective power state.

During the test, pytest detects each step and calculates the RMS (Root Mean Square) current for every detected state. The test then compares the calculated RMS values against the expected values, which have been determined manually through experimental observation

**To Reproduce:**

1. Configure the `stm32l562e_dk` board according to the instructions provided above.
2. Connect both USB ports of the board to your host system. The board will appear as two separate targets:
    - `Power Monitor`: e.g., `/dev/ttyACM0` or `/dev/serial/by-id/usb-STMicroelectronics_PowerShield__Virtual_ComPort_in_FS_Mode__FFFFFFFEFFFF-if00`
    - `Target`: e.g., `/dev/ttyACM1` or `/dev/serial/by-id/usb-STMicroelectronics_STLINK-V3_004##############39-if02`
3. Copy the path of the `Power Monitor` and update the `<path_to_power_monitor>` in the `tests/subsys/pm/power_states/testcase.yaml` file under the `pytest_args` section as follows:
```
pytest_args: [--powershield=<path_to_power_monitor>]
```
4. Run the following command to execute the test:
```
twister --device-testing --device-serial <path_to_target> -p stm32l562e_dk -T tests/subsys/pm/power_states/ -x=CONFIG_BOOT_DELAY=500 -vv
```
**Pytest log:**
```
DEBUG   - PYTEST: k_cpu_idle:
DEBUG   - PYTEST: Expected RMS: 57.00 mA
[... 93 lines cut ...]
(1-2) As previously.
3. Copy the path of the `Power Monitor` and update the `<path_to_power_monitor>` in the `tests/subsys/pm/power_wakeup_timer/testcase.yaml` file under the `pytest_args` section as follows:
```
pytest_args: [--powershield=<path_to_power_monitor>]
```
4. Run the following command to execute the tests:
```
twister --device-testing --device-serial <path_to_target> -p stm32l562e_dk -T tests/subsys/pm/power_wakeup_timer/ -x=CONFIG_BOOT_DELAY=500 -vv
```
**Pytest_log:**
```
DEBUG   - PYTEST: Expected RMS: 0.26 mA
DEBUG   - PYTEST: Tolerance: 0.03 mA
DEBUG   - PYTEST: Current RMS: 0.27 mA
DEBUG   - PYTEST: active:
DEBUG   - PYTEST: Expected RMS: 95.00 mA
DEBUG   - PYTEST: Tolerance: 9.50 mA
DEBUG   - PYTEST: Current RMS: 91.90 mA
DEBUG   - PYTEST: PASSED
```

All the tests mentioned above are available in the following pull request: https://github.com/zephyrproject-rtos/zephyr/pull/80692. This pull request is currently in draft status and will remain so until the RFC is reviewed and applied.

### Dependencies
This change requires access to the `stm32l562e_dk` board and an external (onboarded) `Power Monitor`. It may also necessitate modification of test scripts if additional `power monitors` are used.

nashif, 2024-11-13:
@erwango @bjarki-andreasen @ceolin  please take a look

bjarki-andreasen, 2024-11-13:
I agree! We need to add power usage testing!

Some notes from my side:
* We should measure power usage in watts (joules) :) amps only tell half the story
* We should support power monitors as flexibly as we support debuggers in west. I would likely opt for a https://www.joulescope.com/ for testing
* We should be testing more than power states, power usage of UARTs, I2C, whole applications really, is really important as well :) We should be able to just "enable power measuring" for any test or application at any time :)

I do have ideas for solutions for this as well but that's for later :)

bjarki-andreasen, 2024-11-13:
Can't help it, I would tie expected power measurement in unit tests to output from probe like:

device output:
```
*** Booting Zephyr OS build v4.0.0-rc2 ***
Running TESTSUITE rtc_api_helpers
===================================================================
START - test_validate_calibration_from_frequency
 EXPECT POWER 1.2V 20uA at 0.200 seconds
 PASS - test_validate_calibration_from_frequency in 0.000 seconds
===================================================================
```

power measurement log output
```
POWER 1.16V 18uA at 0.000
POWER 1.16V 17uA at 0.100
POWER 1.10V 22uA at 0.200
```

to then be combined and validated after the test has passed by some higher level runner (west)

gbarkadiusz, 2024-11-15:
> I agree! We need to add power usage testing!
>
> Some notes from my side:
>
> * We should measure power usage in watts (joules) :) amps only tell half the story
> * We should support power monitors as [...]

Regarding those pointed things.
Watts are a good choice for measurement, but they require measuring the target board's supply voltage, which makes the setup a bit more complex. Additionally, I’m not sure if West is the best tool for handling this functionality. It could lead to potential issues with synchronization between West and Twister. Moreover, analyzing the results through Twister (as the tool responsible for testing) might be problematic—but I’m not certain, just raising the question.

bjarki-andreasen, 2024-11-18:
> Regarding these pointed things. Watts are a good choice for measurement, but they require measuring the target board's supply voltage, which makes the setup a bit more complex.

Yeah, with some probes it may not be possible to measure joules, but in these cases, the test suites should still indicate a number of expected jules, and the user should provide a "fixed" voltage for twister to use to calculate joules from the current measurement :)

> Additionally, I’m not sure if West is the best tool for handling this functionality. It could lead to potential issues with synchronization between West and Twister. Moreover, analyzing the results [...]

Looking into it, twister is the tool used to manage tests, so it would be the place to add probes, alongside harnesses IMO :)

> Another question to consider is whether we should focus on measuring the relative changes in power consumption between different power modes/states, or if we should aim to measure the specific values for each application state. For example, should we also measure the power usage of every peripheral available on a specific target?

I would strongly encourage specific measurements :) Test suites are about as controlled an environment we can get, changes somewhere which unexpectedly enables/disables some peripheral should be caught as an anomaly.

gbarkadiusz, 2024-11-20:
@bjarki-andreasen

I noticed you updated the comment—thank you for that!

I have another question regarding the testing setup. I saw that you're using the ztest framework. Could you please explain how you'd like to integrate the measuring functionality with the ztest framework? I'm having some trouble understanding the connection and would appreciate your guidance.

Thank you!

bjarki-andreasen, 2024-11-20:
> @bjarki-andreasen
> 
> I noticed you updated the comment—thank you for that!
> 
> I have another question regarding the testing setup. I saw that you're using the `ztest` framework. Could you please explain how you'd like to integrate the measuring functionality with the `ztest` framework? I'm having some trouble understanding the connection and would appreciate your guidance.
> 
> Thank you!
> 
> PS: I can imagine using Serial/Uart to correspond zephyr app with python script to trigger the measurement. e.g.:
> 
> ```
> void ztest_probe_power(int duration_us, int min_uJ, int max_uJ)
> {
>     const struct device *uart_dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_shell_uart));
>     char cmd[64];
>     snprintf(cmd, sizeof(cmd), "measure_power %d %d %d\n", duration_us, min_uJ, max_uJ);
> 
>     // Send the command over UART
>     for (size_t i = 0; i < strlen(cmd); ++i) {
>         uart_poll_out(uart_dev, cmd[i]);
>     }
> 
>     // Wait for acknowledgment and result
>     char response[128];
>     size_t len = 0;
>     while (len < sizeof(response) - 1) {
>         if (uart_poll_in(uart_dev, &response[len]) == 0) {
>             len++;
>             if (response[len - 1] == '\n') {
>                 break; // End of response
>             }
>         }
>     }
>     response[len] = '\0';
> }
> ```

To send back an answer, we would likely need to enable shell support, and add a shell backend for testing, something that would look like:
```
===================================================================
START - test_power
 PROBE CURRENT AVG - 100us 12uA 14uA
uart:~$ ztest PASS
 PASS - test_power
===================================================================
```
in the shell, where we would listen for `PROBE CURRENT AVG - 100us 12uA 14uA`, start the measurement, and send back the status `PASS.` `ztest` would be the subcommand, taking one parameter, `PASSED` or `FAILED`.

The code could look like:
```
static K_SEM_DEFINE(shell_result, 0, 1);
static bool passed;

void ztest_probe_power(int duration_us, int min_uJ, int max_uJ)
{
        /* Request measurement */
        TC_PRINT("measure_power %d %d %d\n", duration_us, min_uJ, max_uJ);

        /* Wait for shell input */
        k_sem_take(&shell_result, K_FOREVER);

        /* Check result */
        if (!passed) {
                ztest_test_fail();
        }
}
```
excluding the implementation of the `ztest` shell backend.

However, test suites currently run with only console enabled, since its simple, and all we need if the entire test suite is run on target. Enabling the shell takes quite a bit of ROM, and may make some targets incompatible with the test suite. For that reason, I would compromise to something like:
```
void ztest_probe_power(int duration_us, int min_uJ, int max_uJ)
{
        /* Request measurement */
        TC_PRINT("measure_power %d %d %d\n", duration_us, min_uJ, max_uJ);
        
        /* wait "some time" for twister to start test */
        k_sleep(K_MSEC(10));

        /* wait for duration */
        k_sleep(K_USEC(duration_us));
}
```
Its likely less stable then the shell approach, but its an option :)

gbarkadiusz, 2024-11-21:
I see your point. A Zephyr application can request measurements at any time using methods like TC_PRINT or shell commands. The challenge lies in capturing these requests and performing the measurements. We need a mechanism to handle such commands. As you mentioned, we could use the shell backend (e.g., UART) to send commands, but we still need something on the other end to process these requests. Additionally, I’m considering whether we should use the same serial port for communication with the Zephyr application and for verifying tests via Twister. It could make the Twister log messy.

Thinking out loud:
A straightforward approach could be to leverage the `pytest` harness. For example:
```
import pytest
from twister_harness import DeviceAdapter

def test_measure(dut: DeviceAdapter):
    # Read lines from the device's output
    output_lines = dut.readlines()

    # Handle the command from the output
    for line in output_lines:
        if "PROBE POWER" in line:
            # Parse the command, execute measurement logic, and decide pass/fail
            handle_probe_power_command(line, dut)
```
The `handle_probe_power_command` function could process the command, execute the measurement, and communicate the result back to the device or log it appropriately. This keeps the process modular and aligns well with pytest's structure.

bjarki-andreasen, 2024-11-21:
> I see your point. A Zephyr application can request measurements at any time using methods like TC_PRINT or shell commands. The challenge lies in capturing these requests and performing the measurements. We need a mechanism to handle such commands. As you mentioned, we could use the shell backend (e.g., UART) to send commands, but we still need something on the other end to process these requests. Additionally, I’m considering whether we should use the same serial port for communication with the Zephyr application and for verifying tests via Twister. It could make the Twister log messy.
> 
> Thinking out loud: A straightforward approach could be to use the `pytest` harness. For example:
> 
> ```
> import pytest
> from twister_harness import DeviceAdapter
> 
> def test_measure(dut: DeviceAdapter):
>     # Read lines from the device's output
>     output_lines = dut.readlines()
> 
>     # Handle the command from the output
>     for line in output_lines:
>         if "PROBE POWER" in line:
>             # Parse the command, execute measurement logic, and decide pass/fail
>             handle_probe_power_command(line, dut)
> ```
> 
> The `handle_probe_power_command` function could process the command, execute the measurement, and communicate the result back to the device or log it appropriately. This keeps the process modular and aligns well with pytest's structure.

This looks pretty nice :)

bjarki-andreasen, 2024-11-22:
I don't think we should limit ourselves at all :) We could add support for power over timem min/max voltages, average current, rms current etc.

```
ztest_probe_power(); /* in joules */
ztest_probe_current_rms() /* in amps */
ztest_probe_voltage_min_max() /* in volts */
```
Any test can specify whatever it needs, and will get its way if the probe supports the requested measurement.

gbarkadiusz, 2024-11-22:
> I don't think we should limit ourselves at all :) We could add support for power over time, min/max voltages, average current, rms current etc.
> 
> ```
> ztest_probe_power(); /* in joules */
> ztest_probe_current_rms() /* in amps */
> ztest_probe_voltage_min_max() /* in volts */
> ```
> 
> Any test can specify whatever it needs, and will get its way if the probe supports the requested measurement.

Sounds great ;)

bjarki-andreasen, 2024-11-22:
One little update:
```
ztest_probe_power_avg(); /* in watts */
ztest_probe_energy(); /* in joules */
```
power is in watts :)

hakehuang, 2024-11-26:
https://github.com/zephyrproject-rtos/zephyr/pull/77752#pullrequestreview-2274219838

gbarkadiusz, 2024-11-26:
@bjarki-andreasen   FYI.
I performed some exercises on triggering measurements from the Zephyr application. 
```

void ztest_probe_power(int duration_us, int min_uJ, int max_uJ)
{
    /* Request measurement */
    printk("measure_power %d %d %d\n", duration_us, min_uJ, max_uJ);
        
    /* wait for duration */
    k_sleep(K_MSEC(duration_us));
}

ZTEST(testsuite, test_name_of_testcase)
{
    ztest_probe_power(2000000, 1, 3);
}

ZTEST_SUITE(testsuite, NULL, NULL, NULL, NULL, NULL);
```

pytest:
```

def test_foo(dut: DeviceAdapter, request):
    powershield_device = request.config.getoption("--powermonitor")
    PM_Device = PowerMonitor()
    PM_Device.setup(power_device_path = powershield_device)
    while True:
        lines = dut.readlines() 
        # Find the line containing 'measure_power'
        measure_power_line = next((line for line in lines if 'measure_power' in line), None)
        if measure_power_line:
            parts = measure_power_line.split()
            # Extract the values
            command = parts[0]  # "measure_power"
            duration = int(parts[1]) /1000000 # us
            print(duration)
            value2 = int(parts[2])  # 
            value3 = int(parts[3])  # 

            PM_Device.measure(measure_unit = PM_Device.power_monitor_conf.MeasureUnit.Power, time=duration, reset=False)
            print(f"Power: {PM_Device.get_measured_data(unit = PM_Device.power_monitor_conf.MeasureUnit.POWER)} [W]")
            break
        else:
            print("measure_power not found.")
```
Unfortunately, I found a limitation in the onboard power monitor's firmware. Each 'start' command resets the target and starts the measurement, so it can't be triggered during the application.

bjarki-andreasen, 2024-11-26:
> @bjarki-andreasen FYI. I performed some exercises on triggering measurements from the Zephyr application.
> 
> ```
> 
> void ztest_probe_power(int duration_us, int min_uJ, int max_uJ)
> {
>     /* Request measurement */
>     printk("measure_power %d %d %d\n", duration_us, min_uJ, max_uJ);
>         
>     /* wait for duration */
>     k_sleep(K_MSEC(duration_us));
> }
> 
> ZTEST(testsuite, test_name_of_testcase)
> {
>     ztest_probe_power(2000000, 1, 3);
> }
> 
> ZTEST_SUITE(testsuite, NULL, NULL, NULL, NULL, NULL);
> ```
> 
> pytest:
> 
> ```
> 
> def test_foo(dut: DeviceAdapter, request):
>     powershield_device = request.config.getoption("--powermonitor")
>     PM_Device = PowerMonitor()
>     PM_Device.setup(power_device_path = powershield_device)
>     while True:
>         lines = dut.readlines() 
>         # Find the line containing 'measure_power'
>         measure_power_line = next((line for line in lines if 'measure_power' in line), None)
>         if measure_power_line:
>             parts = measure_power_line.split()
>             # Extract the values
>             command = parts[0]  # "measure_power"
>             duration = int(parts[1]) /1000000 # us
>             print(duration)
>             value2 = int(parts[2])  # 
>             value3 = int(parts[3])  # 
> 
>             PM_Device.measure(measure_unit = PM_Device.power_monitor_conf.MeasureUnit.Power, time=duration, reset=False)
>             print(f"Power: {PM_Device.get_measured_data(unit = PM_Device.power_monitor_conf.MeasureUnit.POWER)} [W]")
>             break
>         else:
>             print("measure_power not found.")
> ```
> 
> Unfortunately, I found a limitation in the onboard power monitor's firmware. Each 'start' command resets the target and starts the measurement, so it can't be triggered during the application. I need to find another solution.

In that case, could the current measurement not be started and running constantly in the "background"? We don't need to actually start it only when we care about the samples right?

hakehuang, 2024-11-26:
> Absolutely we need to support it :) flashing, voltage and current measurement in one device, really neat :)

I will add same support when the initial pr merged https://github.com/zephyrproject-rtos/zephyr/pull/80692

hakehuang, 2024-11-26:
> Also, we could write a zephyr sample application which uses an on-board ADC for power measurement, our own "in-house" power probe firmware :)

it is difficult, for power measurement, we need some additional circuits, now only lpcxpresso55s36 has such circuits on board

gbarkadiusz, 2024-11-26:
> In that case, could the current measurement not be started and running constantly in the "background"? We don't need to actually start it only when we care about the samples right?

That was my first thought. WIP. :)

bjarki-andreasen, 2024-11-26:
> > Also, we could write a zephyr sample application which uses an on-board ADC for power measurement, our own "in-house" power probe firmware :)
> 
> it is difficult, for power measurement, we need some additional circuits, now only lpcxpresso55s36 has such circuits on board

We have devicetree models like `voltage-divider`, I would think we could add `shunt-resistor` or `current-shunt-resistor` or something to convert voltage to current :)

bjarki-andreasen, 2024-11-26:
> > > > Also, we could write a zephyr sample application which uses an on-board ADC for power measurement, our own "in-house" power probe firmware :)
> > > 
> > > 
> > > it is difficult, for power measurement, we need some additional circuits, now only lpcxpresso55s36 has such circuits on board
> > 
> > 
> > We have devicetree models like `voltage-divider`, I would think we could add `shunt-resistor` or `current-shunt-resistor` or something to convert voltage to current :) some boards do have shunt resistors, or at least a very obvious jumper to replace with shunt resistor
> 
> The problem is rather, the adc is not working in low power state.

The https://www.nxp.com/docs/en/user-manual/UM11673.pdf is a standalone device, so it will be running a completely different firmware than the DUT, its not measuring itself :)

gbarkadiusz, 2024-11-28:
I performed tests using the Zephyr app and triggered commands by passing them to pytest. The integration worked quite well, but I encountered an issue— I didn’t know how to pass the result of the pytest test back to the Zephyr app. As a workaround, I used a method where Zephyr triggers a command, such as `measure_current_rms`, with two parameters: `[min_uA, max_uA]`. The measured value of the current RMS should then be between these two values. To verify this condition, I used pytest's assertion, like this:

```
assert min_uA < current_rms < max_uA
```

From my perspective, this approach seems valid, but I would appreciate your opinion on whether it's correct.
Since we cannot use ztest directly in this case, the application is a "pure" Zephyr application, structured as follows:
```
#include <string.h>
#include <zephyr/kernel.h>

void ztest_probe_current_rms(int duration_us, int min_uA, int max_uA)
{
    /* Request measurement */
    printk("measure_current_rms %d %d %d\n", duration_us, min_uA, max_uA); 
}

int main(void)
{
    ztest_probe_current_rms(1600000, 25, 30); /*  duration = 1.6s, min_uA = 25, max_uA = 30*/

    // Sleep for 3000ms
    k_msleep(3000);
    
    /*  Anti-sleeping loop */
    while(1)
    {
        arch_nop();
    }

    return 0;
}
```

gbarkadiusz, 2024-11-29:
> Forgot to add a solution here :) I found this https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/testsuite/ztest/src/ztest_shell.c so there is already a shell for test execution, surely we can extend it with test feedback :)

Ztest Shell supports executing a particular test via the shell. However, there are still valid questions about how to trigger the measurement from the test, gather the result, and then determine whether the test passes or fails.

bjarki-andreasen, 2024-11-29:
> > Forgot to add a solution here :) I found this https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/testsuite/ztest/src/ztest_shell.c so there is already a shell for test execution, surely we can extend it with test feedback :)
> 
> Ztest Shell supports executing a specific test via the shell. However, there are still valid questions about how to trigger measurements within the test, collect the results, and determine whether the test passes or fails.
> 
> The main challenge lies in how to communicate between Ztest and the measurement environment.
> 
> **Using Pytest** We can pass the trigger from the Zephyr application to Pytest through the console (e.g., by printing measure_power), and then read the output in Pytest using dut.readlines(). However, the problem arises in how to send the result back to Ztest in order to determine whether the test passes or fails.
> 
> **Using Shell** There is support to execute a specific test via the shell. Pytest can also be used to run the test, collect the triggered command, measure the probe, and perform the necessary actions. The challenge here is determining how to send the result back to Ztest while the test is running.
> 
> **Using Console Subsystem** The console subsystem supports bidirectional communication via serial. In this approach, we need to figure out how to measure the required value in background using twister/west command ?

I looked into using the shell for control, and came up with:
```
static int cmd_control_testcase(const struct shell *sh, size_t argc, char **argv)
{
	const char *action;

	action = argv[1];

	if (!strcmp(action, "pass")) {
		ztest_test_pass();
	} else if (!strcmp(action, "skip")) {
		ztest_test_skip();
	} else if (!strcmp(action, "fail")) {
		ztest_test_fail();
	} else {
		shell_error("unsupported action %s", action);
		return -EINVAL;
	}

	return 0;
}

SHELL_STATIC_SUBCMD_SET_CREATE(
	sub_ztest_harness_cmds,
	SHELL_CMD_ARG(control_testcase, NULL, "", cmd_control_testcase, 1, 0),
	SHELL_SUBCMD_SET_END /* Array terminated. */
);

SHELL_CMD_REGISTER(ztest_harness, &sub_ztest_harness_cmds, "", NULL);
```
creating a simple low overhead shell which is only used for feedback (no running test suites etc. like the full ztest shell). With this, the DUT prints "PROBE POWER ...", pytest does the measurements, and writes back `ztest_harness control_testcase pass` or `ztest_harness control_testcase fail`

Definately all for doing something way simpler using the console directly, primarily to save on ROM/RAM, but we would need to check what the actual difference in ROM/RAM is comparing the shell to the console solution to choose (the shell one seems optimal to me if ROM/RAM is OK)

gbarkadiusz, 2024-11-29:
> I looked into using the shell for control, and came up with:
>
> ```
> static int cmd_control_testcase(const struct shell *sh, size_t argc, char **argv)
> {
> 	const char *action;
>
> 	action = argv[1];
>
> 	if [...]

Have you tested this approach?
Quick fix: 
```
- shell_error("unsupported action %s", action);
+ shell_error(sh, "unsupported action %s", action);
```
but:
```
uart:~$ ztest_harness control_testcase pass
 ERROR: cannot pass in test phase 'framework()', bailing
uart:~$ ztest_harness control_testcase skip
 ERROR: cannot skip in test phase 'framework()', bailing
uart:~$ ztest_harness control_testcase fail
 ERROR: cannot fail in test phase 'framework()', bailing
```

bjarki-andreasen, 2024-11-29:
> Have you tested this approach? Quick fix:
> 
> ```
> - shell_error("unsupported action %s", action);
> + shell_error(sh, "unsupported action %s", action);
> 
> - SHELL_CMD_ARG(control_testcase, NULL, "", cmd_control_testcase, 1, 0),
> + SHELL_CMD_ARG(control_testcase, NULL, "", cmd_control_testcase, 1, 1),
> ```
> 
> but:
> 
> ```
> uart:~$ ztest_harness control_testcase pass
>  ERROR: cannot pass in test phase 'framework()', bailing
> uart:~$ ztest_harness control_testcase skip
>  ERROR: cannot skip in test phase 'framework()', bailing
> uart:~$ ztest_harness control_testcase fail
>  ERROR: cannot fail in test phase 'framework()', bailing
> ```

No, I just wanted to show how to set up the shell, are you in a test case? to test it, do:
```
ZTEST(suite, test)
{
        TC_PRINT("you can control test now\n");
        k_msleep(10000);
        TC_PRINT("too late\n");
}
```
The functions only work while a test case is actively running

gbarkadiusz, 2024-11-29:
The problem lies on enabled shell. If it is enabled, the tests won't be executed. After boot zephyr, the app goes to shell immediately.

bjarki-andreasen, 2024-11-29:
Are you selecting `CONFIG_ZTEST_SHELL=y`? it should be `CONFIG_ZTEST_SHELL=n`, only `CONFIG_SHELL=y` should be enabled

gbarkadiusz, 2024-11-29:
Quick update, It seems to work.
I added
```
static void *shell_setup(void)
{
	/* Let the shell backend initialize. */
	k_msleep(100);
	return NULL;
}

ZTEST_SUITE(testsuite, NULL, shell_setup, NULL, NULL, NULL);
```

pytest
```
import pytest, time
from twister_harness import DeviceAdapter
from twister_harness import Shell

def test_catch_and_respons(dut: DeviceAdapter):
    while True:
        lines = dut.readlines()  # Adjust timeout as needed
        # Find the line containing 'measure_power'
        measure_power_line = next((line for line in lines if 'measure_power' in line), None)
        if measure_power_line:
            print(f"Found line: {measure_power_line}")
            shell = Shell(dut)
            shell.exec_command("ztest_harness control_testcase fail")
            break

```
the log:

```
DEBUG   - PYTEST: DEBUG: Flashing finished
DEBUG   - PYTEST: -------------------------------- live log call ---------------------------------
DEBUG   - PYTEST: DEBUG: #: *** Booting Zephyr OS build v3.7.0-rc2-3875-gac1cc17ae90a ***
DEBUG   - PYTEST: DEBUG: #: Running TESTSUITE testsuite
DEBUG   - PYTEST: DEBUG: #: ===================================================================
DEBUG   - PYTEST: DEBUG: #: 
DEBUG   - PYTEST: DEBUG: #: uart:~$ START - test_name_of_testcase
DEBUG   - PYTEST: DEBUG: #: measure_power 100 1 3
DEBUG   - PYTEST: Found line: measure_power 100 1 3
DEBUG   - PYTEST: DEBUG: #: ztest_harness control_testcase fail
DEBUG   - PYTEST: DEBUG: #: FAIL - test_name_of_testcase in 0.025 seconds
DEBUG   - PYTEST: DEBUG: #: ===================================================================
DEBUG   - PYTEST: DEBUG: #: TESTSUITE testsuite failed.
DEBUG   - PYTEST: DEBUG: #: ------ TESTSUITE SUMMARY START ------
DEBUG   - PYTEST: DEBUG: #: SUITE FAIL -   0.00% [testsuite]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.025 seconds
DEBUG   - PYTEST: DEBUG: #: - FAIL - [testsuite.test_name_of_testcase] duration = 0.025 seconds
DEBUG   - PYTEST: DEBUG: #: ------ TESTSUITE SUMMARY END ------
DEBUG   - PYTEST: DEBUG: #: ===================================================================
DEBUG   - PYTEST: DEBUG: #: RunID: 59cf315eb1495eeae656f6b34c7d2282
DEBUG   - PYTEST: DEBUG: #: PROJECT EXECUTION FAILED

```

bjarki-andreasen, 2024-11-29:
This is a bit more graceful (though not perfect) for waiting on the shell:
```
	const struct shell *sh = shell_backend_get(0);

	zassert_true(WAIT_FOR(shell_ready(sh), 100000, k_msleep(1)));
```

ceolin, 2024-12-02:
> @bjarki-andreasen I thought about the unit of measured value. We have several options. **Pure Average of current. (As you said, it is a half of story).** ![image](https://private-user-images.githubusercontent.com/122030181/388857492-438e893f-4ce8-4243-bc21-2438c4d76a15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDM1NzcsIm5iZiI6MTczMzEwMzI3NywicGF0aCI6Ii8xMjIwMzAxODEvMzg4ODU3NDkyLTQzOGU4OTNmLTRjZTgtNDI0My1iYzIxLTI0MzhjNGQ3NmExNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMTM0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jODcyZDUzNjEyZTRmNjdmMmY1YjhmYWQ0NzJhZGNiZjY5YjY1OGVmNDY0ZDNjOGY3NzFmNTE3YTgzODg1ZmVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.ZRlQ00DV2uYlvvbPcMSrV_myOZuyLTZ1xCQXnJ-h_8I) But the power can be calculated as: ![image](https://private-user-images.githubusercontent.com/122030181/388857589-142de36d-cba2-4638-b836-765ae7110361.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDM1NzcsIm5iZiI6MTczMzEwMzI3NywicGF0aCI6Ii8xMjIwMzAxODEvMzg4ODU3NTg5LTE0MmRlMzZkLWNiYTItNDYzOC1iODM2LTc2NWFlNzExMDM2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMTM0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNDJjYmE3YTQ0YWY5NmNhNTNmMDFhZGIwMWM3N2MwZDYzMTFjM2ViMTg5ZWNmYjI5MDk3NWVmMDQ3NmYyMDJlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RTydEmR7HWm4SrZiyxR-5UA47EcXNoLusP9jMQki-Ak)
> 
> **RMS (Root Mean Square)** is a better metric for fluctuating signals as it considers the variation in current over time. RMS current can be calculated as: ![image](https://private-user-images.githubusercontent.com/122030181/388857051-c440aa2c-6e9a-458e-9b68-b8586c3f8b85.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDM1NzcsIm5iZiI6MTczMzEwMzI3NywicGF0aCI6Ii8xMjIwMzAxODEvMzg4ODU3MDUxLWM0NDBhYTJjLTZlOWEtNDU4ZS05YjY4LWI4NTg2YzNmOGI4NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMTM0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMmY4NTRiNDExNWJiNDViZjMxMzA3Y2QzMDJlZGVhZGI1ZDE4YzliNmQzMTk5MjVhOTUxMGE1OWZkMDdjOTJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.K3nN1X-Y1fdOOzDlUnUzQDt_0eVTAQy-nZZBtvwSp7g) and the power can be calculated a: ![image](https://private-user-images.githubusercontent.com/122030181/388857178-da11540c-938a-491e-be60-d9cfb194532e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDM1NzcsIm5iZiI6MTczMzEwMzI3NywicGF0aCI6Ii8xMjIwMzAxODEvMzg4ODU3MTc4LWRhMTE1NDBjLTkzOGEtNDkxZS1iZTYwLWQ5Y2ZiMTk0NTMyZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMTM0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZjY4NmM5NTdkODc0OWMwZjlmZTU1Zjk5ZTE4OWNmNDdiMGE3ODczNzFjYjhhODRiZTYyMjdmZWY0YWI4MzlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cSRACrCcd3elMT7sZL6deOtbaiDDFt3yW3WkYlDTz3Q)
> 
> **Total Energy Instead of Power.** If the measurement duration ![image](https://private-user-images.githubusercontent.com/122030181/388858176-88536dbf-3d7a-423c-849f-04bbf023d7ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDM1NzcsIm5iZiI6MTczMzEwMzI3NywicGF0aCI6Ii8xMjIwMzAxODEvMzg4ODU4MTc2LTg4NTM2ZGJmLTNkN2EtNDIzYy04NDlmLTA0YmJmMDIzZDdhYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMTM0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mY2U2NmJhYTc4NzI5NmY4YmU5NGFkYzZmNDk0ZTVlODk5ZDdhNGVjNjM4YzM5NTkwMDA2NDBhNGVmZmI5YWY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.p37Sv4ACpgnhpOBd57tQOrze1FawGiQfkBuj9ky_LSI) is known, we can calculate the total energy (𝐸) consumed over the period. This gives a clearer picture of energy usage: ![image](https://private-user-images.githubusercontent.com/122030181/388858725-1ef8ac88-103f-421c-bfc1-d77f9f5f6d80.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDM1NzcsIm5iZiI6MTczMzEwMzI3NywicGF0aCI6Ii8xMjIwMzAxODEvMzg4ODU4NzI1LTFlZjhhYzg4LTEwM2YtNDIxYy1iZmMxLWQ3N2Y5ZjVmNmQ4MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMTM0MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MmM4NmNhYTUyNTA5NTU0Y2EwMGRjMTk0YTYyNzFiYzk2M2VmMzJjZDVkYjFkOTBjZDgxY2IwYjI0YjFjNDk5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Mum4jnACTf5GgfOoqQtLpfPi3dG53l8dCafQmTdPRSk) where: ![image](https://private-user-images.githubusercontent.com/122030181/388858863-cebec587-4d95-44da-860a-0070ae09e303.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOi [... cut ...]
> The question is. How to calculate the expected value? Maybe it is not a good question at this time but let me treat this RFC as a brainstorm and place to share all ideas and warries. :)

Would be nice to support multiple types since they address different things. In general total energy is probably what people want because it is directly related to battery life, but it is not super helpful for debugging power states consumption, for this RMS and pure average of current are better.

I am wondering if RMS is the best measurement unit for testing sleep states, it seems that pure average of current is better for this.

ceolin, 2024-12-02:
@gbarkadiusz thanks ! really appreciate it.

gbarkadiusz, 2024-12-02:
> @gbarkadiusz thanks ! really appreciate it.

It's my pleasure ;)

gbarkadiusz, 2024-12-04:
@bjarki-andreasen @ceolin FYI. 
I encountered a new issue regarding triggering measurements from the Zephyr application after enabling the Power Management (PM) subsystem. It appears that Zephyr disables the shell while the system enters low-power states, causing the shell to become unresponsive. Maybe a quick fix can be to enable shell/uart as wakeup-source.

bjarki-andreasen, 2024-12-04:
> @bjarki-andreasen @ceolin FYI. I encountered a new issue regarding triggering measurements from the Zephyr application after enabling the Power Management (PM) subsystem. It appears that Zephyr [...]

Its more likely PM is suspending the UART device in some low power state as defined by the SoC, hence the shell becomes unresponsive :)

In any case, could the solution be to trigger the measurement while in a higher power state, set some timer to exit low power state, then enter low power? measurement is done by pytest, then system exits low power state and can receive the message :)

erwango, 2024-12-05:
> @bjarki-andreasen @ceolin FYI. I encountered a new issue regarding triggering measurements from the Zephyr application after enabling the Power Management (PM) subsystem. It appears that Zephyr [...]

Did you try to use the following configuration:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/boards/st/power_mgmt/serial_wakeup/boards/stm32l562e_dk.overlay ?

If not working then this is a regression

gbarkadiusz, 2025-01-09:
> Did you try to use the following configuration: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/boards/st/power_mgmt/serial_wakeup/boards/stm32l562e_dk.overlay ?
>
> If not working then [...]

“I tried to use this sample with the STM32L562E-DK, but it didn’t work. I might have been doing something wrong. WIP.

erwango, 2025-01-09:
> I tried to use this sample with the STM32L562E-DK, but it didn't work, or I might have done something wrong. The shell was not responding,

Ok. Would you mind raising a dedicated issue ?

gbarkadiusz, 2025-01-09:
Before that, I'd like to double check that sample with another board. But OFC I'll raise the issue with appropriate description.

gbarkadiusz, 2025-01-09:
@erwango 
After adding stm32l562e_dk to the sample.yaml file, the twister throws after flashing.
`Timed out while monitoring serial output on stm32l562e_dk/stm32l562xx`
So, It's time to raise the issue ;)

SzymonRichert, 2025-01-24:
@erwango @bjarki-andreasen @ceolin, can we  drive it to some initial merge? We can rework this later.

nashif, 2025-04-23:
closing this as done

closed 2025-04-23 by nashif