zephyrproject-rtos/zephyr #81001: nrf54l15dk: Failing to build with CONFIG_ENTROPY_GENERATOR

rettichschnidi, 2024-11-06:
**Describe the bug**

Building for nrf54l15dk fails when enabling Bluetooth and entropy drivers at the same time.

**To Reproduce**

1. `west build --pristine --board nrf54l15dk/nrf54l15/cpuapp zephyr/samples/hello_world/ -- -DCONFIG_BT=y -DCONFIG_ENTROPY_GENERATOR=y`

**Expected behavior**

Build should work. As it does e.g. for the `nrf52840dk`: `west build --pristine --board nrf52840dk/nrf52840 zephyr/samples/hello_world/ -- -DCONFIG_BT=y -DCONFIG_ENTROPY_GENERATOR=y`

**Impact**

Need to chose between BT and ENTROPY_GENERATOR.

**Logs and console output**

```
$ west build --pristine --board nrf54l15dk/nrf54l15/cpuapp zephyr/samples/hello_world/ -- -DCONFIG_BT=y -DCONFIG_ENTROPY_GENERATOR=y
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: <censored>/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.30.5
-- Found Python3: /home/<censored>/code/3rd-party/venv-zephyrproject/bin/python3 (found suitable version "3.12.6", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/<censored>/.cache/zephyr
-- Zephyr version: 4.0.0-rc2 (<censored>/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nrf54l15dk, qualifiers: nrf54l15/cpuapp
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/<censored>/.local/opt/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/<censored>/.local/opt/zephyr-sdk-0.16.8)
-- Found Dtc: /home/<censored>/.local/opt/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: <censored>/zephyrproject/zephyr/boards/nordic/nrf54l15dk/nrf54l15dk_nrf54l15_cpuapp.dts
unit address and first address in 'reg' (0x5004c000) don't match for /soc/peripheral@50000000/vpr@4c000/mailbox@1
-- Generated zephyr.dts: <censored>/zephyrproject/build/zephyr/zephyr.dts
-- Generated pickled edt: <censored>/zephyrproject/build/zephyr/edt.pickle
-- Generated zephyr.dts: <censored>/zephyrproject/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: <censored>/zephyrproject/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: <censored>/zephyrproject/build/zephyr/dts.cmake

warning: Deprecated symbol TINYCRYPT is enabled.

Parsing <censored>/zephyrproject/zephyr/Kconfig
Loaded configuration '<censored>/zephyrproject/zephyr/boards/nordic/nrf54l15dk/nrf54l15dk_nrf54l15_cpuapp_defconfig'
Merged configuration '<censored>/zephyrproject/zephyr/samples/hello_world/prj.conf'
Merged configuration '<censored>/zephyrproject/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '<censored>/zephyrproject/build/zephyr/.config'
Kconfig header saved to '<censored>/zephyrproject/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/<censored>/opt/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/<censored>/.local/opt/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done (3.1s)
-- Generating done (0.1s)
-- Build files have been written to: <censored>/zephyrproject/build
-- west build: building application
[1/189] Preparing syscall dependency handling

[2/189] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.0-rc2 (<censored>/zephyrproject/zephyr), build: v4.0.0-rc2-55-gd4b7bf986c6a
[93/189] Building C object zephyr/subsys/random/CMakeFiles/subsys__random.dir/random_entropy_device.c.obj
FAILED: zephyr/subsys/random/CMakeFiles/subsys__random.dir/random_entropy_device.c.obj 
In file included from <censored>/zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:98,
ccache /home/<censored>/.local/opt/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNRF54L15_XXAA -DNRF_APPLICATION -DNRF_CONFIG_CPU_FREQ_MHZ=128 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I<censored>/zephyrproject/build/zephyr/include/generated/zephyr -I<censored>/zephyrproject/zephyr/include -I<censored>/zephyrproject/build/zephyr/include/generated -I<censored>/zephyrproject/zephyr/soc/nordic -I<censored>/zephyrproject/zephyr/soc/nordic/nrf54l/. -I<censored>/zephyrproject/zephyr/soc/nordic/common/. -I<censored>/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -I<censored>/zephyrproject/zephyr/subsys/bluetooth -I<censored>/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I<censored>/zephyrproject/zephyr/modules/cmsis/. -I<censored>/zephyrproject/modules/hal/nordic/nrfx -I<censored>/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I<censored>/zephyrproject/modules/hal/nordic/nrfx/mdk -I<censored>/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -I<censored>/zephyrproject/modules/crypto/tinycrypt/lib/include -isystem <censored>/zephyrproject/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros <censored>/zephyrproject/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/<censored>/.local/opt/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi -imacros <censored>/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=<censored>/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=<censored>/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=<censored>/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -MD -MT zephyr/subsys/random/CMakeFiles/subsys__random.dir/random_entropy_device.c.obj -MF zephyr/subsys/random/CMakeFiles/subsys__random.dir/random_entropy_device.c.obj.d -o zephyr/subsys/random/CMakeFiles/subsys__random.dir/random_entropy_device.c.obj -c <censored>/zephyrproject/zephyr/subsys/random/random_entropy_device.c
                 from <censored>/zephyrproject/zephyr/include/zephyr/toolchain.h:50,
                 from <censored>/zephyrproject/zephyr/include/zephyr/sys/atomic.h:13,
                 from <censored>/zephyrproject/zephyr/subsys/random/random_entropy_device.c:7:
[... 57 lines cut ...]
   96 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
<censored>/zephyrproject/zephyr/include/zephyr/device.h:238:37: note: in expansion of macro 'DEVICE_NAME_GET'
  238 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
<censored>/zephyrproject/zephyr/include/zephyr/device.h:255:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  255 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
<censored>/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:65:49: note: in expansion of macro 'DEVICE_DT_GET'
   65 | static const struct device *const dev_entropy = DEVICE_DT_GET(DT_NODELABEL(rng));
      |                                                 ^~~~~~~~~~~~~
[116/189] Building C object zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build <censored>/zephyrproject/build
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.167.8
 - Commit: d4b7bf986c6a5ea0eedaf2d2b5e2a22567b1dac9

**Additional context**

- The error comes from this line of code: https://github.com/zephyrproject-rtos/zephyr/blob/d4b7bf986c6a5ea0eedaf2d2b5e2a22567b1dac9/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c#L65

cvinayak, 2024-11-06:
There is no supported entropy driver for nRF54L15 yet in the upstream Zephyr Project.
When there is one, it will need to support its APIs being invoked from thread and ISR execution contexts, and change the code in link layer to use a DTS choose entropy device.

nordic-piks, 2024-11-07:
Triaged internally

rettichschnidi, 2024-11-10:
Just realized: `NETWORKING` selects `ENTROPY_GENERATOR`. So - due to this bug, a nRF54 based application can not support both, BT an IP communication. This limitation affects the product I am working on directly, pretty much rules out the usage of the nRF54L15. The prototypes using a nRF52840 where still fine.

rettichschnidi, 2024-11-10:
Adding `CONFIG_ENTROPY_BT_HCI=n` allows the build to succeed. I do no know the consequences of this however.

This builds:
```
west build --pristine --board nrf54l15dk/nrf54l15/cpuapp zephyr/samples/hello_world/ -- -DCONFIG_BT=y -DCONFIG_NETWORKING=y -DCONFIG_ENTROPY_BT_HCI=n
```

jukkar, 2024-11-11:
For testing purposes with networking enabled, you could set `CONFIG_TEST_RANDOM_GENERATOR=y` to get things compiled. This is not a permanent solution as the generated random numbers are not truly random.

cvinayak, 2024-11-11:
> For testing purposes with networking enabled, you could set `CONFIG_TEST_RANDOM_GENERATOR=y` to get things compiled. This is not a permanent solution as the generated random numbers are not truly random.

@jukkar hmm... i was not aware of this, can I use this in nRF54L Controller on target? (yes, of course it is test generator) it does provide the Zephyr Entropy APIs and dts node to be chosen, right?

jukkar, 2024-11-11:
I have used this Kconfig option to test networking things when random number generator is not available. Not sure if it would work for you in your use case.

rettichschnidi, 2025-04-11:
Thanks to the awesome effort by Nordic, this issue has been resolved as part of https://github.com/zephyrproject-rtos/zephyr/pull/83972

Thanks a lot!

closed 2025-04-11 by rettichschnidi