zephyrproject-rtos/zephyr #81008: HTTP Server dynamic IPs

RIPHeisenberg, 2024-11-06:
Hi everyone,

I am currently trying out the Zephyr 3.7 http_server example. It does work fine. I haven't however found out how to assign a dynamic IP address to the Servers properties.

The HTTP_SERVICE_DEFINE() Macro only allows for a dynamic port. As described in the Doc strings.

 * @param _host IP address or hostname associated with the service.
 * @param[inout] _port Pointer to port associated with the service.

Would it be possible to change the _host IP address property to a pointer?
One could use the http_server_start() and _stop() functions to rerun the socket initialization.

The objective of my current project is to use the Zephyr dhcp api to generate a fitting _host IP and then assign it at runtime.
I could write the socket management myself but I really like your HTTP_Server implementation.

Thanks for your interest

jukkar, 2024-11-06:
You can either leave host as empty `""`, or as NULL. This was fixed in commit 3949873886a7831f5ba7a1bc02120813dafa35d2

RIPHeisenberg, 2024-11-19:
Hi Jukkar, thanks for the responds. Seems as if everything is running smooth now in Zephyr 4.0.0.

jukkar, 2025-02-12:
Closing as this is fixed in 4.0. No plans for changes in v3.7