zephyrproject-rtos/zephyr #81014: drivers: sensors: rtio No error reporting from bus controller to application

FlorianWeber1018, 2024-11-06:
**Is your enhancement proposal related to a problem? Please describe.**

**Describe the solution you'd like**

**Additional context**

teburd, 2024-11-06:
@bperseghetti @jgoppert this likely relates to the issue with rtio and cognipilot as well

FlorianWeber1018, 2025-03-17:
How to proceed here? 
Should i make a draft PR for discussion? @teburd @ubieda Any thoughts on that?  (redirecting cpe´s <-> flag for callbacks so that  callbacks will be called in case of an error or success <-> another solution i don´t know yet)

ubieda, 2025-03-17:
@FlorianWeber1018 I'm good with having the callback handling the error. Feel free to file a draft PR to move forward this feature. Thank you!

teburd, 2025-03-17:
The callback handling the error will require some small changes to rtio_executor.c and likely a review of existing callback users in the tree, otherwise I think its the right approach as well.

Pulls very welcome!

teburd, 2025-05-29:
@FlorianWeber1018 were you still looking to tackle this or should I take it?

FlorianWeber1018, 2025-06-03:
i am still looking to tackle it but i am full of work. I will try tackle it this week.

PR #93227 (2025-09-16): rtio: More useful callback OPs
Callbacks now take a result parameter which may, if the callback was linked to by a previous submissions, have the result code from the first submission that failed.

Fixes #81014
closed 2025-09-16 by henrikbrixandersen