zephyrproject-rtos/zephyr #81021: net: IP fragementation failures are not handled properly

krish2718, 2024-11-06:
**Describe the bug**

As part of 802.11 AMSDU testing, we do ping tests with larger payloads, and we enable `CONFIG_NET_IPV4_FRAGMENT=y` and seeing issues in the nRF70 driver. Below are the issues found in IP fragmentation code:

1. In case of any failure in IP fragmentation (v4/v6) due to out of memory `ENOMEM` or out of buffers `ENOBUFS` the packet is sent unfragmented to the L2 which violates MTU rules.
2. In case intermediate fragments fail, the code attempts to send whole unfragmented packet even after sending the initial few fragments.

See code https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/ipv4_fragment.c#L636 and https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/ipv6_nbr.c#L822.

**To Reproduce**

1. on NRF7002DK build `samples/net/wifi/shell` with `CONFIG_NET_IPV4_FRAGMENT=y` and `CONFIG_NET_IPV4_FRAGMENT_MAX_PKT=24`.
2. Connect to an AP and ping `net ping <AP> -s 4000`
3. Observe warning from nRF70 driver:
```
[00:03:33.439,270] <err> net_pkt: Data buffer (1520) allocation failed.
[00:03:33.439,453] <err> wifi_nrf: nrf_wifi_hal_buf_map_tx: Invalid TX buf_len (3042) for (4)
[00:03:33.439,483] <err> wifi_nrf: tx_cmd_prep_callbk_fn: nrf_wifi_hal_buf_map_tx failed
[00:03:33.439,514] <err> wifi_nrf: tx_cmd_prepare: build_mac80211_hdr failed
[00:03:33.439,514] <err> wifi_nrf: tx_cmd_init: tx_cmd_prepare failed
[00:03:34.462,280] <err> net_pkt: Data buffer (1520) allocation failed.
```

**Expected behavior**
Ping should be successful even for larger values. In case of OOM failures, the stack should drop the entire chain of fragments and free the original packet. Need to see if RFC specifies a way to inform the receiver about failure to send the fragments to avoid waiting till reassembly timeout occurs. (`15ses`).

Adding support for retying sending of fragments later in case of memory failure might be complicated.

**Impact**
802.11 A-MSDU feature might not work.

**Logs and console output**
NA
**Environment (please complete the following information):**

 - OS:Linux
 - Toolchain Zephyr SDK,
 - Commit SHA or Version used: 4.0.0-rc1

**Additional context**
This frag code was supposedly taken from Intel's original implementation and needs a thorough review and tests.

jukkar, 2024-11-07:
It is easy just to drop the packet if there was an allocation failure when creating the individual fragments. The system has already sent the fragments that were allocated before the allocation failure. If this is not desired, then this increases memory requirements considerably because we would need to keep the fragments in memory and only send them all after a successful fragmentation.

jukkar, 2024-11-07:
cc: @rlubos

rlubos, 2024-11-07:
Retransmission is the responsibility of the upper layers so I agree that dropping the packet seems to be the most straightforward way to handle this.

krish2718, 2024-11-07:
Agreed, dropping is the easiest, and this feature is seldom used in IoT, so, not worth the effort to implement retry or other mechanisms.

PR #81061 (2024-11-08): Drop IP packet if there is a fragmentation failure
If we could not fragment the IPv4 or IPv6 packet, then drop it and do not try to send it. Let the upper layer re-send the packet if needed. It is causing more trouble if we try to send the packet and not honor the MTU setting.

Fixes #81021
commit 41431637 (2024-11-08): net: ipv4: fragmentation: Drop the packet if fragmentation fails

If we could not fragment the IPv4 packet, then drop it and do
not try to send it. Let the upper layer re-send the packet if
needed. It is causing more trouble if we try to send the packet
and not honor the MTU setting.

Fixes #81021
closed 2024-11-08 by mmahadevan108