zephyrproject-rtos/zephyr #81023: Bluetooth: Controller: ISO: BIG Complete event contains invalid/unset values

Thalley, 2024-11-06:
**Describe the bug**
When the controller finished setting up the BIG, it sends a BIG Complete Event which is parsed by the host, but the event is missing some values / some values are uninitialized. 

**To Reproduce**
1) Build and run zephyr/tests/bsim/bluetooth/host/iso/bis/tests_scripts/bis.sh with https://github.com/zephyrproject-rtos/zephyr/pull/80999
2) See error 

**Expected behavior**
Expect that all values in the event are valid

**Impact**
Not insignificant as the values are expose to the application layer

**Logs and console output**
```
d_00: @00:00:00.013751  [00:00:00.013,732] <inf> bis_broadcaster: ISO Channel 0x81a689c connected
d_00: @00:00:00.013947 ERROR: (WEST_TOPDIR/zephyr/tests/bsim/bluetooth/host/iso/bis/src/bis_broadcaster.c:229): Invalid ISO interval 0x0000
d_00: @00:00:00.013947  The TESTCASE FAILED (test return code 2)
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/80999

**Additional context**
N/A

Thalley, 2024-11-06:
@cvinayak this also seems to be an issue for unicast: https://github.com/zephyrproject-rtos/zephyr/pull/81029 where both c to p and p to c latencies are 0.

cvinayak, 2025-02-12:
@Thalley can you confirm if this issue can now be closed, fixed by #85300 and #85358 ?

Thalley, 2025-02-13:
> [@Thalley](https://github.com/Thalley) can you confirm if this issue can now be closed, fixed by [#85300](https://github.com/zephyrproject-rtos/zephyr/pull/85300) and [#85358](https://github.com/zephyrproject-rtos/zephyr/pull/85358) ?

My PR, https://github.com/zephyrproject-rtos/zephyr/pull/80999, is now passing CI so looks like the BIG complete event now contains all valid values :)

closed 2025-02-13 by Thalley