zephyrproject-rtos/zephyr #81025: samples: wifi: board_frdm_rw612 fails to connect to WiFi APs, Entropy failed to Init

andrewnyland, 2024-11-06:
**Describe the bug**

Demonstrated on NXP's FRDM_RW612 development board, with the `net/wifi` sample. Attempted to connect to a few different wifi networks with various security levels. Also attempted to use alternate drivers with no success or apparent change. Wifi `scan` works fine.

**To Reproduce**

1. `cd ~/zephyrproject/zephyr/samples/net/wifi`
2. `west update` as needed
3. `west build -p -b frdm_rw612`
4. `west flash`
5. Open serial port to view device logs

1/2 approaches to see issue, after above 5 steps:
1. Open shell
2. Run `wifi scan`, expect to see list of nearby WiFi networks with SSID/sec type/etc - this works
3. Run `wifi connect -s "SSID" -p "passphrase" -k #`
4. Expect [LOG 1](#md-log-1)
6. Expect delay until network event
7. Expect [LOG 2](#md-log-2)

2/2 approaches to see issue, after above 5 steps:
1. Reset board to show boot prompt
2. Expect [LOG 3](#md-log-3)

**Expected behavior**

After running `wifi connect <params>`, it is expected to connect to the AP and ideally be given an IP address via DHCP (dependent on network setup obviously but this is expected for my given network and comparable boards supported by Zephyr). 

**Impact**

Moderately inconvenient for board usage and investigation, as WiFi is one of its main provided and advertised abilities. Ethernet can be used instead.

Completely blocking for running wifi benchmarks and zperf performance tests - which was the purpose of using this sample today.

**Logs and console output**

<a id="md-log-1">LOG 1:</a> `wifi connect` logs 1
```
assertion failed at WEST_TOPDIR/modules/lib/hostap/port/mbedtls/supp_psa_api.c:438 - actual:-137 expected:0
Connection requested
```

<a id="md-log-2">LOG 2:</a> `wifi connect` logs 2
```
[00:26:56.800,204] <inf> wpa_supp: ml: SME: Trying to authenticate with 3c:37:86:66:c6:c3 (SSID='SSID_name' freq=5765 MHz)
[00:26:56.817,743] <inf> wpa_supp: ml: Trying to associate with 3c:37:86:66:c6:c3 (SSID='SSID_name' freq=5765 MHz)
uart:\~$ PKG_TYPE: BGA
Set BGA tx power table data 
[00:26:58.349,650] <inf> wpa_supp: ml: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US 
[00:26:58.362,187] <inf> wpa_supp: ml: CTRL-EVENT-ASSOC-REJECT bssid=3c:37:86:66:c6:c3 status_code=1 qca_driver_reason=assoc_no_resp_received
[00:27:07.209,744] <inf> wpa_supp: ml: SME: Trying to authenticate with 3c:37:86:66:c6:c3 (SSID='SSID_name' freq=5765 MHz)
[00:27:07.226,763] <inf> wpa_supp: ml: Trying to associate with 3c:37:86:66:c6:c3 (SSID='SSID_name' freq=5765 MHz)
uart:\~$ PKG_TYPE: BGA
Set BGA tx power table data 
[00:27:07.260,712] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: received event 2
[00:27:07.271,654] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: associated to nxp_wlan_network
[00:27:07.283,881] <inf> wpa_supp: ml: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US 
[00:27:07.296,623] <inf> wpa_supp: ml: Associated with 3c:37:86:66:c6:c3
[00:27:07.306,218] <inf> wpa_supp: ml: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
uart:\~$         assertion failed at WEST_TOPDIR/modules/lib/hostap/port/mbedtls/supp_psa_api.c:390 - actual:-137 expected:0
[00:27:07.330,026] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: received event 10
[00:27:07.340,805] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: deauth leaving
[00:27:07.351,540] <inf> wpa_supp: ml: CTRL-EVENT-DISCONNECTED bssid=3c:37:86:66:c6:c3 reason=1 locally_generated=1
[00:27:07.364,722] <inf> wpa_supp: ml: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
[00:27:07.376,604] <inf> wpa_supp: ml: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="SSID_name" auth_failures=1 duration=10 reason=WRONG_KEY
[00:27:07.391,675] <inf> wpa_supp: ml: CTRL-EVENT-DSCP-POLICY clear_all
Connected
Connection request failed (2)
```

<a id="md-log-3">LOG 3:</a> Bootup logs
```
uart:\~$ 
Wi-Fi cau temperature : 26
uart:\~$ MAC Address: C0:95:DA:01:60:2D 
PKG_TYPE: BGA
Set BGA tx power table data 
[00:00:01.220,419] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: received event 14
uart:\~$ *** Booting Zephyr OS build v4.0.0-rc1-76-g743761d7d1d7 ***
[00:00:01.237,361] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN initialized
uart:\~$ supp_psa_crypto_init failed [00:00:01.251,870] <inf> wifi_supplicant: wpa_supplicant initialized
[00:00:01.265,358] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: received event 16
[00:00:01.276,122] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: PS_ENTER
[00:00:01.286,156] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: received event 16
[00:00:01.296,955] <dbg> nxp_wifi: nxp_wifi_wlan_event_callback: WLAN: PS_ENTER
[00:00:01.308,682] <err> bwpa_supp: Init of random number generator failed
```

**Environment (please complete the following information):**

 - OS: macOS M3 Pro
 - Toolchain: Zephyr SDK 0.16.5, version 4.0.0-rc1, CMake 3.30.0
 - Commit SHA or Version used: 743761d7d1d7be7ef7cf0614e37ead5eb748b79e

**Additional context**

MaochenWang1, 2024-11-08:
Hi @andrewnyland 
pls see this PR: https://github.com/zephyrproject-rtos/hal_nxp/pull/463
which can fix this issue.

dleach02, 2024-11-08:
@andrewnyland could you verify that #81115 addresses the issue you found? If it does, give that PR a +1 for us.

closed 2024-11-08 by mmahadevan108

andrewnyland, 2024-11-08:
@dleach02 Yeah, just seeing this now, looks like it was already accepted but I'll go ahead and test it on my end in a few minutes.