zephyrproject-rtos/zephyr #81036: eSPI Virtual Wire "Valid bit" is observed only on updated Virtual Wire

vphirema, 2024-11-07:
**Is your feature request related to a problem? Please describe.**
On Pantherlake Intel SoC, "valid" bit of SLP_Sx is set only for the Virtual Wire that undergoes changes. However, that wasn’t the case in Previous generation of Intel SoC.

On Meteorlake: when S5 is received ->S4 & S3 are valid, when S4 is received both ->S5 & S3 are still valid.
On Pantherlake : when S5 is received ->S4 & S3 are Not Valid, when S4 is received-> S5 is made Invalid & S3 is still invalid.

**Describe the solution you'd like**

For the backward compatibility, add a new Kconfig and ignore the "Valid bit" masking if Konfig is enabled.

Novoton:
https://github.com/zephyrproject-rtos/zephyr/blob/4300a4c33b0f6becc43d1e3a130ebee19c411c6b/drivers/espi/espi_npcx.c#L875
https://github.com/zephyrproject-rtos/zephyr/blob/4300a4c33b0f6becc43d1e3a130ebee19c411c6b/drivers/espi/espi_npcx.c#L891

ITE:
https://github.com/zephyrproject-rtos/zephyr/blob/4300a4c33b0f6becc43d1e3a130ebee19c411c6b/drivers/espi/espi_it8xxx2.c#L733

MCHP:
Not applicable as valid bit is not checked

**Describe alternatives you've considered**
From the discussion with eSPI experts "New implementations seem spec compliant. If Valid bit is not set, the previous value should be considered still valid".

**Additional context**
example eSPI capture
[image]

Table 9: Virtual Wire Index Definition
https://downloadmirror.intel.com/27055/327432%20espi_base_specification%20R1-5.pdf

vphirema, 2024-11-07:
cc: @keith-zephyr , @galak ,  @nashif , @Dino-Li , @MulinChao

vphirema, 2024-11-07:
PR created 
https://github.com/zephyrproject-rtos/zephyr/pull/81096

albertofloyd, 2025-02-20:
@vphirema  why is this reopened?

nashif, 2025-02-20:
the PR should have closed this...

closed 2025-02-20 by nashif