zephyrproject-rtos/zephyr #81044: Zephyr firmware broken when using full RAM, atleast on nRF SoCs

cvinayak, 2024-11-07:
**Describe the bug**
Non-functional samples in Zephyr when samples or application use all of the RAM, atleast in nRF SoCs. Hey, I want to use every bit of my precious SoC!

A sample (`hello_world`) worked correctly if there was ~37 bytes free on nRF52840, and similar free bytes on nRF53 network core. Tested on these two to fail. Have not tested if the failure is on all vendor SoCs supported by Zephyr.

Is there any hidden issues if this a symptom in Zephyr kernel/build, that can cause faulty applications that are in productions, in field and/or with customers.

Discovered as part of #80852.

Use the below snippet in `hello_world` sample, reduce the value `37` anything lower down to `0` to reproduce a non-functional `hello_world` sample.

The number `5696` is the RAM usage of the `hello_world` sample without the use of `ram_fill` variable.

Ozone debugger disconnects during calls to `z_sys_init_run_level(INIT_LEVEL_PRE_KERNEL_2);` i think, reporting Bus Faults.
 
```
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index c550ab461cb..632383f2fdc 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -6,6 +6,8 @@
 
 #include <stdio.h>
 
+static volatile uint8_t ram_fill[256 * 1024 - 5696 - 37] __attribute__((retain));
+
 int main(void)
 {
        printf("Hello World! %s\n", CONFIG_BOARD_TARGET);

```
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir -p build/hello_world; cd build/hello_world
2. cmake -GNinja -DBOARD=nrf52840dk/nrf52840 ../../samples/hello_world
3. ninja flash
4. See, there is no output on console

**Expected behavior**
Sample to functional as expect when using full RAM available on nRF SoCs.

**Impact**
Annoyance, run-time application failure on near RAM usages. No build failure of imminent issue.

**Logs and console output**
None

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.8
 - Commit SHA or Version used: c3466b14d09

**Additional context**
NA

cvinayak, 2024-11-07:
Maintainers or release managers are free to remove the release blocker label, if this is an acceptable release notes "Known Issue" candidate.

nordicjm, 2024-11-07:
The reason for this is that you have a heap enabled with the default value of **use all available RAM**, it seems that this will fail if there is actually no RAM left (arguably that is expected). `CONFIG_COMMON_LIBC_MALLOC=y` enables malloc (enabled by default), and `CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=-1` uses all remaining RAM for it. If you disable the heap or set the size to 0 then it successfully boots.

nordic-piks, 2024-11-07:
Triaged internally

mmahadevan108, 2024-11-07:
cc @andyross @peter-mitsis , can you help take a look at this issue reported.

@cvinayak is this a regression?

cvinayak, 2024-11-07:
@mmahadevan108 The issue might always have been, I have not tested if older release has same issue.

mmahadevan108, 2024-11-08:
@cvinayak , I am removing the release blocker label from this PR as from discussion with the release team this does not seem like a release blocker.

Thalley, 2024-11-27:
> @cvinayak can this be closed now?

I don't think so. The issue here is that we run out of RAM even when the reported RAM allocation is less than 100%. That means that we cannot find RAM issues at build time, and thus don't catch various issues in CI. This will also be an issue for users expecting to be able to use RAM up to 100%, where in reality they may only be able to use 99.5% or so.

cvinayak, 2024-11-29:
As @Thalley has explained, the build should fail if 100% of the RAM cannot be utilized at runtime.

fabiobaltieri, 2024-12-10:
hey @cvinayak, I'll lower this to medium since it does not look like there's a plan to fix this for this release, feel free to readjust if you think that's not the case

closed 2025-02-23 by github-actions[bot]