zephyrproject-rtos/zephyr #81050: tests: benchmarks: sched: qemu_riscv assertion fail

ycsin, 2024-11-07:
**Describe the bug**
qemu_riscv hits assertion when running the `tests/benchmarks/sched`. The benchmark works if #80027 is reverted.

**To Reproduce**
Steps to reproduce the behavior:
1. Check out current `main`
2. `west build -b qemu_riscv64 -p auto -t run zephyr/tests/benchmarks/sched `
3. See error

**Expected behavior**
The benchmark should run without asserting

**Impact**
Can't run the `sched` benchmark

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-rc2-115-gc3466b14d09b ***
Running 0x8000c0a8
ASSERTION FAIL [arch_irq_unlocked(key) || _kernel.cpus[0].current->base.thread_state & (((1UL << (0))) | ((1UL << (3))))] @ WEST_TOPDIR/zephyr/kernel/include/kswap.h:99
        Context switching while holding lock!
E: 
E:  mcause: 11, Environment call from M-mode
E:   mtval: 0
E:      a0: 0000000000000004    t0: 0000000000000000
E:      a1: 0000000000000063    t1: 000000000000004c
E:      a2: 000000008000f628    t2: 0000000000000044
E:      a3: 000000008000c3b0    t3: 000000000000002a
E:      a4: 0000000000000000    t4: 000000000000002e
E:      a5: 000000008000c528    t5: 000000000000007f
E:      a6: 0000000000000009    t6: 0000000000000010
E:      a7: 0000000000000009
E:      sp: 000000008000f660
E:      ra: 0000000080002966
E:    mepc: 0000000080000926
E: mstatus: 0000000a00021800
E: 
E:      s0: 000000008000c048    s6: 000000008000a488
E:      s1: 000000008000c528    s7: 000000008000c098
E:      s2: 0000000000000000    s8: 000000000000000a
E:      s3: 000000008000c3b0    s9: 0000000000000000
E:      s4: 000000008000c0a8   s10: 0000000a00020088
E:      s5: 00000000000003f2   s11: 0000000000000000
E: 
E: call trace:
E:       0: sp: 000000008000f660 ra: 0000000080000926 [assert_post_action+0x8]
E:       1: sp: 000000008000f690 ra: 00000000800005d4 [main+0x14e]
E:       2: sp: 000000008000f6e0 ra: 0000000080003264 [z_thread_timeout+0x0]
E:       3: sp: 000000008000f700 ra: 0000000080002250 [bg_thread_main+0x124]
E:       4: sp: 000000008000f718 ra: 000000008000212c [bg_thread_main+0x0]
E:       5: sp: 000000008000f730 ra: 000000008000212c [bg_thread_main+0x0]
E:       6: sp: 000000008000f740 ra: 0000000080000916 [z_thread_entry+0x22]
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x8000c3b0 (main)
E: Halting system
```

**Environment (please complete the following information):**

 - OS: MacOS
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: v4.0.0-rc2-115-gc3466b14d09b

peter-mitsis, 2024-11-07:
I am able to reproduce the problem. I am also able to further narrow down the offending commit to SHA ID cedd36106b3fdb2159ba1e03e05daeee1a31389a . Continuing to dig.

peter-mitsis, 2024-11-07:
In the referenced commit, interrupts are locked for too long. They should be unlocked after the call to z_unpend_first_thread() in the sched benchmark.

How the error crept in. I added the arch_irq_lock()/arch_irq_unlock() pair as it was needed for the optimization done for z_unpend_first_thread(). I was thinking to leave the interrupts locked for the whole of the various timings so that unlocking them would not interfere with any of the timings. However, I failed to take into account the CONFIG_USE_SWITCH configuration option. That one requires that interrupts not be locked when we call k_yield().

I think CI did not catch it because we "lucked out" on the selection of platforms.

A fix will be coming shortly.

PR #81106 (2024-11-08): tests: Fix sched benchmark exception on qemu_riscv64
Corrects an issue that was introduced when the interrupt locking/unlocking was added to the 'sched' benchmark by unlocking the interrupts before the context switch done by k_yield(), but after the call to z_unpend_first_thread().

Fixes #81050