zephyrproject-rtos/zephyr #81100: GPIO Port Design Limitation of only 32 pins

nslowell, 2024-11-07:
The GPIO Port API design assumes/forces/limits the max number of pins per port to be 32.  The problem is when you've got chips that have pins groups into sizes greater than 32.  They share the address space, they share the same interrupt.

We're typically used to Register mappings like this
PORTA-DIR
PORTA-DAT
PORTA-IEN
PORTA-ISTAT
PORTB-DIR
PORTB-DAT
PORTB-IEN
PORTB-ISTAT
where PORT A has up to 32 pins that use interrupt N and PORT B has up to 32 pins that use interrupt N+1.

What about chips where there's just one port:
PORT-DIRA
PORT-DIRB
PORT-DATA
PORT-DATB
PORT-IENA
PORT-IENB
PORT-ISTATA
PORT-ISTATB
and all pins go through interrupt N.

I realize Zephyr isn't Linux, but there continues to be a lot of borrowing.  Linux's GPIO API easily handles 32+ gpio ports.  The running theme is there are individual pin functions and then there are array-based functions, not bitmask-based functions.

I think/hope shared interrupts will allow for breaking up the port into multiple ports, but the addressing and coding complexity remains.

Has anyone thought about this?  Any plans to improve GPIO API to address this shortcoming?

Ayush1325, 2025-05-13:
Hello, is anyone already working on this?

nslowell, 2025-05-23:
Here's what I ended up developing locally, separate from gpio.h but inheriting what it could, if someone wants to run with it. 

```
#ifndef DRIVERS_GPIOP_H_
#define DRIVERS_GPIOP_H_

#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/drivers/gpio.h>

#ifdef __cplusplus
extern "C" {
#endif

struct gpiop_callback;

/**
 * @brief GPIO Pin callback function type.
 *
 * This is the function signature for GPIO Pin callback functions.
 *
 * @param dev Pointer to the device structure for the driver instance.
 * @param cb Pointer to the gpiop_callback structure.
 */
typedef void (*gpiop_callback_t)(const struct device *dev, struct gpiop_callback *cb);

/**
 * @brief GPIO Pin callback structure
 *
 * Used to register a callback in the driver instance callback list.
 * As many callbacks as needed can be added as long as each of them
 * are unique pointers of struct gpiop_callback.
 * Beware such structure should not be allocated on stack.
 *
 * Note: To help setting it, see gpiop_init_callback() below
 */
struct gpiop_callback {
	/** This is meant to be used in the driver and the user should not
	 * mess with it (see drivers/gpio/gpio_utils.h)
	 */
	sys_snode_t node;

	/** Actual callback function being called when relevant. */
	gpiop_callback_t handler;

    /** User data */
    void *user_data;
};

/**
 * @brief GPIO Pin driver API structure.
 *
 * This structure holds the function pointers for the GPIO Pin driver API.
 */
struct gpiop_driver_api {
    /**
     * @brief Configure a GPIO pin.
     *
     * @param dev Pointer to the device structure for the driver instance.
     * @param pin The pin number to configure.
     * @param flags Configuration flags for the pin.
     * @return 0 on success, negative error code on failure.
     */
    int (*config)(const struct device *dev, gpio_pin_t pin, gpio_flags_t flags);

    /**
     * @brief Set the value of a GPIO pin.
     *
     * @param dev Pointer to the device structure for the driver instance.
     * @param pin The pin number to set.
[... 187 lines cut ...]
 *
 * @param dev Pointer to the device structure for the driver instance.
 * @param pin The pin number to configure interrupts for.
 * @param flags Configuration flags for the pin.
 * @return 0 on success, negative error code on failure.
 */
static inline int gpiop_configure_interrupt(const struct device *dev, gpio_pin_t pin,
                       gpio_flags_t flags)
{
    const struct gpiop_driver_api *api = (const struct gpiop_driver_api *)dev->api;

    enum gpio_int_trig trig =
        (enum gpio_int_trig)(flags & (GPIO_INT_LOW_0 | GPIO_INT_HIGH_1));
    enum gpio_int_mode mode =
        (enum gpio_int_mode)(flags & (GPIO_INT_EDGE | GPIO_INT_DISABLE | GPIO_INT_ENABLE));

    return api->configure_interrupt(dev, pin, mode, trig);
}

#ifdef __cplusplus
}
#endif

#endif /* DRIVERS_GPIOP_H_ */
```