zephyrproject-rtos/zephyr #81119: stm32: Extend wake-up pins support to whole family

erwango, 2024-11-08:
**Is your enhancement proposal related to a problem? Please describe.**
STM32 Wake up pins support was introduced by https://github.com/zephyrproject-rtos/zephyr/pull/73058/.
This port was limited to L4 U5 and WL series, but other series remain unvovered

**Describe the solution you'd like**
Extend STM32 Wake up pin feature coverage to the whole STM32 series using example provided in #73058.

erwango, 2024-11-08:
> L4 too

Sure, it was mentioned ;)

> Because other series didn't support PowerOff in Zephyr,

Ha, right. Thanks for the update. So it would need to be added as well then.

mathieuchopstm, 2025-05-27:
Current status from a quick check:
- [x] STM32C0
- [ ] STM32F0
- [x] STM32F1
- [ ] STM32F2
- [ ] STM32F3
- [x] STM32F4
- [ ] STM32F7
- [x] STM32G0
- [ ] STM32G4
- [ ] STM32H5
- [ ] STM32H7
- [ ] STM32H7R/S
- [ ] STM32L0
- [ ] STM32L1
- [x] STM32L4
- [ ] STM32L5
- [ ] STM32MP1x
- [ ] STM32MP13
- [ ] STM32N6
- [ ] STM32U0
- [ ] STM32U3 (not merged in `main` yet)
- [x] STM32U5
- [x] STM32WB
- [ ] STM32WB0
- [ ] STM32WBA
- [x] STM32WL

ivanwagner, 2025-08-21:
@mathieuchopstm you can tick the STM32WBA family solved via #93911

mathieuchopstm, 2026-08-25:
H5 implementation (merged): https://github.com/zephyrproject-rtos/zephyr/pull/116628

C5 implementation (ongoing): https://github.com/zephyrproject-rtos/zephyr/pull/116633

U0 implementation (ongoing): https://github.com/zephyrproject-rtos/zephyr/pull/116499

U3 implementation (ongoing): https://github.com/zephyrproject-rtos/zephyr/pull/113038