zephyrproject-rtos/zephyr #81125: Drivers: SPI: `nxp,imx-lpspi` DMA usage incompatible with most SPI drivers

matt-wood-ct, 2024-11-08:
**Is your enhancement proposal related to a problem? Please describe.**
Yes, when using LPSPI with Zephyr SPI device drivers you cannot enable DMA as it does not manage coherency internally, the driver has the expectation the usage will manage cache coherency so in practice this means that virtually all mainstream SPI device drivers can/will misbehave in unusual and unpredictable manners.  

I noticed this on a project which utilised and IMXRT1040 + a NRF52 series IC connected using `zephyr,bt-hci-spi` which is a very high traffic driver so thus is best implemented using DMA.  
I also ran into the issues when using the `worldsemi,ws2812-spi` and `jedec,spi-nor` drivers.  
Those are just the in-tree drivers I use in this project, but I would suspect this issue applies universally given the cause.

**Describe the solution you'd like**
I would like the `nxp,imx-lpspi` driver to implement a cache coherency scheme similar to what the existing STM32H series driver does which also had DCache issue where by if caching is enabled and the memory provided to the transact function is cacheable then it uses a small internal buffer for transactions to enforce coherency in all cases while enabling applications to provide large nocache buffers if their application requires it  (see 0e72d63a0177744a49b6a0be71e1c9c012fa121c and #71048)

**Describe alternatives you've considered**
It would be possible to do the opposite, say that all SPI device drivers need to account for coherency and make their buffers no cache, but this will likely cause significantly more work & problems.

**Additional context**
For my active project I have taking the `nxp,imx-lpspi` out of tree and added a bodge where in the transact function I flush the buffer locations then in the DMA callback I invalidate the cache over the RX buffer, this functionally works but is technically not thread safe (when the buffers are allocated from stack or 32byte aligned & rounded up it is safe).

matt-wood-ct, 2024-11-08:
To clarify what I mean when I say "unusual and unpredictable manners". Generally this manifested as random junk data coming out of the SPI and the driver also receiving back bad data which does not match bus captures, but all of which inconsistent build to build as it depends on what uses adjacent memory areas. Many of instances showed 0xAA coming out of the SPI, which is not surprising as the unused memory is stuffed with 0xAA as part of the high water mark and memory protection schemes.

Raymond0225, 2024-11-08:
I believe that this DMA coherency issue exists on all DMA related peripheral devices such like uart, i2c i2s spi, it is better we come up with a total solution for it.

matt-wood-ct, 2025-07-08:
> What solution are you looking for here?

The LPSPI driver should provide coherency for the data used in transactions at the point they hit the DMA, ST already did this on their driver which has the same issue ref https://github.com/zephyrproject-rtos/zephyr/pull/71048  
this is the precedent, other vendors should follow it unless the TSC provide a better way to manage it at a higher level, otherwise every single SPI bus device driver will need to adopt coherency management specially to tolerate these platforms.  

These issues only apply to platforms with DCache like the IMXRT series, so supporting peripherals attached to an one will need to come with some baggage otherwise it simply does not work properly.  

I am sure these same issues apply to the UART and I2S peripherals, and IMO they should also be fixed, the SPI specifically is especially troubling because it is frequently used for external memory so high bandwidth usage is extremely common, in these applications DMA is virtually mandatory.  
In our case we are also using HCI SPI for bluetooth so these issues are severe, we still have to maintain a bodge out of tree driver to make the SPI work at all but I am certain the side effects of the poor fix will be creating bugs throughout.

teburd, 2025-07-09:
> > What solution are you looking for here?
> 
> The LPSPI driver should provide coherency for the data used in transactions at the point they hit the DMA

@carlocaione had worked on adding annotations to memory regions in DTS to help do this more generally, but this work never seemed to get into solving this for devices and peripheral drivers broadly.

See the memory attributes PRs starting at https://github.com/zephyrproject-rtos/zephyr/pull/60049

But in the case where your dts annotated the memory regions correctly, and we had (maybe we have it?) a memory attribute API for inspecting if a address has a memory attribute (e.g. dcached region, dma accessible) then this could be done generically.

Basically a driver before using DMA should...

1. Check if the buffer is in a DMA friendly region if DMA is in use
2. Check if the buffer needs invalidation/flushing (its a cached region)

Those are two really simple queries into device tree if the attributes and regions are defined.

The only thing I really don't want to see myself and have -1'ed is a memcpy to "fix" buffers for callers that give incorrectly placed buffers. This is, imho, wrong as its a massive performance overhead hidden to the caller which should be immediately exposed instead.

closed 2026-10-15 by henrikbrixandersen

matt-wood-ct, 2026-10-15:
After nearly 2 years it is finally done.  

Thanks all🙏