zephyrproject-rtos/zephyr #81136: tests/subsys/llext/simple/llext.simple.loader_build fails to build for native_posix in main

aescolar, 2024-11-08:
**Describe the bug**
The test tests/subsys/llext/simple/llext.simple.loader_build fails to build for native_posix in main

11c350e2e635a9a47a02203432d31ba4a0e1f0ce changed a linker script snippet to set no_syscall_impl (in section llext_no_syscall_impl) hard in address 0, but in a way that sets the address of any further section/symbols after that until the current address is set to something else.

**Side notes**

* Is this llext_no_syscall_impl needed even if USERSPACE is not enabled?
* Note: native_posix is deprecated, so maybe just filtering it is an option. But still, it does not feel too nice to have sideffects in snippets.
* Just doing this (assuming the current way of doing things is necessary) would avoid the issue:
```diff
+       silly = .;
        SECTION_PROLOGUE(llext_no_syscall_impl, 0 (COPY), )
        {
          *(llext_no_syscall_impl)
        }
+       . = silly;
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=native_posix ../tests/subsys/llext/simple/ && ninja
3. ninja
4. See error

**Expected behavior**
No build failures in enabled platforms.

**Impact**
2 failures in weekly CI (for native_posix and native_posix//64)
https://github.com/zephyrproject-rtos/zephyr/runs/32564635154
(note there is some other platforms failing this test)

**Logs and console output**
```
/usr/bin/ld.bfd: zephyr/zephyr.elf: error: PHDR segment not covered by LOAD segment
collect2: error: ld returned 1 exit status
```

**Environment (please complete the following information):**

 - OS: Ubuntu 24.04
 - Toolchain (e.g Zephyr SDK, ...): gcc 14.2
 - Commit SHA or Version used: Main since 11c350e2e635a9a47a02203432d31ba4a0e1f0ce

**Additional context**
It fails to build since 11c350e2e635a9a47a02203432d31ba4a0e1f0ce

aescolar, 2024-11-08:
CC @ edersondisouza @pillo79

pillo79, 2024-11-08:
@aescolar definitely an unwanted side effect. Thanks for pinging and providing a workaround, will prepare a PR right away.

commit 50ab6953 (2024-11-08): linker: llext: avoid modifying current address in linker script snippet

The linker script snippet that places the llext_no_syscall_impl section
has the unfortunate side effect of overwriting the current linker
address. This can lead to unexpected behavior of the caller script.

Saving the current address before defining the section and restoring it
afterwards avoids this issue.

Fixes #81136.
closed 2024-11-08 by mmahadevan108