zephyrproject-rtos/zephyr #81139: samples: tfm_integration: tfm_regression: Build failure for MPS2 AN521

jayashree505, 2024-11-08:
**Describe the bug**

Unable to build the TF-M regression tests for MPS2 AN521 platform. I get configuration errors. 
**To Reproduce**

```
west build -p -b mps2_an521_ns samples/tfm_integration/tfm_regression_test/
```

**Expected behavior**

The regression build executable should be produced without any configuration errors. 

**Impact**

I am unable to check the results of regression build. 

**Logs and console output**

```
js@JSRINIV3-L01:~/workdisk/zephyr_pristine/zephyr$ west build -p -b mps2_an521_ns samples/tfm_integration/tfm_regression_test/
-- west build: making build dir /home/js/workdisk/zephyr_pristine/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/js/workdisk/zephyr_pristine/zephyr/samples/tfm_integration/tfm_regression_test
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/js/.cache/zephyr
-- Zephyr version: 3.5.0 (/home/js/workdisk/zephyr_pristine/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mps2_an521_ns
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.4 (/home/js/zephyr-sdk-0.16.4)
-- Found toolchain: zephyr 0.16.4 (/home/js/zephyr-sdk-0.16.4)
-- Found Dtc: /home/js/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/js/workdisk/zephyr_pristine/zephyr/boards/arm/mps2_an521/mps2_an521_ns.dts
-- Generated zephyr.dts: /home/js/workdisk/zephyr_pristine/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/js/workdisk/zephyr_pristine/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/js/workdisk/zephyr_pristine/zephyr/build/zephyr/dts.cmake
Parsing /home/js/workdisk/zephyr_pristine/zephyr/Kconfig
Loaded configuration '/home/js/workdisk/zephyr_pristine/zephyr/boards/arm/mps2_an521/mps2_an521_ns_defconfig'
Merged configuration '/home/js/workdisk/zephyr_pristine/zephyr/samples/tfm_integration/tfm_regression_test/prj.conf'
Configuration saved to '/home/js/workdisk/zephyr_pristine/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/js/workdisk/zephyr_pristine/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/js/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/js/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/js/workdisk/zephyr_pristine/zephyr/build
-- west build: building application
[1/131] Preparing syscall dependency handling

[2/131] Generating include/generated/version.h
-- Zephyr version: 3.5.0 (/home/js/workdisk/zephyr_pristine/zephyr), build: zephyr-v3.5.0
[8/131] Generating ../../tfm/CMakeCache.txt
FAILED: tfm/CMakeCache.txt /home/js/workdisk/zephyr_pristine/zephyr/build/tfm/CMakeCache.txt 
cd /home/js/workdisk/zephyr_pristine/zephyr/build/tfm && /usr/bin/cmake -GNinja -DTFM_TOOLCHAIN_FILE=/home/js/workdisk/zephyr_pristine/modules/tee/tf-m/trusted-firmware-m/toolchain_GNUARM.cmake -DCROSS_COMPILE=/home/js/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi -DCMAKE_BUILD_TYPE=MinSizeRel -DTFM_PLATFORM=arm/mps2/an521 -DCONFIG_TFM_BUILD_LOG_QUIET=ON -DCONFIG_TFM_MEMORY_USAGE_QUIET=OFF -DCONFIG_TFM_SPM_BACKEND="IPC" -DTEST_S=ON -DTEST_NS=ON -DBL2=TRUE -DMCUBOOT_IMAGE_VERSION_S=0.0.0+0 -DMCUBOOT_IMAGE_VERSION_NS=0.0.0+0 -DNS=TRUE -DTFM_ISOLATION_LEVEL=2 -DMCUBOOT_IMAGE_NUMBER=2 -DTFM_EXCEPTION_INFO_DUMP=OFF -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_PARTITION_NS_AGENT_MAILBOX=OFF -DTFM_PARTITION_PROTECTED_STORAGE=ON -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE=ON -DTFM_PARTITION_CRYPTO=ON -DTFM_PARTITION_PLATFORM=ON -DTFM_PARTITION_INITIAL_ATTESTATION=OFF -DTFM_PARTITION_FIRMWARE_UPDATE=OFF -DCRYPTO_RNG_MODULE_DISABLED=FALSE -DCRYPTO_KEY_MODULE_DISABLED=FALSE -DCRYPTO_AEAD_MODULE_DISABLED=FALSE -DCRYPTO_MAC_MODULE_DISABLED=FALSE -DCRYPTO_HASH_MODULE_DISABLED=FALSE -DCRYPTO_CIPHER_MODULE_DISABLED=FALSE -DCRYPTO_ASYM_SIGN_MODULE_DISABLED=FALSE -DCRYPTO_ASYM_ENCRYPT_MODULE_DISABLED=FALSE -DCRYPTO_KEY_DERIVATION_MODULE_DISABLED=FALSE -DQCBOR_PATH= -DMCUBOOT_PATH=/home/js/workdisk/zephyr_pristine/bootloader/mcuboot -DCONFIG_TFM_ENABLE_FP=OFF -DMBEDCRYPTO_PATH=/home/js/workdisk/zephyr_pristine/modules/crypto/mbedtls -DTFM_TEST_REPO_PATH=/home/js/workdisk/zephyr_pristine/modules/tee/tf-m/trusted-firmware-m/../tf-m-tests -DPSA_ARCH_TESTS_PATH=/home/js/workdisk/zephyr_pristine/modules/tee/tf-m/trusted-firmware-m/../psa-arch-tests /home/js/workdisk/zephyr_pristine/modules/tee/tf-m/trusted-firmware-m
CMake Warning at cmake/version.cmake:22 (message):
  Actual TF-M version is not available from Git repository.  Settled to
  v1.8.0
Call Stack (most recent call first):
  CMakeLists.txt:22 (include)

-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/js/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/GNUInstallDirs.cmake:239 (message):
  Unable to determine default CMAKE_INSTALL_LIBDIR directory because no
  target architecture is known.  Please enable at least one language before
  including GNUInstallDirs.
Call Stack (most recent call first):
  /home/js/workdisk/zephyr_pristine/modules/crypto/mbedtls/CMakeLists.txt:42 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Failed
CMake Error at config/check_config.cmake:20 (message):
  INVALID CONFIG: NOT PLATFORM_FLIH_IRQ_TEST_SUPPORT AND TEST_NS_FLIH_IRQ
Call Stack (most recent call first):
  /home/js/workdisk/zephyr_pristine/modules/tee/tf-m/tf-m-tests/test/config/check_config.cmake:37 (tfm_invalid_config)
  config/check_config.cmake:114 (include)
  CMakeLists.txt:82 (include)

-- Configuring incomplete, errors occurred!
See also "/home/js/workdisk/zephyr_pristine/zephyr/build/tfm/CMakeFiles/CMakeOutput.log".
See also "/home/js/workdisk/zephyr_pristine/zephyr/build/tfm/CMakeFiles/CMakeError.log".
ninja: build stopped: subcommand failed.
```
**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows) Windows (on WSL)
 - Commit SHA or Version used - a6eef0ba3755f2530c5ce93524e5ac4f5be30194

ithinuel, 2024-11-13:
@jayashree505 are you still having the issue?  
I just tried to reproduce on the current tip of `main` and on the sha you linked and both built fine.

Make sure you run west update after changing zephyr's reference.

wearyzen, 2025-01-30:
Hi @jayashree505, I too do not see any build issues on the tip. Could you please confirm if you still see this issue or if we can close the issue?

closed 2025-04-16 by github-actions[bot]