zephyrproject-rtos/zephyr #81176: twister: Test suite name parsed as a test case if not using ZTest

vChavezB, 2024-11-09:
## **Describe the bug**

For harnesses that are not ZTest, the suite name declared in the yaml file is parsed as a test case. 

Example:

```yaml
tests:
  base.my_test_suite:
    harness: "my_harness"
```

In this example, `base.my_test_suite`, is parsed as a test case. This is wrong and will make twister complain when running the unit tests. Example:

```
INFO    - 1/1 native_sim/native/64      tests/cpputest/base/base.my_test_group             PASSED (native 0.005s)
INFO    -                                    base.my_test_group                                                          NONE
INFO    -                                    base.my_test_group.my_test_group.test_1                       PASSED
```

In this log, we can see that `base.my_test_group.my_test_group.test_1` was discovered while running the test.  The first  test case (i.e., ` base.my_test_group` is the suite name I defined in the yaml.

## **To Reproduce**

I am developing my own harness for cpputest, so this should allow to test it with an example:

```
git clone https://github.com/vChavezB/zephyr -b twister_cpputest --depth=1
cd twister_cpputest
python ./scripts/twister /tests/cpputest/base
```

## **Expected behavior**

Test suite names from the `yaml` should not be considered as test cases. The relevant code is here

https://github.com/zephyrproject-rtos/zephyr/blob/c50777a8431157bd4ab3a2c2e2a37e8151ae07da/scripts/pylib/twister/twisterlib/testsuite.py#L462-L463

## **Impact**

Harnesses that are not `ztest` create always a fake test case obtained from the test suite names. This wrongly adds a new testcase which will output an error as it has Twister status NONE.

vChavezB, 2024-11-09:
I was checking other harnesses, and I see for example that Pytest clears the test cases before running:

https://github.com/zephyrproject-rtos/zephyr/blob/c50777a8431157bd4ab3a2c2e2a37e8151ae07da/scripts/pylib/twister/twisterlib/harness.py#L544-L546

In this case, we dont get the error because pytests clears any test case before running the tests.

However, for harnesses that inherit from class `Harness`, there is no way to do this. 

I could do something hacky to cleary the testcases:

```python
class MyHarness(Harness):
    def __init__(self):
        self.started = False

    def handle(self, line):
        if not self.started:
            self.instance.testcases = []
            self.started = True
```

But I think it would be better to not add a fake test case if the harnesss does not support parsing the source code.

vChavezB, 2025-01-14:
This is still a pending issue.

nashif, 2025-01-20:
@golowanow  Can you please take a look at this and #74372.

golowanow, 2025-01-21:
> Harnesses that are not ztest will always create a fake testcase derived from the test-suite names. 
> This erroneously adds a new testcase that does not exist.

currently it is Twister behavior for non-Ztest Test Suites where it creates a 'standalone' default Test Case:

https://github.com/zephyrproject-rtos/zephyr/blob/176676d81fdc44a62fe6f5ffe857d0f521ee1f24/doc/develop/test/twister.rst?plain=1#L283-L294
	

> I am developing my own harness for cpputest,  .. I see for example that Pytest clears the test cases before running:

In the new harness you should add test cases the similar way, as other harnesses, e.g. recently added #83297.

Please rebase https://github.com/zephyrproject-rtos/zephyr/pull/74372 and let's resolve your problem there if it still remains.

closed 2025-01-21 by golowanow

vChavezB, 2025-03-04:
Thanks for the clarification. I will clear the list of `self.instance.testcases`