zephyrproject-rtos/zephyr #81202: Stack canaries not working on modified FlexRAM iMX RT1050 (SRAM = DTCM)

harristomy, 2024-11-10:
**Describe the bug**

If stack canaries is enabled (CONFIG_STACK_CANARIES=y), FlexRAM has been modified with a larger DTCM, and `zephyr,sram=&dtcm`, then the FW crashes when entering the `z_arm_platform_init` function, specifically on the `ldr r3 [r3 #0]` instruction.

**Target platform:** (IMX1050) MIMXRT1052CVL5B

**To Reproduce**

1. Use any sample application (`hello_world` or `basic/blinky`)
2. In prj.conf, add `CONFIG_STACK_CANARIES=y` and `CONFIG_ENTROPY_GENERATOR=y`
3. Add a devicetree overlay where the FlexRAM is modified to have a larger DTCM and `zephyr,sram=&dtcm`
4. Try run the FW 🙏🏽 
5. Should see the FW crash on the `z_arm_platform_init` function

**Expected behavior**

HelloWorld or led blinking observed when stack canaries is enabled.

**Impact**

Can't use the protection of stack canaries in modified FlexRAM applications

**Logs and console output**

None as the crash happens in the first few instructions of the reset handler.

**Environment (please complete the following information):**

 - **OS:** Linux container on WSL2 (Ubuntu 22.04 LTS / Windows 11)
 - **Toolchain:** Zephyr SDK v0.16.8
 - **Version used:** Zephyr 3.7 LTS

**Additional context**

The value of register r3 on the `ldr r3 [r3 #0]` instruction is beyond the default address space of the DTCM region, and as the `memc_flexram_dt_partition()` function is called after stack operations have been performed and the __stack_chk_guard is set to an initially inaccessible part of memory, this could be what's causing the crash. 

**Potential fix**

Changing the FlexRAM config before z_arm_platform_init is called with assembly code (not using memc_flexram_dt_partition()) allows us to use modified FlexRAM, SRAM=DTCM, and stack canaries enabled. I'm not sure if it's possible to generate the FlexRAM configuration register values from the devicetree, otherwise I would've made a PR 🥲

decsny, 2025-01-27:
@harristomy is this still a problem after the changes to use soc_reset_hook ?

harristomy, 2025-04-09:
Hi @decsny , I'm in the process of upgrading our environment to Zephyr 4.1, so will have a look at soc_reset_hook and let you know

harristomy, 2025-04-09:
@decsny I haven't tested it yet, but by inspection I would guess the problem still exists. The first check of the stack canary variable is done before the FlexRAM reconfiguring routine is called.

harristomy, 2025-04-15:
@decsny tested, and this issue still occurs. I've attached a screenshot below of the assembly in question.
The CPU goes to the Boot ROM due to the instruction that is highlighted. DTCM is set to 320KB for this application, and as the linker places the value at the end of DTCM (and has no knowledge of when it is configured), this instruction fails as the FlexRAM hasn't been configured yet. 
Our workaround was to implement an assembly routine that reconfigures this before soc_reset_hook (previously z_arm_platform_init) is called. Sorry if I've restated anything obvious.

[image]

harristomy, 2025-06-17:
Hmm, I must have done something wrong with our board configuration then, I'll have a look. Cheers for this @jacob-wienecke-nxp

harristomy, 2025-06-18:
@jacob-wienecke-nxp could you also enable `CONFIG_STACK_CANARIES_ALL=y`?

closed 2025-09-22 by github-actions[bot]