zephyrproject-rtos/zephyr #81217: doc: smp: Wrong length

nordicjm, 2024-11-11:
**Describe the bug**
Length is a document is wrong

**Expected behavior**
Being right

**Impact**
Is wrong

henrikbrixandersen, 2024-11-11:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

nordicjm, 2024-11-11:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Did not bother as it's a backport of a doc fix.

nashif, 2024-11-11:
> Did not bother as it's a backport of a doc fix.

tbh, I'd rather merge such a doc fix without an issue and not deal with bug reports like this, this bug report does not have the minimal language required to understand what is going on here, where is the issue, what file, what is the context? smp? Is this symmetric multi processing maybe?  

This issue eventually will be linked into release notes, someone who wants to know how changes might impact them upgrading to 3.7.1 might open this to see what is the context and what is the bug being fixed. Lacking the basic information of where the bug is and its impact might leave someone wondering.

nordicjm, 2024-11-12:
Are we ok to open/backport doc-only changes without an issue? I opened this to avoid the CI failure with needing an issue. There was no bug in the code at all, just in the documentation a part had 3 bytes instead of 5

PR #81158 (2024-11-11): [Backport v3.7-branch] doc: fix body length in SMP UART transport
Backport a77d499e2dae37d7b549f5c1954db1582763fd12 from #80611.

Fixes #81217