zephyrproject-rtos/zephyr #81230: Bluetooth: Host: bt_le_set_auto_conn() doesn't work

ivaniushkov, 2024-11-11:
**Describe the bug**
Zephyr Host has a function called [bt_le_set_auto_conn()](https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0-rc3/subsys/bluetooth/host/conn.c#L3860) which is supposed to auto-reconnect when the ACL connection disconnected.
Unfortunately, the function doesn't work as expected.
There are at least two issues with the current implementation of this function:
- When bt_le_set_auto_conn() is called from the application, the application never gets disconnected callback. It happens because in the [hci_disconn_complete()](https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0-rc3/subsys/bluetooth/host/hci_core.c#L983) handler the connection state [is set to BT_CONN_DISCONNECTED](https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0-rc3/subsys/bluetooth/host/hci_core.c#L1002) and right after that it [is set to BT_CONN_SCAN_BEFORE_INITIATING](https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0-rc3/subsys/bluetooth/host/hci_core.c#L1025), so notify_disconnected() handler is never called because it is only called from [deferred_work()](https://github.com/zephyrproject-rtos/zephyr/blob/v4.0.0-rc3/subsys/bluetooth/host/conn.c#L2233)
- Another issue is that a device never reconnects to the Peripheral. Not sure exactly why it happens, but it looks like some parts of the auto-reconnect feature are missing in conn.c and scan.c

**To Reproduce**
I modified samples/bluetooth/central sample to reproduce the issue.
The changes include: removed disconnection, added bt_le_set_auto_conn() to device_found(), increased RSSI to -30dBm to connect only to the device located very close.
Steps to reproduce the behavior:
1. cd central_auto
2. west build -bnrf52840dk_nrf52840
3. west flash

It required additional device to be used as a Peripheral

**Expected behavior**
The following messages are expected in the sample output:
```
Device found: 52:84:E1:BC:96:03 (random) (RSSI -30)
Connected: 52:84:E1:BC:96:03 (random)
Disconnected: 52:84:E1:BC:96:03, reason 0x08
Connected: 52:84:E1:BC:96:03 (random)
```

**Observed behavior**
```
Disconnected: 52:84:E1:BC:96:03, reason 0x08
```
message is never printed when Peripheral is disconnected. It happens because notify_disconnected() is never called.
Also, the second `Connected: 52:84:E1:BC:96:03 (random)` is not printed.

**Suggested solution**
I would suggest to just remove this function because the use-cases are unclear. Also it looks like it didn't work as expected for a long time

[central_auto.zip](https://github.com/user-attachments/files/17704048/central_auto.zip)

ivaniushkov, 2024-11-12:
bt_conn_le_create_auto() function is similar to bt_le_set_auto_conn() but uses BT_FILTER_ACCEPT_LIST feature. I guess it may have the same problems

jhedberg, 2024-11-14:
Bluetooth WG meeting: decided that it should be fine to remove for 4.1, however the removal should follow the documented process for stable API breakage: https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-breaking-api-changes

jhedberg, 2025-01-20:
Taken care of by #81757

closed 2025-01-20 by jhedberg