zephyrproject-rtos/zephyr #81239: Zephyr OS 4.0.0-rc3 does not work on mimxrt1060_evkb

andrisk-dev, 2024-11-11:
Examples (e.g. Hello World) does not work on NXP mimxrt1060_evkb with Zephyr 4.0.0-rc3. The example crashes with Hard Fault - Imprecise data access error (BFSR.IMPRECISERR flag raised). The error is caused by initialization code (main function is not reached).

I tried - disable optimization (CONFIG_NO_OPTIMIZATIONS=y)

The same example on the same platform works on Zephyr 3.7.0 and Zephyr 3.6.0.

**To Reproduce**

- In zephyr repository main branch (tag: v4.0.0-rc3) build the example:

west build -b mimxrt1060_evkb samples\hello_world -p always

- debug the example

west debug 
or other method of debugging (west flash, Ozone ... )

The example crashes

**Expected behavior**

The example should output hello world message on console.

**Impact**
Not able to test new features, the new version seems to be breaking NXP platforms.

**Registers**
[image]

**Environment (please complete the following information):**

 - OS: Windows 11 Enterprise
 - Toolchain: Zephyr SDK,
 - Commit SHA: c50777a8431157bd4ab3a2c2e2a37e8151ae07da

cvinayak, 2024-11-11:
Is there any chance related as https://github.com/zephyrproject-rtos/zephyr/issues/81044?
Symptoms are hard fault in initialization before reaching main(), right?

nashif, 2024-11-12:
@andrisk-dev would be nice if you could bisect it.

andrisk-dev, 2024-11-12:
> @andrisk-dev would be nice if you could bisect it.

OK I will do it.

closed 2024-11-21 by Thalley