zephyrproject-rtos/zephyr #81253: doc: Documentation typo in 'boards/weact/blackpill_f401cc/doc/index'

thetechmusician, 2024-11-12:
**Links and text typo**

In the section [supported features](https://docs.zephyrproject.org/latest/boards/weact/blackpill_f401cc/doc/index.html#supported-features), 'blackpill_f401ce' is mentioned instead of 'blackpill_f401cc'.

"The default configuration can be found in [boards/weact/blackpill_f401ce/blackpill_f401ce_defconfig](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/weact/blackpill_f401ce/blackpill_f401ce_defconfig)".
The above link should have been [this link](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/weact/blackpill_f401cc/blackpill_f401cc_defconfig) instead.

**Environment**

* Page: `boards/weact/blackpill_f401cc/doc/index`
* Version: main
* SHA-1: f2c2cc334d486ec613463774ac009df3982b3221

kartben, 2024-11-12:
@thetechmusician thanks for noticing it and opening an issue! Can you please send a pull request to fix it, by any chance?

thetechmusician, 2024-11-12:
@kartben added pull request for the fix [#81291](https://github.com/zephyrproject-rtos/zephyr/pull/81291)

PR #81291 (2024-11-13): doc: fix typo in blackpill_f401cc docs
Fix the text and links where it mentions 'blackpill_f401ce' instead of 'blackpill_f401cc'.
Fixes #81253
closed 2024-11-13 by mmahadevan108