zephyrproject-rtos/zephyr #81261: Issue through RAM / ROM summary

bayannagetteboina-eaton, 2024-11-12:
Hello there,

In order to summarize the used size vs free space in the v3 repo across all modules, we found that some of the calculations are not matching correctly when all those modules are totaled.

When we run the west build, it gives the following information about RAM size:
The total size for ROOT is “28565”:

1.  ROOT: 28565
2.  hidden: 17527
3.  no paths: 7614
4.  C:\: 1131
5.  WORKSPACE: 4
6.  ZEPHYR_BASE: 3420

The sum of rows #2 to #7 should be equal to ROOT, but it is not matching as expected.

When we sum the values from row #2 to row #7:
17527+7614+1131+4+3420=29696
This total (29696) does not match the ROOT value (28565). The discrepancy is:
29696−28565=1131

To further investigate, you might want to:

Check for overlapping data or verify calculation methods.

If you have access to more detailed logs or can provide additional context, that might help pinpoint the exact cause of the discrepancy.

Could you please investigate this further and let us know?

Regards,
Bayanna

ttmut, 2024-11-12:
Please use the bug template to report bugs.

https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

dcpleung, 2024-11-12:
You need to provide steps to reproduce.

bayannagetteboina-eaton, 2025-01-20:
Hi team,

When we do the compilation from Linux environment the total of individual modules seems to be correct. However, when wo run the same compilation on windows 10, the totals are not matching.

We are currently checking this issue at our end, will definitely reach out to you if any such help is required on this.

Thanks,
Bayanna

aniruddhanalgirkar-eaton, 2025-01-20:
[image]
[image]

**Describe the bug:**
In our application, there is a discrepancy in the total size (RAM/ROM) of individual modules. We are using the west build -t ram_report command to check this, but it produces different results on Linux and Windows.”
**Expected behavior**
The size(RAM/ROM) of individual modules must match when totaled in any application.

**Impact**
we are getting inaccurate memory profiling results.

**Logs and console output**
PFA images which shows the difference clearly.

**Environment** 

**OS**: Linux & Windows-10, 11
**Toolchain**: zephyr-sdk-0.16.5-1
**Commit SHA or Version used** : zephyr version 3.6.0
**Additional context**: we want to measure & automate the memory footprint over the PR checks, etc. Hence, we need this information consistent across different OS.

bayannagetteboina-eaton, 2025-02-28:
Hello there,

We are writing to follow up on the bug report we submitted on 20th Jan 2025. The issue is that the size(RAM/ROM) of individual modules must match when totaled in any application, and it is crucial for our operations to have it resolved as soon as possible.

Could you please provide an update on the status of this issue? If there are any additional details or steps required from our end, please let us know.

Thank you for your prompt attention to this matter.

Regards,
Bayanna

dcpleung, 2025-02-28:
Still need steps to reproduce on things we can test in this repo. We do not have your application and your build environment, and this makes it impossible for us to debug.

dcpleung, 2025-03-17:
I tried it on Windows with `blinky` and `frdm_k64f`, and I cannot reproduce it. I simply cannot debug something that I am unable to reproduce.

closed 2025-04-01 by github-actions[bot]