zephyrproject-rtos/zephyr #81267: Checkpatch and check_compliance.py disagree on spacing

SeppoTakalo, 2024-11-12:
**Describe the bug**

Using `checkpatch.pl` to check your changes:
```
$ checkpatch.pl -G upstream/main..
...
689bfcc0cc5ef56dd4e407d533474606d53cb384:63: ERROR:SPACING: space required before the open brace '{'
#63: FILE: tests/net/lib/coap_client/src/main.c:69:
+		if (memcmp(last_token[i], &(uint8_t[8]){0, 0, 0, 0, 0, 0, 0, 0}, 8) == 0) {
```

And fixing that produces warning on another check tool
```
$ ./scripts/ci/check_compliance.py -c upstream/main..HEAD -m ClangFormat
...
You may want to run clang-format on this change:
-		if (memcmp(last_token[i], &(uint8_t[8]) {0, 0, 0, 0, 0, 0, 0, 0}, 8) == 0) {
+		if (memcmp(last_token[i], &(uint8_t[8]){0, 0, 0, 0, 0, 0, 0, 0}, 8) == 0) {
```

**To Reproduce**

1. Create commit with a line above.
2. Run checkpatch and check_compliance.pl

**Expected behavior**

I expect both tools to agree on styling as both are used in CI to verify PRs.

**Impact**

annoyance

mmahadevan108, 2024-11-13:
This is similar to https://github.com/zephyrproject-rtos/zephyr/issues/78214

keith-zephyr, 2025-01-15:
This is addressed by https://github.com/zephyrproject-rtos/zephyr/pull/83374

closed 2025-03-31 by github-actions[bot]