zephyrproject-rtos/zephyr #81268: drivers/serial/uart_nrfx_uarte.c Builds with warnings nrf54l20pdk/nrf54l20/cpuapp and therefore fails in twister

aescolar, 2024-11-12:
**Describe the bug**
drivers/serial/uart_nrfx_uarte.c Builds with warnings nrf54l20pdk/nrf54l20/cpuapp and therefore fails in twister for at least these tests in main:

* tests/drivers/build_all/w1/drivers.w1.build
* tests/drivers/uart/uart_async_api/drivers.uart.async_api
* tests/drivers/uart/uart_async_api/drivers.uart.async_api.rtt
* tests/drivers/uart/uart_basic_api/drivers.uart.basic_api
* tests/drivers/uart/uart_basic_api/drivers.uart.basic_api.poll
* tests/drivers/uart/uart_basic_api/drivers.uart.basic_api.shell

**To Reproduce**

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf54l20pdk/nrf54l20/cpuapp ../tests/drivers/uart/uart_async_api 
3. ninja
4. See error

**Expected behavior**
No build warnings & twister failures

**Impact**
Failing tests in main and therefore in PRs which trigger these.

**Logs and console output**
```
FAILED: zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNRF54L20_ENGA_XXAA -DNRF_APPLICATION -DNRF_CONFIG_CPU_FREQ_MHZ=128 -DNRF_SKIP_CLOCK_CONFIGURATION -DNRF_SKIP_GLITCHDETECTOR_DISABLE -DNRF_SKIP_TAMPC_SETUP -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/__w/zephyr/zephyr/twister-out/nrf54l20pdk_nrf54l20_cpuapp/tests/drivers/build_all/w1/drivers.w1.build/w1/zephyr/include/generated/zephyr -I/__w/zephyr/zephyr/include -I/__w/zephyr/zephyr/twister-out/nrf54l20pdk_nrf54l20_cpuapp/tests/drivers/build_all/w1/drivers.w1.build/w1/zephyr/include/generated -I/__w/zephyr/zephyr/soc/nordic -I/__w/zephyr/zephyr/soc/nordic/nrf54l/. -I/__w/zephyr/zephyr/soc/nordic/common/. -I/__w/zephyr/zephyr/subsys/testsuite/include -I/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zephyr/zephyr/modules/cmsis/. -I/__w/zephyr/modules/hal/nordic/nrfx -I/__w/zephyr/modules/hal/nordic/nrfx/drivers/include -I/__w/zephyr/modules/hal/nordic/nrfx/mdk -I/__w/zephyr/zephyr/modules/hal_nordic/nrfx/. -isystem /__w/zephyr/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Werror -Os -imacros /__w/zephyr/zephyr/twister-out/nrf54l20pdk_nrf54l20_cpuapp/tests/drivers/build_all/w1/drivers.w1.build/w1/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/opt/toolchains/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /__w/zephyr/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/zephyr/zephyr/tests/drivers/build_all/w1=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -MD -MT zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj -MF zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj.d -o zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj -c /__w/zephyr/zephyr/drivers/serial/uart_nrfx_uarte.c
In file included from /__w/zephyr/zephyr/drivers/serial/uart_nrfx_uarte.c:15:
In function 'nrf_uarte_configure',
    inlined from 'uarte_nrfx_configure' at /__w/zephyr/zephyr/drivers/serial/uart_nrfx_uarte.c:504:2:
/__w/zephyr/modules/hal/nordic/nrfx/hal/nrf_uarte.h:877:23: error: 'uarte_cfg.frame_timeout' may be used uninitialized [-Werror=maybe-uninitialized]
  877 |                     | (uint32_t)p_cfg->frame_timeout
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/serial/uart_nrfx_uarte.c: In function 'uarte_nrfx_configure':
/__w/zephyr/zephyr/drivers/serial/uart_nrfx_uarte.c:443:28: note: 'uarte_cfg.frame_timeout' was declared here
  443 |         nrf_uarte_config_t uarte_cfg;
      |                            ^~~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: Ubuntu 22.04 CI
 - Toolchain (e.g Zephyr SDK, ...): 0.17
 - Commit SHA or Version used: Main as of now

**Additional context**

aescolar, 2024-11-12:
Note the HAL uses `NRF_UARTE_HAS_FRAME_TIMEOUT` while the drive uses `UARTE_HAS_FRAME_TIMEOUT`

aescolar, 2024-11-12:
Adding an extra uart23 & uart24 in dts/common/nordic/nrf54l20.dtsi fixes it.
But, @nordic-krch is there any reason to not just use the nrfx `NRF_UARTE_HAS_FRAME_TIMEOUT` macro instead?

nordic-krch, 2024-11-22:
> But, @nordic-krch is there any reason to not just use the nrfx NRF_UARTE_HAS_FRAME_TIMEOUT macro instead

yes, it could be used. When I was starting work on frame timeout hal did not supported it and i added propoerty in DT but i think that it should be ok - going more towards relying on information from DT than on MDK (to be able to using DT idx and do not need to match it with peripheral index). You can see it the other way round, it detected that DT was incomplete ;)

PR #81664 (2024-11-28): dts: common: nordic: nrf54l20: Add missing nodes
Add missing nodes: i2c23, i2c24, spi23, spi24, uart23, uart24.

Fixes #81268.
closed 2024-11-28 by kartben