zephyrproject-rtos/zephyr #81270: Zephyr OS QEMU: Network Initialization Fails (Cannot Bind to IP, Defaults to 0.0.0.0)

learnerr2024, 2024-11-12:
I'm trying to emulate a smart bulb on Zephyr OS using QEMU with networking support. My setup is designed to run an HTTP server on a TAP interface (tap0) configured with specific IPv4 addresses. However, I keep encountering network initialization errors, and Zephyr defaults to binding to [0.0.0.0](http://0.0.0.0/) instead of the expected IP. Despite configuring tap0 on the host, Zephyr doesn’t seem to recognize the network interface.

Environment

Zephyr OS Version: v4.0.0-rc1

Host OS: Ubuntu

QEMU: Installed via Zephyr SDK

Steps Taken

Configured tap0 Interface on Host:

```
``sudo ip tuntap add dev tap0 mode tap

sudo ip link set tap0 up

sudo ip addr add [192.0.2.1/24](http://192.0.2.1/24) dev tap0

sudo sysctl -w net.ipv4.ip_forward=1`
```

2.overlay-tap.conf:

``CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_SOCKETS=y

CONFIG_NET_CONFIG_SETTINGS=y

CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.2"

CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.1"

CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.1"

CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"

``

3.prj.conf

`#CONFIG_HTTP_SERVER=y

CONFIG_NETWORKING=y

CONFIG_NET_IPV4=y

CONFIG_NET_TCP=y

CONFIG_NET_SOCKETS=y

CONFIG_NET_SLIP_TAP=n

CONFIG_NET_SOCKETS_POSIX_NAMES=y

# Enable logging

CONFIG_LOG=y

CONFIG_NET_LOG=y

`

4. HTTP Server Code (main.c)

``#include <zephyr/kernel.h>

#include <zephyr/net/socket.h>

#include <zephyr/logging/log.h>

LOG_MODULE_REGISTER(http_server, LOG_LEVEL_INF);

#define HTTP_PORT 80

#define HTTP_RESPONSE "HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\nContent-Length: 13\r\n\r\nHello, World!"

void start_http_server(void)

{

struct sockaddr_in addr;

int server_fd, client_fd;

char buffer[128];

addr.sin_family = AF_INET;

addr.sin_port = htons(HTTP_PORT);

addr.sin_addr.s_addr = htonl(INADDR_ANY);

server_fd = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);

if (server_fd < 0) {

LOG_ERR("Failed to create socket");

return;

}

bind(server_fd, (struct sockaddr *)&addr, sizeof(addr));

listen(server_fd, 5);

while (1) {

client_fd = accept(server_fd, NULL, NULL);

if (client_fd < 0) {

continue;

}

recv(client_fd, buffer, sizeof(buffer) - 1, 0);

send(client_fd, HTTP_RESPONSE, sizeof(HTTP_RESPONSE) - 1, 0);

close(client_fd);

}

close(server_fd);

}

void main(void)

{

LOG_INF("Starting HTTP server...");

start_http_server();

}`

`

west build -b qemu_x86 -- -DOVERLAY_CONFIG=overlay-tap.conf

west build -t run

output:

`[00:00:00.000,000] <err> net_if: There is no network interface to work with!

[00:00:00.000,000] <wrn> net_config: No auto-started network interface - network-bound app initialization skipped.

[00:00:00.000,000] <wrn> net_if: No default interface found!

[00:00:00.000,000] <err> net_ctx: Cannot bind to 0.0.0.0`

- Why is Zephyr not binding to the IP address [192.0.2.2](http://192.0.2.2/) as configured in overlay-tap.conf?

- How can I ensure that QEMU initializes the network interface correctly for Zephyr?

- Are there any specific configurations I might be missing to get networking working properly in QEMU with Zephyr?

jukkar, 2024-11-12:
There are ready made scripts found in net-tools project for setting up the tap interface. This is described here https://docs.zephyrproject.org/latest/connectivity/networking/qemu_setup.html#networking-with-qemu

I just tested this and it works fine:
```
cd zephyrproject/tools/net-tools
./loop-socat.sh
```
and in another window
```
cd zephyrproject/tools/net-tools
sudo ./loop-slip-tap.sh
```

And I started http-server sample like this:
```
cd zephyrproject/zephyr
west build -p -b qemu_x86 -t run -d ../build/http-server \
samples/net/sockets/http_server/ -- \
-DCONFIG_NET_HTTP_SERVER_LOG_LEVEL_DBG=y \
-DCONFIG_NET_SHELL=y -DCONFIG_NET_LOG=y
```

learnerr2024, 2024-11-13:
Thanks for your help!

I dont have any real device so I tried to emulate a smart bulb seeking reference from this link:https://docs.zephyrproject.org/latest/samples/net/sockets/http_server/README.html

But I am facing some error, I need help to resolve it

here are the files that i used:
main.c
`/*
 * Copyright (c) 2023, Emna Rekik
 * Copyright (c) 2024, Nordic Semiconductor
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdio.h>
#include <inttypes.h>
#include <zephyr/kernel.h>
#include <zephyr/net/tls_credentials.h>
#include <zephyr/net/http/server.h>
#include <zephyr/net/http/service.h>
#include <zephyr/net/net_ip.h>
#include <zephyr/net/socket.h>
#include <zephyr/device.h>
#include <zephyr/sys/util.h>
#include <zephyr/drivers/led.h>
#include <zephyr/data/json.h>
#include <zephyr/sys/util_macro.h>

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(smart_bulb_emulator, LOG_LEVEL_DBG);

struct bulb_command {
    bool bulb_state;
    int brightness; // Represented as a percentage (0-100)
};

static const struct json_obj_descr bulb_command_descr[] = {
    JSON_OBJ_DESCR_PRIM(struct bulb_command, bulb_state, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct bulb_command, brightness, JSON_TOK_NUMBER),
};

static const struct device *bulb_dev = DEVICE_DT_GET_ANY(gpio_leds);

static uint8_t index_html_gz[] = {
#include "index.html.gz.inc"
};

static uint8_t main_js_gz[] = {
#include "main.js.gz.inc"
};

static struct http_resource_detail_static index_html_gz_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_STATIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
            .content_encoding = "gzip",
            .content_type = "text/html",
    },
    .static_data = index_html_gz,
    .static_data_len = sizeof(index_html_gz),
};

static struct http_resource_detail_static main_js_gz_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_STATIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
            .content_encoding = "gzip",
            .content_type = "text/javascript",
    },
[... 410 lines cut ...]
output: `west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..

*** Booting Zephyr OS build v4.0.0-rc1-22-g14478cafa6a4 ***
[00:00:00.020,000] <wrn> net_sock_tls: No entropy device on the system, TLS communication is insecure!
[00:00:00.020,000] <inf> net_config: Initializing network
[00:00:00.020,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.120,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] <inf> smart_bulb_emulator: TLS setup completed
[00:00:00.120,000] <err> net_http_server: All services failed (0)
[00:00:00.120,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:00.120,000] <inf> net_http_server: Re-starting server (-3)
[00:00:01.130,000] <err> net_http_server: All services failed (0)
[00:00:01.130,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:01.130,000] <inf> net_http_server: Re-starting server (-3)
[00:00:02.140,000] <err> net_http_server: All services failed (0)
[00:00:02.140,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:02.140,000] <inf> net_http_server: Re-starting server (-3)
uart:~$ qemu-system-i386: terminating on signal 2

(.venv) iottry@iottry:~/zephyrproject/smart_bulb_emulation$ ninja: build stopped: interrupted by user.`

learnerr2024, 2024-11-13:
Thanks for your help! I am beginning in this field

I dont have any real device so I tried to emulate a smart bulb seeking reference from this link:https://docs.zephyrproject.org/latest/samples/net/sockets/http_server/README.html

But I am facing some error, I need help to resolve it

here are the files that i used:
main.c
/*
 * Copyright (c) 2023, Emna Rekik
 * Copyright (c) 2024, Nordic Semiconductor
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#include <stdio.h>
#include <inttypes.h>
#include <zephyr/kernel.h>
#include <zephyr/net/tls_credentials.h>
#include <zephyr/net/http/server.h>
#include <zephyr/net/http/service.h>
#include <zephyr/net/net_ip.h>
#include <zephyr/net/socket.h>
#include <zephyr/device.h>
#include <zephyr/sys/util.h>
#include <zephyr/drivers/led.h>
#include <zephyr/data/json.h>
#include <zephyr/sys/util_macro.h>

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(smart_bulb_emulator, LOG_LEVEL_DBG);
struct bulb_command {
    bool bulb_state;
    int brightness; // Represented as a percentage (0-100)
};
static const struct json_obj_descr bulb_command_descr[] = {
    JSON_OBJ_DESCR_PRIM(struct bulb_command, bulb_state, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct bulb_command, brightness, JSON_TOK_NUMBER),
};
static const struct device *bulb_dev = DEVICE_DT_GET_ANY(gpio_leds);
static uint8_t index_html_gz[] = {
#include "index.html.gz.inc"
};
static uint8_t main_js_gz[] = {
#include "main.js.gz.inc"
};
static struct http_resource_detail_static index_html_gz_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_STATIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
            .content_encoding = "gzip",
            .content_type = "text/html",
    },
    .static_data = index_html_gz,
    .static_data_len = sizeof(index_html_gz),
};
static struct http_resource_detail_static main_js_gz_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_STATIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
            .content_encoding = "gzip",
            .content_type = "text/javascript",
    },
    .static_data = main_js_gz,
    .static_data_len = sizeof(main_js_gz),
};
static int bulb_status_handler(struct http_client_ctx *client, enum http_data_status status,
                               uint8_t *buffer, size_t len, struct http_response_ctx *response_ctx,
                               void *user_data)
{
    static uint8_t status_buf[64];
[... 356 lines cut ...]
output: `west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..

*** Booting Zephyr OS build v4.0.0-rc1-22-g14478cafa6a4 ***
[00:00:00.020,000] <wrn> net_sock_tls: No entropy device on the system, TLS communication is insecure!
[00:00:00.020,000] <inf> net_config: Initializing network
[00:00:00.020,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.120,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] <inf> smart_bulb_emulator: TLS setup completed
[00:00:00.120,000] <err> net_http_server: All services failed (0)
[00:00:00.120,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:00.120,000] <inf> net_http_server: Re-starting server (-3)
[00:00:01.130,000] <err> net_http_server: All services failed (0)
[00:00:01.130,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:01.130,000] <inf> net_http_server: Re-starting server (-3)
[00:00:02.140,000] <err> net_http_server: All services failed (0)
[00:00:02.140,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:02.140,000] <inf> net_http_server: Re-starting server (-3)
uart:~$ qemu-system-i386: terminating on signal 2

(.venv) iottry@iottry:~/zephyrproject/smart_bulb_emulation$ ninja: build stopped: interrupted by user.`

jukkar, 2024-11-13:
Could you adjust the code / log output text in the comments to make it readable (now it is hard to read), you need to use triple backticks (and placing them in separate lines). Use the preview button to verify that things look ok. I adjusted the initial message but the latter one has too much stuff going on.

learnerr2024, 2024-11-13:
Sure..
I will place the files in separate comments for better reading. Give me some minutes

learnerr2024, 2024-11-13:
main.c
/*
 * Copyright (c) 2023, Emna Rekik
 * Copyright (c) 2024, Nordic Semiconductor
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#include <stdio.h>
#include <inttypes.h>
#include <zephyr/kernel.h>
#include <zephyr/net/tls_credentials.h>
#include <zephyr/net/http/server.h>
#include <zephyr/net/http/service.h>
#include <zephyr/net/net_ip.h>
#include <zephyr/net/socket.h>
#include <zephyr/device.h>
#include <zephyr/sys/util.h>
#include <zephyr/drivers/led.h>
#include <zephyr/data/json.h>
#include <zephyr/sys/util_macro.h>

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(smart_bulb_emulator, LOG_LEVEL_DBG);
struct bulb_command {
    bool bulb_state;
    int brightness; // Represented as a percentage (0-100)
};
static const struct json_obj_descr bulb_command_descr[] = {
    JSON_OBJ_DESCR_PRIM(struct bulb_command, bulb_state, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct bulb_command, brightness, JSON_TOK_NUMBER),
};
static const struct device *bulb_dev = DEVICE_DT_GET_ANY(gpio_leds);
static uint8_t index_html_gz[] = {
#include "index.html.gz.inc"
};
static uint8_t main_js_gz[] = {
#include "main.js.gz.inc"
};
static struct http_resource_detail_static index_html_gz_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_STATIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
            .content_encoding = "gzip",
            .content_type = "text/html",
    },
    .static_data = index_html_gz,
    .static_data_len = sizeof(index_html_gz),
};
static struct http_resource_detail_static main_js_gz_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_STATIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
            .content_encoding = "gzip",
            .content_type = "text/javascript",
    },
    .static_data = main_js_gz,
    .static_data_len = sizeof(main_js_gz),
};
static int bulb_status_handler(struct http_client_ctx *client, enum http_data_status status,
                               uint8_t *buffer, size_t len, struct http_response_ctx *response_ctx,
                               void *user_data)
{
    static uint8_t status_buf[64];
    int ret = snprintf(status_buf, sizeof(status_buf), "{\"state\": %s, \"brightness\": %d}",
                       bulb_dev ? "on" : "off", 100);
     response_ctx->body = status_buf;
    response_ctx->body_len = ret;
    response_ctx->final_chunk = true;
    return 0;
}
static struct http_resource_detail_dynamic bulb_status_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_DYNAMIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_GET),
    },
    .cb = bulb_status_handler,
    .user_data = NULL,
};
static void parse_bulb_post(uint8_t *buf, size_t len)
{
    int ret;
    struct bulb_command cmd;
    const int expected_return_code = BIT_MASK(ARRAY_SIZE(bulb_command_descr));
    ret = json_obj_parse(buf, len, bulb_command_descr, ARRAY_SIZE(bulb_command_descr), &cmd);
    if (ret != expected_return_code) {
        LOG_WRN("Failed to fully parse JSON payload, ret=%d", ret);
        return;
    }
    LOG_INF("POST request setting bulb state %d, brightness %d%%", cmd.bulb_state, cmd.brightness);
    if (bulb_dev != NULL) {
        if (cmd.bulb_state) {
            led_on(bulb_dev, 0);
        } else {
            led_off(bulb_dev, 0);
        }
    }
}
static int bulb_control_handler(struct http_client_ctx *client, enum http_data_status status,
                                uint8_t *buffer, size_t len, struct http_response_ctx *response_ctx,
                                void *user_data)
{
    static uint8_t post_payload_buf[32];
    static size_t cursor;
    if (status == HTTP_SERVER_DATA_ABORTED) {
        cursor = 0;
        return 0;
    }
    if (len + cursor > sizeof(post_payload_buf)) {
        cursor = 0;
        return -ENOMEM;
    }
    memcpy(post_payload_buf + cursor, buffer, len);
    cursor += len;
    if (status == HTTP_SERVER_DATA_FINAL) {
        parse_bulb_post(post_payload_buf, cursor);
        cursor = 0;
    }
    return 0;
}
static struct http_resource_detail_dynamic bulb_control_resource_detail = {
    .common = {
            .type = HTTP_RESOURCE_TYPE_DYNAMIC,
            .bitmask_of_supported_http_methods = BIT(HTTP_POST),
    },
    .cb = bulb_control_handler,
    .user_data = NULL,
};
HTTP_RESOURCE_DEFINE(index_html_gz_resource, test_http_service, "/",
                     &index_html_gz_resource_detail);

HTTP_RESOURCE_DEFINE(main_js_gz_resource, test_http_service, "/main.js",
                     &main_js_gz_resource_detail);
HTTP_RESOURCE_DEFINE(bulb_status_resource, test_http_service, "/bulb_status", &bulb_status_resource_detail);
HTTP_RESOURCE_DEFINE(bulb_control_resource, test_http_service, "/bulb_control", &bulb_control_resource_detail);
int main(void)
{
   void setup_tls()
{
LOG+INF("TLS setup completed")
}
 setup_tls();
    http_server_start();
    return 0;
}

learnerr2024, 2024-11-13:
prj.conf:
General Configuration
CONFIG_MAIN_STACK_SIZE=3072
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y
CONFIG_ZVFS_OPEN_MAX=32
CONFIG_POSIX_API=y
CONFIG_FDTABLE=y
CONFIG_ZVFS_POLL_MAX=32
CONFIG_EVENTFD=y
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_SHELL=y
CONFIG_NET_STATISTICS=y
CONFIG_NET_STATISTICS_USER_API=y
CONFIG_NET_LOG=y
CONFIG_JSON_LIBRARY=y
CONFIG_HTTP_PARSER_URL=y
CONFIG_HTTP_PARSER=y
CONFIG_HTTP_SERVER=y
CONFIG_HTTP_SERVER_WEBSOCKET=y
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=128
CONFIG_NET_BUF_TX_COUNT=128
CONFIG_NET_CONTEXT_NET_PKT_POOL=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=32
CONFIG_NET_MAX_CONN=32
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=60000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=2048
CONFIG_MBEDTLS_USE_PSA_CRYPTO=n
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_NET_SOCKETS_TLS_MAX_CONTEXTS=6
CONFIG_TLS_CREDENTIALS=y
CONFIG_TLS_MAX_CREDENTIALS_NUMBER=5
CONFIG_NET_TCP_TIME_WAIT_DELAY=0
CONFIG_GPIO=n
CONFIG_LED=n
CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=n
CONFIG_NET_HTTP_LOG_LEVEL_DBG=n
CONFIG_NET_IPV6_LOG_LEVEL_DBG=n
CONFIG_NET_IPV6_ND_LOG_LEVEL_DBG=n

learnerr2024, 2024-11-13:
main.js:
main.js 
/*
 * Revised main.js for Zephyr HTTP Server with Smart Bulb Emulator
 * SPDX-License-Identifier: Apache-2.0
 */
// Function to fetch uptime information from the server
async function fetchUptime() {
    try {
        // Updated endpoint to match the server's /bulb_status path
        const response = await fetch("/bulb_status");
        if (!response.ok) {
            throw new Error(`Response status: ${response.status}`);
        }
        const json = await response.json();
        const uptime = document.getElementById("uptime");
        uptime.innerHTML = "Uptime: " + json.uptime + " milliseconds";
    } catch (error) {
        console.error("Error fetching uptime:", error.message);
    }
}
// Function to send LED state to the server
async function postLed(state) {
    try {
        // Updated payload structure to match the C code's expected JSON format
        const payload = JSON.stringify({ "bulb_state": state, "brightness": state ? 100 : 0 });

        // Updated endpoint to match the server's /bulb_control path
        const response = await fetch("/bulb_control", { method: "POST", body: payload });
        if (!response.ok) {
            throw new Error(`Response status: ${response.status}`);
        }
    } catch (error) {
        console.error("Error posting LED state:", error.message);
    }
}
// Function to set network statistics in the UI
function setNetStat(json_data, stat_name) {
    document.getElementById(stat_name).innerHTML = json_data[stat_name];
}
// Main function to initialize event listeners and WebSocket
window.addEventListener("DOMContentLoaded", (ev) => {
    // Fetch the uptime once per second
    setInterval(fetchUptime, 1000);
    // POST to the /bulb_control endpoint when the LED buttons are pressed
    const led_on_btn = document.getElementById("led_on");
    led_on_btn.addEventListener("click", () => {
        console.log("LED ON button clicked");
        postLed(true);
    });
    const led_off_btn = document.getElementById("led_off");
    led_off_btn.addEventListener("click", () => {
        console.log("LED OFF button clicked");
        postLed(false);
    });
    // Set up WebSocket connection to handle network statistics
    const ws = new WebSocket(`ws://${location.host}/ws`); // Set WebSocket path to `/ws`
    ws.onmessage = (event) => {
        const data = JSON.parse(event.data);
        setNetStat(data, "bytes_recv");
        setNetStat(data, "bytes_sent");
        setNetStat(data, "ipv6_pkt_recv");
        setNetStat(data, "ipv6_pkt_sent");
        setNetStat(data, "ipv4_pkt_recv");
        setNetStat(data, "ipv4_pkt_sent");
        setNetStat(data, "tcp_bytes_recv");
        setNetStat(data, "tcp_bytes_sent");
    };

    ws.onerror = (error) => {
        console.error("WebSocket error:", error);
    };
});

learnerr2024, 2024-11-13:
output:
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..

*** Booting Zephyr OS build v4.0.0-rc1-22-g14478cafa6a4 ***
[00:00:00.020,000] <wrn> net_sock_tls: No entropy device on the system, TLS communication is insecure!
[00:00:00.020,000] <inf> net_config: Initializing network
[00:00:00.020,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.120,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] <inf> smart_bulb_emulator: TLS setup completed
[00:00:00.120,000] <err> net_http_server: All services failed (0)
[00:00:00.120,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:00.120,000] <inf> net_http_server: Re-starting server (-3)
[00:00:01.130,000] <err> net_http_server: All services failed (0)
[00:00:01.130,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:01.130,000] <inf> net_http_server: Re-starting server (-3)
[00:00:02.140,000] <err> net_http_server: All services failed (0)
[00:00:02.140,000] <err> net_http_server: Failed to initialize HTTP2 server
[00:00:02.140,000] <inf> net_http_server: Re-starting server (-3)
uart:~$ qemu-system-i386: terminating on signal 2

(.venv) iottry@iottry:~/zephyrproject/smart_bulb_emulation$ ninja: build stopped: interrupted by user.`

jukkar, 2024-11-13:
Your comments are difficult to read because of lack of proper formatting.
What I would suggest to you is take the `samples/net/sockets/http_server` sample as a base and then adjust it by removing things that you do not need, and adding new stuff that you need. The `http_server` sample with qemu supports also TAP if you want to use it, other alternative is using normal Ethernet interface `zeth` which works with native-sim and Qemu. Instructions how to do use the interfaces can be found in the documentation (I had it linked somewhere above).