zephyrproject-rtos/zephyr #81281: samples: with_mcuboot: MCUBOOT_UPDATE_FOOTER_SIZE build warning

butok, 2024-11-12:
**Describe the bug**
There is a build warning of the zephyr\samples\sysbuild\with_mcuboot sample that can be a potential bug:

```
warning: MCUBOOT_UPDATE_FOOTER_SIZE (defined at subsys/dfu/Kconfig:55) was assigned the value
'0x2000' but got the value ''. Check these unsatisfied dependencies: MCUBOOT_IMG_MANAGER (=n),
IMG_MANAGER (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE and/or look up
MCUBOOT_UPDATE_FOOTER_SIZE in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build the with_mcuboot sample for any supported platform:
    `west build -p always -b nrf52840dk/nrf52840 --sysbuild samples/sysbuild/with_mcuboot`
3. See log with the MCUBOOT_UPDATE_FOOTER_SIZE warning :
```
-- west build: making build dir C:\_ddm\ZEPHYR\zephyr_main\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): sysbuild_default
-- Found Python3: C:/Users/nxa12205/AppData/Local/Programs/Python/Python310/python.exe (found suitable version "3.10.11", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/_ddm/ZEPHYR/zephyr_main/zephyr/.cache
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nrf52840dk, qualifiers: nrf52840
Parsing C:/_ddm/ZEPHYR/zephyr_main/zephyr/share/sysbuild/Kconfig
Loaded configuration 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/_sysbuild/empty.conf'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/samples/sysbuild/with_mcuboot/sysbuild.conf'
Configuration saved to 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/_sysbuild/autoconf.h'
--
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
-- Application: C:/_ddm/ZEPHYR/zephyr_main/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.28.1
-- Found Python3: C:/Users/nxa12205/AppData/Local/Programs/Python/Python310/python.exe (found suitable version "3.10.11", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/_ddm/ZEPHYR/zephyr_main/zephyr/.cache
-- Zephyr version: 4.0.0-rc3 (C:/_ddm/ZEPHYR/zephyr_main/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nrf52840dk, qualifiers: nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (C:/Users/nxa12205/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (C:/Users/nxa12205/zephyr-sdk-0.17.0)
-- Found Dtc: C:/Users/nxa12205/.mcuxpressotools/dtc-msys2/tools/usr/bin/dtc.exe (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: C:/_ddm/ZEPHYR/zephyr_main/zephyr/boards/nordic/nrf52840dk/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: C:/_ddm/ZEPHYR/zephyr_main/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/zephyr.dts
-- Generated pickled edt: C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/edt.pickle
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/dts.cmake
Parsing C:/_ddm/ZEPHYR/zephyr_main/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/boards/nordic/nrf52840dk/nrf52840dk_nrf52840_defconfig'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_main/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_main/bootloader/mcuboot/boot/zephyr/boards/nrf52840dk_nrf52840.conf'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/samples/sysbuild/with_mcuboot/sysbuild/mcuboot.conf'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/.config.sysbuild'
Configuration saved to 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/.config'
Kconfig header saved to 'C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/nxa12205/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/nxa12205/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
MCUBoot bootloader key file: C:/_ddm/ZEPHYR/zephyr_main/bootloader/mcuboot/root-rsa-2048.pem
Calculated maximum number of sectors: 119
[... 61 lines cut ...]
Generating files from C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/with_mcuboot/zephyr/zephyr.elf for board: nrf52840dk
image.py: sign the payload
image.py: sign the payload

[1/275] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.0-rc3 (C:/_ddm/ZEPHYR/zephyr_main/zephyr), build: v4.0.0-rc3-4-g65511eacf9d1
[275/275] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       31890 B        48 KB     64.88%
             RAM:       23680 B       256 KB      9.03%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from C:/_ddm/ZEPHYR/zephyr_main/zephyr/build/mcuboot/zephyr/zephyr.elf for board: nrf52840dk

[16/16] Completed 'mcuboot'

```

**Expected behavior**
No MCUBOOT_UPDATE_FOOTER_SIZE warning

**Environment (please complete the following information):**

 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.17
 - Commit SHA: 65511eacf9d17bf22a6cf28c3ace467a86a159f5

nordicjm, 2024-11-12:
Not a bug, the value is assigned to the image so it can calculate how large an update can be, if an application does not enable the feature then you get a warning, thus the warning is expected.

nordicjm, 2024-11-12:
Nothing to fix, expected behaviour

closed 2024-11-12 by nordicjm

butok, 2024-11-12:
> Nothing to fix, expected behaviour

Can we somehow avoid/workaround this warning?