zephyrproject-rtos/zephyr #81300: SMF smf_set_handled() bug preventing event propagation to parent states.

gbmhunter, 2024-11-12:
I think there might be a bug that causes `smf_set_handled()` to cause future calls to `smf_run_state()` to not propagate events up the state hierarchy in HSMs in certain situations.

I am using a state hierarchy in where there is a state `S1` which has multiple child states `S1A`, `S1B`, e.t.c. `S1A` calls `smf_set_handled()` on one event, and then a second event comes it to cause the state machine to transition to other child states like `S1B`. At some point the state machine transitions back to `S1A`.

It is here that I've noticed that an event can come in to `S1A`, and the parent state `run()` functions are not called. I am definitely not calling `smf_set_handled()` or `smf_set_state()` in the `s1a_run()` function, thus I'm not handling the event and would expect the parent `run()` functions to be called in backwards order.

Unfortunately the code that causes bug is private and I cannot share it. I tried to recreate a minimal working example but haven't yet been able to trigger the bug (my attempt is here: https://github.com/gbmhunter/zephyr-examples/tree/main/apps/state-machine-framework).

Further evidence that has me thinking this is a bug with the SMF library is that if I modify `zephyr/lib/smf/smf.c` so that `handled` is set to `false` everytime `smf_run_state()` is called, the bug goes away.

```c
int32_t smf_run_state(struct smf_ctx *const ctx)
{
	struct internal_ctx *const internal = (void *)&ctx->internal;
	internal->handled = false; // Added this line!!!!!!!!!!!!

	/* No need to continue if terminate was set */
	if (internal->terminate) {
		return ctx->terminate_val;
	}
```

I am running Zephyr `v3.7.0`, the Zephyr SDK `v0.17.0` and building for `native_sim`. Running this on WSL (Ubuntu).

gbmhunter, 2024-11-14:
Update: I've found that the bug can still occur, and had to add `internal->new_state = NULL;` to the start of `smf_run_state()` also. Now it is:

```c
int32_t smf_run_state(struct smf_ctx *const ctx)
{
	struct internal_ctx *const internal = (void *)&ctx->internal;
	internal->handled = false; // Added this
 	internal->new_state = NULL; // Added this

	/* No need to continue if terminate was set */
	if (internal->terminate) {
		return ctx->terminate_val;
	}
```

henrikbrixandersen, 2025-01-06:
> Update: I've found that the bug can still occur, and had to add `internal->new_state = NULL;` to the start of `smf_run_state()` also. Now it is:
>
> ```c
> int32_t smf_run_state(struct smf_ctx *const [...]

Thank you for investigating and reporting this.

The above seems like a correct fix to me. Could you please open a PR (see https://docs.zephyrproject.org/latest/contribute/index.html) and propose those changes?

gbmhunter, 2025-01-08:
@henrikbrixandersen can do! I'll hopefully get the PR open in the next few days.

closed 2025-01-23 by kartben

glenn-andrews, 2025-06-19:
Thank you for finding this. Could you add on to the tests to prevent any future regressions?

PR #83728 (2025-01-23): lib: smf: Fix handled bug causing events to not propagate.
When using the SMF for a project discovered that events would sometimes not propagate to parent states correctly. Could not create a minimum reproducable test case for this, but it was found that these changes fixed the bug. This resets the handled and new_state variables on entry to smf_run_state().

Closes #81300.