zephyrproject-rtos/zephyr #81313: Add support for RX architecture

duynguyenxa, 2024-11-13:
**Background:** We are proposing to add full support for the Renesas RX architecture to Zephyr OS. This support will enable developers to run Zephyr on RX-based microcontrollers, expanding the platform’s reach to Renesas’ RX family.

**For first basic support of the architecture we propose the below features**

- **RX Architecture Core Support**
  -  Initial implementation of core features for RX architecture, first target will be RXv1 core (only Little endian will be supported as first).
  -  The features listed as **required** in Zephyr [Architecture porting guide](https://docs.zephyrproject.org/latest/hardware/porting/arch.html) will be supported first which include:
      - The early boot sequence
      - Interrupt and exception handling
      - Thread context switching
      - Thread creation and termination
      - Device drivers: Basic driver support which include Clock control, GPIO, Pinctrl, UART.
      - Utility libraries
      - CPU idling/power management
      - Fault management
      - Linker scripts and toolchains: Our plan is to try to support [GCC RX toolchain](https://llvm-gcc-renesas.com/rx-download-toolchains/) as a Zephyr Variant External toolchain, and GCC RX build on Zephyr SDK
  - SOC: Our target for this support will be the Renesas [RX130 MCU](https://www.renesas.com/en/products/microcontrollers-microprocessors/rx-32-bit-performance-efficiency-mcus/rx130-cost-optimized-high-performance-32-bit-microcontroller-enhanced-touch-key-function-and-5v-operation?srsltid=AfmBOopf6QvGGPa3K0kr6ugZ_yfphdd68XnEJW6rOyUb9gpAYCRgjduJ)
  - Board: Our target for this support will be the Renesas [RSKRX130 512KB Board](https://www.renesas.com/en/products/microcontrollers-microprocessors/rx-32-bit-performance-efficiency-mcus/rx130-512kb-starter-kit-renesas-starter-kit-rx130-512kb)

**References**
[Renesas RX Brochure](https://www.renesas.com/en/document/bro/rx-family-brochure?r=469296)
[Renesas RX Family Overview](https://www.renesas.com/en/products/microcontrollers-microprocessors/rx-32-bit-performance-efficiency-mcus)
[Zephyr Architecture porting guide](https://docs.zephyrproject.org/latest/hardware/porting/arch.html)

nordicjm, 2024-11-13:
> Linker scripts and toolchains: Our plan is to try to support [GCC RX toolchain](https://llvm-gcc-renesas.com/rx-download-toolchains/) as a Zephyr Variant External toolchain, and GCC RX build on Zephyr SDK

@stephanosio

duynguyenxa, 2024-11-13:
@nashif , @KhiemNguyenT  , @stephanosio , @nordicjm , Please help to share your opinion about above proposal.
Especially topic relate to GCC RX toolchain and license for using rx-driver-package as hal layer

nordicjm, 2024-11-13:
@gmarull for possible comment on HAL. Note that all binary blobs must be stripped for that repo to be included in zephyr

stephanosio, 2024-11-13:
> Our plan is to try to support [GCC RX toolchain](https://llvm-gcc-renesas.com/rx-download-toolchains/) as a Zephyr Variant External toolchain

If it is more or less a vanilla GCC, a separate toolchain variant should not be necessary -- `ZEPHYR_TOOLCHAIN_VARIANT=cross-compile` can be used.

> and GCC RX build on Zephyr SDK

Is the upstream (i.e. binutils and GCC) RX architecture support well maintained? If so, adding it to the Zephyr SDK should be fairly easy.

If the upstream RX architecture support is incomplete and/or outdated compared to the Renesas GNU toolchain fork, it will likely take a lot of work to get it integrated into the Zephyr SDK -- we had a very hard time integrating Microblaze, another proprietary architecture like RX, into the SDK because the upstream binutils and GCC were buggy and missing many patches.

gmarull, 2024-11-13:
License looks fishy:

```
License
These software:
can be used for any users.
can be re-distributed by any users. (excluding r_emwin_rx)
can be integrated with specified open source code.
Amazon FreeRTOS has been now allowed, Renesas would accept other open source.
use condition is basically [MIT License](https://github.com/renesas/rx-driver-package/blob/master/LICENSE), and details are shown in [PDF file in FIT Modules directory](https://github.com/renesas/rx-driver-package/tree/master/doc/license).
are generated by SmartConfigurator as well, the generated code (each FIT Modules and Code Generator parts) license is same as previous link.
```

It's definitely not MIT.

gmarull, 2024-11-13:
Also, it looks like HAL is bloated with unnecessary stuff: see e.g. https://github.com/renesas/rx-driver-package/tree/master/source/r_bsp/r_bsp_vx.xx/r_bsp/board. HAL repos should be restricted to driver enablement code.

nashif, 2024-11-13:
is there qemu support for this architecture? How are we going to verify it works in CI?

duynguyenxa, 2024-11-14:
> is there qemu support for this architecture? How are we going to verify it works in CI?

@nashif , According to qemu document, it has support for R5F562N8 MCU which is using RXv1 Core, I'll add an action on this support to add qemu target for RX.

duynguyenxa, 2024-11-14:
> License looks fishy:
>
> ```
> License
> These software:
> can be used for any users.
> can be re-distributed by any users. (excluding r_emwin_rx)
> can be integrated with specified open source code.
> Amazon [...]

In the license PDF file, there is mention for Open Source Software usage exception. If the condition is like below, would it be allowed to use in Zephyr hal repository.
https://github.com/renesas/rx-driver-package/blob/master/doc/license/L-292(Rev1_2)_rx_mcus_rdp_sla_en.pdf

Article 3 (PROHIBITIONS)
1. Except for the rights expressly granted herein, Licensee shall not (i) remove, obliterate or modify any copyright, trademark or other proprietary notice or mark appearing on the Licensed Software by REL and/or its licensors; (ii) use, copy or modify all or any portion of the Licensed Software; (iii) distribute or sublicense the Licensed Software, in whole or in part, to any third parties; (iv) use the Licensed Software with any Open Source Software; or (v) use the Licensed Software for any purpose other than in accordance with the terms and conditions of this Agreement. **Notwithstanding (iv) above, Licensee may use the Licensed Software only with** (a) Amazon FreeRTOS provided by Amazon Web Services, Inc, (b) wolfSSL provided by wolfSSL Inc. (with respect to (b), restricted to open source software provided under GNU General Public License), (c) **Zephyr RTOS provided by Linux Foundation,** (d) Azure RTOS provided by Microsoft Corporation or (e) mbed TLS provided by Arm Ltd. Amazon FreeRTOS, wolfSSL, Zephyr RTOS, Azure RTOS and mbed TLS shall be collectively referred to as “Permitted Open Source Software”.
2. In the event of using Permitted Open Source Software, Licensee shall (i) **comply with respective terms and conditions of Permitted Open Source Software** and (ii) agree that (a) REL does not have any responsibility and obligation with respect to Permitted Open Source Software and (b) Licensee shall continue to be bound by the terms and conditions set forth herein for use of Developed Copies even in cases Developed Copies are used in conjunction with Permitted Open Source Software. Licensee shall take all necessary measures to avoid the application of the terms and conditions of Permitted Open Source Software to the Licensed Software.
3. The provisions of this Article 3 shall survive any termination of this Agreement for the life of the relevant rights.

KhiemNguyenT, 2024-11-18:
@gmarull Thanks for your comment. This will be checked with someone who know this matter well.

duynguyenxa, 2024-12-03:
> If it is more or less a vanilla GCC, a separate toolchain variant should not be necessary -- `ZEPHYR_TOOLCHAIN_VARIANT=cross-compile` can be used.
>
>
> Is the upstream (i.e. binutils and GCC) RX [...]

@stephanosio , Sorry for taking too long to reply this, currently the upstream GCC support for RX architecture is not maintained by Renesas and very different compared to the Renesas GNU toolchain fork, is it acceptable for an architecture support, for now I think we can only support the cross-compile solution. We will try to make some effort in upstreaming the GCC RX toolchain and necessary library in the near future.
Is it acceptable that an architecture support on Zephyr without ZephyrSDK support?

closed 2025-05-02 by kartben