zephyrproject-rtos/zephyr #81337: Update macros statements to allow -Werror=undef use

rkhouryUBILITE, 2024-11-13:
Bug description:

After enabling `-Werror=undef`, a set of undefined configurations appear while building various zephyr apps. The error would look like this:

```
[build] /workspaces/zephyr/zephyrproject/zephyr/drivers/serial/uart_ns16550.c:946:5: error: "CONFIG_UART_INTERRUPT_DRIVEN" is not defined, evaluates to 0 [-Werror=undef]
[build]   946 | #if CONFIG_UART_INTERRUPT_DRIVEN
[build]       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] cc1: some warnings being treated as errors

```
To reproduce:

Simply enable add `-Werror=undef` to the compiler flags.

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

alwa-nordic, 2024-11-14:
As a start I suggest we run the regex search replace: `^(\s*)#if \s*(CONFIG_[A-Z_]+)\s*$` to `$1#if defined($2)`. This will resolve 556 instances of this problem.

fabiobaltieri, 2024-11-26:
Hey, feel free to send a fix for the specific file, though if you want it to be fixed systematically there should be a conversastion to enable this in CI which is a much bigger problem.