zephyrproject-rtos/zephyr #81363: driver: adc: adc_raw_to_millivolts not usable with 24bit ADC resolution

faloj, 2024-11-13:
With 24 bits ADC readings overflow can occurs in adc_raw_to_millivolts as the internal calculation is done on a 32 bits integer.

This is fixed by using a 64 bits integer but this also requires adc_gain_invert to take a int64_t* instead of an int32t* a it's second argument, therefore slightly changing the adc API.

faloj, 2024-11-14:
Fix proposition : #81398

mrrosen, 2024-11-22:
This is actually a larger issue in Zephyr. While the ADC API and documentation leaves the actual size of each sample up to the driver (based on the resolution), in practice, most of Zephyr assumes that samples are 16-bits and most drivers tend to return 16-bits (ex: voltage divider fetches only 16-bit values).

It might be better if resolution determined sample size so no efficiency is lost for lower resolution samples but there are well-known rules for higher resolutions; probably to the nearest power of 2 to simplify accessing and doing math on samples. So, 6- and 8-bit resolution would be 1 byte; 10-, 12-, 16-bit resolution would be 2 bytes; 24-bit would be 4 bytes. It does mean that any generic driver or service (such as voltage divider) on top of the ADC would need to account for this by checking the resolution and providing a correctly sized buffer however which is more work.

closed 2025-02-09 by github-actions[bot]