zephyrproject-rtos/zephyr #81371: canbus: isotp: mimxrt1180_evk_cm33: canbus.isotp.conformance.fd.txdl_32 and 64 test failure

hakehuang, 2024-11-14:
**Describe the bug**

tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.txdl_32 and 
tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.txdl_64 failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1180_evk_cm33
 - What have you tried to diagnose or workaround this issue?
 - this is failure since this platform is added
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - N/A

**To Reproduce**
```
west twister -p mimxrt1180_evk/mimxrt1189/cm33 -T tests/subsys/canbus/isotp/conformance/canbus.isotp.conformance.fd.txdl_32
```
**Expected behavior**

Test PASS

**Impact**

canbus isotp conformance

**Logs and console output**

```
*** Booting Zephyr OS build v4.0.0-rc3 ***
Running TESTSUITE isotp_conformance
=====================================*** Booting Zephyr OS build v4.0.0-rc3 ***
Running TESTSUITE isotp_conformance
===================================================================
START - test_canfd_mandatory_padding

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:1040: isotp_conformance_test_canfd_mandatory_padding: (ret not equal to 0)
Timeout waiting for msg. ret: -11
 FAIL - test_canfd_mandatory_padding in 0.519 seconds
===================================================================
START - test_canfd_rx_dl_validation

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:275: send_frame_series: (ret not equal to 0)
Sending msg 0 failed.
 FAIL - test_canfd_rx_dl_validation in 0.025 seconds
===================================================================
START - test_receive_data

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:275: send_frame_series: (ret not equal to 0)
Sending msg 0 failed.
 FAIL - test_receive_data in 0.025 seconds
===================================================================
START - test_receive_data_blocks

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:315: add_rx_msgq: (filter_id equal to -ENOSPC)
Filter full
 FAIL - test_receive_data_blocks in 0.014 seconds
===================================================================
START - test_receive_sf
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:423: isotp_conformance_test_receive_sf: (ret not equal to ISOTP_N_OK)
Binding failed [-11]
 FAIL - test_receive_sf in 0.019 seconds
===================================================================
START - test_receive_sf_ext
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/conformance/src/main.c:479: isotp_conformance_test_receive_sf_ext: (ret not equal to ISOTP_N_OK)
[... 102 lines cut ...]
 - FAIL - [isotp_conformance.test_send_sf] duration = 0.014 seconds
 - FAIL - [isotp_conformance.test_send_sf_ext] duration = 0.014 seconds
 - FAIL - [isotp_conformance.test_send_sf_fixed] duration = 0.014 seconds
 - FAIL - [isotp_conformance.test_send_timeouts] duration = 0.024 seconds
 - FAIL - [isotp_conformance.test_sender_fc_errors] duration = 0.014 seconds
 - FAIL - [isotp_conformance.test_stmin] duration = 0.014 seconds

SUITE FAIL -  50.00% [isotp_conformance_mode_check]: pass = 1, fail = 1, skip = 0, total = 2 duration = 0.029 seconds
 - FAIL - [isotp_conformance_mode_check.test_bind] duration = 0.018 seconds
 - PASS - [isotp_conformance_mode_check.test_send] duration = 0.011 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: dd7e2366ba4ff9ef48cfb8957414ee88
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.0.0-rc3/rc2

other can driver test is ok.

mmahadevan108, 2024-11-14:
@lucien-nxp, can you take a look at this.

lucien-nxp, 2024-11-15:
#81426

hakehuang, 2024-11-27:
similar canfd support issues in tests/drivers/can/api/drivers.can.api.rtr  and tests/drivers/can/api/drivers.can.api

```
SUITE FAIL -  64.29% [canfd]: pass = 9, fail = 5, skip = 1, total = 15 duration = 0.118 seconds
 - PASS - [canfd.test_canfd_get_capabilities] duration = 0.001 seconds
 - FAIL - [canfd.test_filters_preserved_through_classic_to_fd_mode_change] duration = 0.029 seconds
```
and
```
 - FAIL - [canfd.test_filters_preserved_through_classic_to_fd_mode_change] duration = 0.029 seconds
 - FAIL - [canfd.test_filters_preserved_through_fd_to_classic_mode_change] duration = 0.028 seconds
```

william-tang914, 2025-10-23:
Hi @hakehuang , this issue has been fixed after https://github.com/zephyrproject-rtos/zephyr/pull/97681 is merged to mainline.
This is run result on `mimxrt1180_evk/mimxrt1189/cm33`, you could try this test case on mainline.
```
*** Booting Zephyr OS build v4.2.0-6748-gef818fecc13b ***
Running TESTSUITE isotp_conformance
===================================================================
START - test_canfd_mandatory_padding
 SKIP - test_canfd_mandatory_padding in 0.001 seconds
===================================================================
START - test_canfd_rx_dl_validation
 SKIP - test_canfd_rx_dl_validation in 0.001 seconds
===================================================================
START - test_receive_data
 PASS - test_receive_data in 0.294 seconds
===================================================================
START - test_receive_data_blocks
 PASS - test_receive_data_blocks in 1.148 seconds
===================================================================
START - test_receive_sf
I: SF DL does not fit. Ignore
 PASS - test_receive_sf in 0.207 seconds
===================================================================
START - test_receive_sf_ext
I: SF DL does not fit. Ignore
 PASS - test_receive_sf_ext in 0.207 seconds
===================================================================
START - test_receive_sf_fixed
 PASS - test_receive_sf_fixed in 0.208 seconds
===================================================================
START - test_receive_timeouts
E: Timeout while waiting for CF
 PASS - test_receive_timeouts in 1.006 seconds
===================================================================
START - test_receiver_fc_errors
E: Sequence number mismatch
I: Got unexpected frame. Ignore
I: Got unexpected frame. Ignore
I: Got unexpected frame. Ignore
I: Got unexpected frame. Ignore
I: Got unexpected frame. Ignore
I: Got unexpected frame. Ignore
E: Pkt length is 4089 but buffer has only 768 bytes
 PASS - test_receiver_fc_errors in 0.639 seconds
===================================================================
START - test_send_data
 PASS - test_send_data in 0.447 seconds
===================================================================
START - test_send_data_blocks
 PASS - test_send_data_blocks in 1.000 seconds
===================================================================
START - test_send_sf
 PASS - test_send_sf in 0.203 seconds
===================================================================
START - test_send_sf_ext
 PASS - test_send_sf_ext in 0.202 seconds
===================================================================
START - test_send_sf_fixed
 PASS - test_send_sf_fixed in 0.203 seconds
===================================================================
START - test_send_timeouts
E: Reception of next FC has timed out
E: Reception of next FC has timed out
E: Reception of next FC has timed out
 PASS - test_send_timeouts in 3.828 seconds
===================================================================
START - test_sender_fc_errors
E: Got overflow FC frame
I: Got to many wait frames
 PASS - test_sender_fc_errors in 0.622 seconds
===================================================================
START - test_stmin
 PASS - test_stmin in 0.265 seconds
===================================================================
TESTSUITE isotp_conformance succeeded
Running TESTSUITE isotp_conformance_mode_check
===================================================================
START - test_bind
 PASS - test_bind in 0.001 seconds
===================================================================
START - test_send
 PASS - test_send in 0.002 seconds
===================================================================
TESTSUITE isotp_conformance_mode_check succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [isotp_conformance]: pass = 15, fail = 0, skip = 2, total = 17 duration = 10.481 seconds
 - SKIP - [isotp_conformance.test_canfd_mandatory_padding] duration = 0.001 seconds
 - SKIP - [isotp_conformance.test_canfd_rx_dl_validation] duration = 0.001 seconds
 - PASS - [isotp_conformance.test_receive_data] duration = 0.294 seconds
 - PASS - [isotp_conformance.test_receive_data_blocks] duration = 1.148 seconds
 - PASS - [isotp_conformance.test_receive_sf] duration = 0.207 seconds
 - PASS - [isotp_conformance.test_receive_sf_ext] duration = 0.207 seconds
 - PASS - [isotp_conformance.test_receive_sf_fixed] duration = 0.208 seconds
 - PASS - [isotp_conformance.test_receive_timeouts] duration = 1.006 seconds
 - PASS - [isotp_conformance.test_receiver_fc_errors] duration = 0.639 seconds
 - PASS - [isotp_conformance.test_send_data] duration = 0.447 seconds
 - PASS - [isotp_conformance.test_send_data_blocks] duration = 1.000 seconds
 - PASS - [isotp_conformance.test_send_sf] duration = 0.203 seconds
 - PASS - [isotp_conformance.test_send_sf_ext] duration = 0.202 seconds
 - PASS - [isotp_conformance.test_send_sf_fixed] duration = 0.203 seconds
 - PASS - [isotp_conformance.test_send_timeouts] duration = 3.828 seconds
 - PASS - [isotp_conformance.test_sender_fc_errors] duration = 0.622 seconds
 - PASS - [isotp_conformance.test_stmin] duration = 0.265 seconds

SUITE PASS - 100.00% [isotp_conformance_mode_check]: pass = 2, fail = 0, skip = 0, total = 2 duration = 0.003 seconds
 - PASS - [isotp_conformance_mode_check.test_bind] duration = 0.001 seconds
 - PASS - [isotp_conformance_mode_check.test_send] duration = 0.002 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

henrikbrixandersen, 2025-10-23:
@william-tang914 Awesome, thank you for testing this.

closed 2025-10-24 by hakehuang