zephyrproject-rtos/zephyr #81380: test: twister: hwmap: missing serial_pty definition

hakehuang, 2024-11-14:
**Describe the bug**

the serial_pty definition is missing. in hardwaremap.py

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - any
 - What have you tried to diagnose or workaround this issue?
 - No
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - this seems missing cherry-pick 78f688a3e710fc8294fbc99d1bc4e676100ef213

**To Reproduce**

```
scripts/twister --device-testing --hardware-map map.yaml -T samples/
```

**Expected behavior**

**Impact**

**Logs and console output**

```
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.1-rc1
INFO    - Using 'zephyr' toolchain.
Traceback (most recent call last):
  File "/home/ubuntu/nxp/frdm_ke15z/zephyr/scripts/twister", line 215, in <module>
    ret = main(options, default_options)
  File "/home/ubuntu/nxp/frdm_ke15z/zephyr/scripts/pylib/twister/twisterlib/twister_main.py", line 115, in main
    ret = hwm.discover()
  File "/home/ubuntu/nxp/frdm_ke15z/zephyr/scripts/pylib/twister/twisterlib/hardwaremap.py", line 169, in discover
    self.load(self.options.hardware_map)
  File "/home/ubuntu/nxp/frdm_ke15z/zephyr/scripts/pylib/twister/twisterlib/hardwaremap.py", line 239, in load
    flash_before = self.options.flash_before and (not (flash_with_test or serial_pty))
UnboundLocalError: local variable 'serial_pty' referenced before assignment
Keeping artifacts untouched

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.1-rc1

hakehuang, 2024-11-15:
@nashif  please check this, cherry-pick the https://github.com/zephyrproject-rtos/zephyr/commit/78f688a3e710fc8294fbc99d1bc4e676100ef213 will fix this, and then I can run a board testing for this release, Thanks

nashif, 2024-11-20:
merged

PR #81444 (2024-11-20): [Backport v3.7-branch] scripts: twister: Fix serial_py referenced before assignment
Backport 78f688a3e710fc8294fbc99d1bc4e676100ef213 from #80661.

Fixes #81380
closed 2024-11-20 by nashif