zephyrproject-rtos/zephyr #81405: west debug: openocd: console garbled with `--tui` option

erwango, 2024-11-14:
**Describe the bug**

gdb option `--tui` to split screen and display source along with gdb console is available as an openocd runner option.
Though, when used with `west debug`, it provides a garbled output:

[image]

When using `--tui` gdb arg outside of `west debug`, it works as expected
[image]

Using:
```
$ ~/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s ~/zephyrproject/zephyr/boards/st/nucleo_l152re/support -s ~/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f ~/zephyrproject/zephyr/boards/st/nucleo_l152re/support/openocd.cfg -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' '-c init' '-c targets' '-c halt'
```
Then, in another window
```
$ ~/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb-py -ex 'target extended-remote :3333' build/nucleo_l152re/zephyr/zephyr.elf -tui -ex load
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_f401re sample/hello_word
2. west debug -d build --tui
4. See error

**Expected behavior**
`--tui` should 

**Impact**
`west debug` not usable with --tui option

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows): Ubuntu
 - Toolchain (e.g Zephyr SDK, ...): zephyr-sdk-0.17.0
 - Commit SHA or Version used: v4.0.0.rc3

**Additional context**
I suspect a timing issue between gdb server and client command

kartben, 2024-12-03:
@pdgendt thoughts on this one?

pdgendt, 2024-12-04:
I don't have something at hand to test the `openocd` runner, what's the output of `west debug` without the `--tui` option?

My guess would be that the `gdb` server is also started as a child proces for `west`?

erwango, 2024-12-04:
>  what's the output of west debug without the --tui option?

Works as expected.
[image]

pdgendt, 2024-12-04:
> Works as expected.

Yes, I didn't doubt that, but from the garbled image there's a print `Listening on por` in between all the output. Is this also printed when not passing `--tui`? This to confirm my suspicion of having the server output interfering with the client output.

pdgendt, 2024-12-04:
Does the following naive fix help?

```diff
diff --git a/scripts/west_commands/runners/openocd.py b/scripts/west_commands/runners/openocd.py
index 9b39791e1ee..0e477135687 100644
--- a/scripts/west_commands/runners/openocd.py
+++ b/scripts/west_commands/runners/openocd.py
@@ -396,7 +396,7 @@ class OpenOcdBinaryRunner(ZephyrBinaryRunner):
         self.print_gdbserver_message()

         if command in ('attach', 'debug'):
-            self.run_server_and_client(server_cmd, gdb_cmd)
+            self.run_server_and_client(server_cmd, gdb_cmd, stderr=subprocess.DEVNULL)
         elif command == 'rtt':
             self.print_rttserver_message()
             server_proc = self.popen_ignore_int(server_cmd)
```

erwango, 2024-12-04:
> Does the following naive fix help?

Nope
```
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
FATAL ERROR: command exited with status 1: ~/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb-py -tui -ex 'target extended-remote :3333' build/disco_l475_iot1/zephyr/zephyr.elf -ex load
```

kartben, 2024-12-04:
@erwango I am not familiar with TUI but it looks like west has supposedly included support for it since almost forever. Just curious if you were using it successfully before and it broke at some point?

pdgendt, 2024-12-04:
New attempt:

```diff
diff --git a/scripts/west_commands/runners/openocd.py b/scripts/west_commands/runners/openocd.py
index 9b39791e1ee..d97cec1fe1e 100644
--- a/scripts/west_commands/runners/openocd.py
+++ b/scripts/west_commands/runners/openocd.py
@@ -396,7 +396,12 @@ class OpenOcdBinaryRunner(ZephyrBinaryRunner):
         self.print_gdbserver_message()

         if command in ('attach', 'debug'):
-            self.run_server_and_client(server_cmd, gdb_cmd)
+            server_proc = self.popen_ignore_int(server_cmd, stderr=subprocess.DEVNULL)
+            try:
+                self.run_client(gdb_cmd)
+            finally:
+                server_proc.terminate()
+                server_proc.wait()
         elif command == 'rtt':
             self.print_rttserver_message()
             server_proc = self.popen_ignore_int(server_cmd)
```

erwango, 2024-12-04:
> New attempt:

and this will be the last... It works like a charm, now. Thanks !

kartben, 2024-12-04:
Confirmed!
It does look like this will need to be applied (factorized?) to other runners too. I can confirm that a similar fix seems to be needed for e.g. pyocd runner, so I wouldn't be surprized if it's needed for jlink too... maybe others?

pdgendt, 2024-12-04:
Maybe some option like `suppress_server_output` could be introduced for all runners?

pdgendt, 2024-12-10:
> @pdgendt @erwango sounds like you have a workaround, are any of you planning to open a PR for this?

I'll open a PR with the current suggestion, I don't have the bandwidth for a more generic solution.

PR #82817 (2024-12-12): scripts: west_commands: runners: openocd: Fix debugging with TUI
Prevent server output when running gdb with --tui to prevent garbled console output.

Fixes #81405
closed 2024-12-12 by kartben