zephyrproject-rtos/zephyr #81458: hwmv2: series and families application folders

erwango, 2024-11-15:
**Is your enhancement proposal related to a problem? Please describe.**
As one of the additions of hwmv2, it was added the possibility to factorize configuration (dts overlays, Kconfig fragments) for a given soc under a new application folder named `soc/`.
This new folder provide the same functionality of `boards/` folder, but allows to put all configuration that depends on a soc into a single file, hence sparing the need to replicate these configurations in various boards that uses the same soc version.

This new possibility is likely pretty useful in down stream projects (see [related golioth blog post](https://blog.golioth.io/use-zephyr-soc-config-files-to-streamline-hardware-types/?utm_content=315431487&utm_medium=social&utm_source=linkedin&hss_channel=lcp-71311047) as example.

Though, in zephyr tree, the possibility of usages of `socs/` folder is relatively limited because this is not often that several boards share the exact same soc, and even if it happens, within in tree applications (samples, tests), we usually want to have one board for each soc but not more. Sometime even, we just want to have one board per whole soc series as aim is to maximize test coverage at series level while minimizing the CI load.

**Describe the solution you'd like**
In the same vein as `socs\` folders, add `series\` and `families\` application folders.
Using these folder it would be easy to provide application specific configuration at the right level while minimizing the number of files to maintain, limit CI load and ensure tests and samples compatibility with the maximum of boards.

For instance:
Replace 
```
samples/drivers/dac/
├── boards
│   ├── b_u585i_iot02a.overlay
│   ├── disco_l475_iot1.overlay
│   ├── nucleo_f091rc.overlay
│   ├── nucleo_f207zg.overlay
│   ├── nucleo_f429zi.overlay
│   ├── nucleo_f746zg.overlay
│   ├── nucleo_f767zi.overlay
│   ├── nucleo_g071rb.overlay
│   ├── nucleo_g431rb.overlay
│   ├── nucleo_g474re.overlay
│   ├── nucleo_h743zi.overlay
│   ├── nucleo_l073rz.overlay
│   ├── nucleo_l152re.overlay
│   ├── nucleo_l552ze_q.overlay
│   ├── nucleo_u575zi_q.overlay
│   ├── nucleo_wl55jc.overlay
│   ├── stm32f3_disco.overlay
│   └── stm32l562e_dk.overlay
```
with
```
samples/drivers/dac/
├── families
│   └── stm32.overlay

``` 

About CI load, one of the major issues in CI load is the frequent use of filter that impose to prebuild board configurations before being able to apply the filters. `series/` and `families` folder could help reducing the usage of those filters as well.

For instance:
```
  drivers.spi.stm32_spi_16bits_frames.loopback:
    extra_args:
      - EXTRA_CONF_FILE="overlay-stm32-spi-16bits.conf"
      - DTC_OVERLAY_FILE="overlay-stm32-spi-16bits.overlay"
    filter: CONFIG_SOC_FAMILY_STM32
    platform_allow:
      - nucleo_h743zi
      - nucleo_h753zi
      - nucleo_h745zi_q/stm32h745xx/m4
      - nucleo_h745zi_q/stm32h745xx/m7
```
could be replaced by `families/stm32.conf` and `families/stm32.overlay`, with the additional benefit that this configuration would apply also when running `west`, not only when running twister.
 
**Describe alternatives you've considered**
Introduce `Kconfig.defconfig` files in samples and tests (if that's even doable).

erwango, 2024-11-15:
^^ @zephyrproject-rtos/testing

ajf58, 2024-11-29:
This would be very useful.

A suggestion:

Could this be namespaced either in the folder name, or by being a set of subfolders? Eg.
```
samples/drivers/adc/
├── soc_series
│   └── rp2350.overlay
samples/drivers/dac/
├── soc_family
│   |── rpi_pico.overlay
│   └── stm32.overlay
```
or using subdirectories:
```
samples/drivers/adc/soc/
├── series
│   └── rp2350.overlay
samples/drivers/adc/soc/
├── family
│   |── rpi_pico.overlay
│   └── stm32.overlay
```
This might be easier to understand than `families` or `series(es)` without some kind of qualifier/prefix.

hakehuang, 2025-08-22:
the board variation on io/pin settings, can not be covered by this, any idea to enhance?

erwango, 2025-08-22:
> the board variation on io/pin settings, can not be covered by this, any idea to enhance?

I think that is a different topic. Would you mind open a dedicated issue ?

ajf58, 2025-09-07:
@zephyrproject-rtos/testing what's the next step here? Is there interest/would their be "buy-in" on this? From my (bias) perspective this feels more elegant than the current situation.

erwango, 2025-09-08:
> @ajf58 , we need do a full regression on this. @anasnashif maybe github CI need involved as well, I will post my report here once done.

There is nothing to test yet

hakehuang, 2025-09-08:
> There is nothing to test yet

oh, sorry, I assume you will have a pull request, or still in discussion on how to?

erwango, 2025-09-10:
> It's a fairly big chunk of work that would affect everyone. I don't want to create a PR and then have someone say "we don't want this.". Therefore my question is: would there be interest in this?

Obviously, I'll support such work. 

> for me, it is yes or no, it depends on how elegant the solution would be.

Can you elaborate ? For instance, what's you opinion on https://github.com/zephyrproject-rtos/zephyr/issues/81458#issuecomment-3210431966 ?

Also, @tejlmand @PerMac what's you take on this ?

hakehuang, 2025-09-10:
> Can you elaborate ? For instance, what's you opinion on [#81458 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/81458#issuecomment-3210431966) ?

my concern is about the platform_allow, it will getting out of control soon
```
  drivers.spi.stm32_spi_16bits_frames.loopback:
    extra_args:
      - EXTRA_CONF_FILE="overlay-stm32-spi-16bits.conf"
      - DTC_OVERLAY_FILE="overlay-stm32-spi-16bits.overlay"
    filter: CONFIG_SOC_FAMILY_STM32
    platform_allow:
      - nucleo_h743zi
      - nucleo_h753zi
      - nucleo_h745zi_q/stm32h745xx/m4
      - nucleo_h745zi_q/stm32h745xx/m7
```

erwango, 2025-10-17:
> but I've put a TSC label on this

Architecture review would be the first step. I did the change.
@carlescufi aim is to have a go on the direction before starting. Not absolutely required if we get feedback from @tejlmand such as:
- points to pay attention to, things to avoid, condition to make it acceptable 
- advices on how to proceed

mbolivar, 2025-11-10:
@erwango thanks for filing this issue and showing your intent via a docs patch, very helpful. I support the idea.

I have a suggestion, which would be to group these under socs/ instead of having parallel series/ or family/ directories.

For example, and based on your docs patch, I would suggest `socs/<SOC>/<FAMILY>/<BOARD_QUALIFIERS>.overlay` instead of `families/<FAMILY>_<BOARD_QUALIFIERS>.overlay`. Similar suggestions for series.

The reason why is that this will prevent us from having to enforce global uniqueness of family or series names across SoC vendors, which could be a head-ache in the long term.

erwango, 2025-11-12:
>  I support the idea.

Great, thanks!

> I have a suggestion, which would be to group these under socs/ instead of having parallel series/ or family/ directories.

Agree, please also note 2 others alternative proposals in this direction:
https://github.com/zephyrproject-rtos/zephyr/issues/81458#issuecomment-2508697492
https://github.com/zephyrproject-rtos/zephyr/issues/81458#issuecomment-3210431966

For the doc update, I only took the most simple one, but it would have to be updated depending on the solution adopted.

> For example, and based on your docs patch, I would suggest `socs/<SOC>/<FAMILY>/<BOARD_QUALIFIERS>.overlay` instead of `families/<FAMILY>_<BOARD_QUALIFIERS>.overlay`. Similar suggestions for series.

I'm not sure to follow. Having to use a preliminary `/<SOC>/` directory would cancel the possibility of actually factorizing overlay for a family. Am I missing something ?

tejlmand, 2025-11-28:
@erwango thanks for raising this.

There are for sure some good thoughts regarding this, and sharing of config settings across boards (or sometimes samples) is very useful.
Especially for downstream users, who then can ease their own maintenance.
And when they are in charge, they also know and easier understand their own structure.

In Zephyr itself, we must find the right balance between making Zephyr as easy-to-go-to for new-comers as well as striking a good balance for maintaining samples.

Though I myself like the idea, then I will try to put myself in place of users of Zephyr.

Zephyr itself is already complex enough and we should be careful on not making it even harder.
One can say, `this enhancement just adds to existing pattern`, and yes, true for some parts, but where I see limitation is especially in predicting which files to be picked up:

Currently, building for `nrf5340dk/nrf5340/cpuapp` and files used are directly coupled, like this:
```
nrf5340dk/nrf5340/cpuapp
|----------------------|
               |
               |-----------> boards/nrf5340dk_nrf5340_cpuapp.conf

          |------------|
               |
               |-----------> socs/nrf5340_cpuapp.conf
```

but allowing for families or series makes this prediction impossible:
```
nrf5340dk/nrf5340/cpuapp
          |------------|
                |
                |-------------> family/?.conf
                |-------------> series/?.conf
```

sure you can lookup the `soc.yml` for the SoC and see its family or series there, and there usually is some relation from the SoC to the family / series, but it's not always consistent.

For example:
- SoC: nrf5340,     family: nordic_nrf,      series: nrf53
- SoC: nrf54l15,    family: nordic_nrf,      series: nrf54l
- SoC: esp32,       family: espressif_esp32, series: esp32
- SoC: esp32s3,     family: espressif_esp32, series: esp32s2
- SoC: stm32c051xx, family: stm,             series: stm32c0x
- ...

some series and families are easy to spot from their name, others are harder (nrf54l15, should I include the `l` or not for series ?).
And some are even has a series name which is a subset of another series.
I'm also sure there are even harder SoCs to guess than those above.

Users can also try a build for the board and look at the CMake output to know which files was actually picked up, but that's a bit late and actually assumes the user is familiar with this part of the build system and knows what to look for.

Though I do like the idea, and I do see the need, especially for downstream users handling multiple projects, then I'm more concerned on using the principle in Zephyr itself as it makes an already complex system even more complex / harder to know why a given file is selected for configuration.

I think the idea itself should be explored more, and I would especially like some ideas on how to handle the family / series challenge described, so that any solution becomes as easy to understand as the current board target --> file name relation.