zephyrproject-rtos/zephyr #81493: stm32: stm32u0 clock dt-binding: the RTC_SEL macro should use the register BDC register

ExaltZephyr, 2024-11-17:
In [stm32u0_clock.h:90](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/dt-bindings/clock/stm32u0_clock.h#L90)
The `RTC_SEL` maco should use the `BDCR_REG` register value instead of `CSR_REG` (there's no such thing):
```diff
- #define RTC_SEL(val)		STM32_DOMAIN_CLOCK(val, 3, 8, CSR_REG)
+ #define RTC_SEL(val)		STM32_DOMAIN_CLOCK(val, 3, 8, BDCR_REG)
```

henrikbrixandersen, 2024-11-17:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

rruuaanng, 2024-11-18:
If we can, I'll commit a PR :)

closed 2024-11-20 by nashif