zephyrproject-rtos/zephyr #81498: Fails to Initialize HTTP2 Server in Zephyr OS

learnerr2024, 2024-11-17:
I tried everything I can and I need help to solve this. I don't have any real device. Please help!
Thanks in advance.

output:
-- west build: running target run
```

[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build v4.0.0-rc1-22-g14478cafa6a4 ***
[00:00:00.020,000] net_sock_tls: No entropy device on the system, TLS communication is insecure!
[00:00:00.020,000] net_config: Initializing network
[00:00:00.020,000] net_config: IPv4 address: 192.0.2.1
[00:00:00.120,000] net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] net_config: IPv6 address: 2001:db8::1
[00:00:00.120,000] smart_bulb_emulator: TLS setup completed
[00:00:00.120,000] net_http_server: All services failed (0)
[00:00:00.120,000] net_http_server: Failed to initialize HTTP2 server
[00:00:00.120,000] net_http_server: Re-starting server (-3)
[00:00:01.130,000] net_http_server: All services failed (0)
[00:00:01.130,000] net_http_server: Failed to initialize HTTP2 server
[00:00:01.130,000] net_http_server: Re-starting server (-3)
[00:00:02.140,000] net_http_server: All services failed (0)
[00:00:02.140,000] net_http_server: Failed to initialize HTTP2 server
[00:00:02.140,000] net_http_server: Re-starting server (-3)
uart:~$ qemu-system-i386: terminating on signal 2
```
prj.conf
//General config
CONFIG_MAIN_STACK_SIZE=3072
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y
CONFIG_ZVFS_OPEN_MAX=32
CONFIG_POSIX_API=y
CONFIG_FDTABLE=y
CONFIG_ZVFS_POLL_MAX=32
// Eventfd
CONFIG_EVENTFD=y
// Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_SHELL=y
CONFIG_NET_STATISTICS=y
CONFIG_NET_STATISTICS_USER_API=y
CONFIG_NET_LOG=y
// JSON
CONFIG_JSON_LIBRARY=y
// HTTP parser
CONFIG_HTTP_PARSER_URL=y
CONFIG_HTTP_PARSER=y
CONFIG_HTTP_SERVER=y
CONFIG_HTTP_SERVER_WEBSOCKET=y
// Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=128
CONFIG_NET_BUF_TX_COUNT=128
CONFIG_NET_CONTEXT_NET_PKT_POOL=y
// IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=32
CONFIG_NET_MAX_CONN=32
// Network address config
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
// TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=60000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=2048
CONFIG_MBEDTLS_USE_PSA_CRYPTO=n
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_NET_SOCKETS_TLS_MAX_CONTEXTS=6
CONFIG_TLS_CREDENTIALS=y
CONFIG_TLS_MAX_CREDENTIALS_NUMBER=5
//Networking tweaks
//Required to handle large number of consecutive connections,
//e.g. when testing with ApacheBench.
CONFIG_NET_TCP_TIME_WAIT_DELAY=0
// Device drivers
CONFIG_GPIO=n
CONFIG_LED=n
// Network debug config
CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=n
CONFIG_NET_HTTP_LOG_LEVEL_DBG=n
CONFIG_NET_IPV6_LOG_LEVEL_DBG=n
CONFIG_NET_IPV6_ND_LOG_LEVEL_DBG=n

learnerr2024, 2024-11-18:
Please Help! If anything else needed let me know

jukkar, 2024-11-18:
Related discussion in https://github.com/zephyrproject-rtos/zephyr/discussions/81386

learnerr2024, 2024-11-18:
I closed it.
I need solution of this error
thanks in adavnce

closed 2024-12-04 by carlescufi