zephyrproject-rtos/zephyr #81508: Unable to run bluetooth peripheral_hids sample on ESP32S3-Devkitc

Zhi-Yih-Lim, 2024-11-18:
**Describe the bug**
When attempting to run the "peripheral_hids" sample on ESP32S3 devkitc, CPU crashes on FATAL EXCEPTION when attempting to invoke the "bt_enable()" function.
Building and flashing the "peripheral" sample results in the same error.

**To Reproduce**
1. cd ~/zephyrproject/zephyr
2. west blobs fetch hal_espressif
3. west build -p -b esp32s3_devkitc/esp32s3/procpu samples/bluetooth/peripheral_hids
4. west flash
5. west espressif monitor

**Logs and console output**
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc958f8,len:0x2b34
load:0x40374000,len:0x118d4
0x40374000: _WindowOverflow4 at /home/parallels/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

SHA-256 comparison failed:
Calculated: d4fed89d944d3015a901d3ed683cc8919d876461686a3104bd25874245ebe766
Expected: 00000000b0bb0000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x4037f938
0x4037f938: __start at /home/parallels/zephyrproject/zephyr/soc/espressif/common/loader.c:231

I (99) boot: ESP Simple boot
I (99) boot: compile time Nov 17 2024 22:05:55
W (100) boot: Unicore bootloader
I (100) spi_flash: detected chip: gd
I (101) spi_flash: flash io: dio
W (104) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (116) boot: chip revision: v0.2
I (119) boot.esp32s3: Boot SPI Speed : 40MHz
I (123) boot.esp32s3: SPI Mode       : DIO
I (127) boot.esp32s3: SPI Flash Size : 8MB
I (131) boot: Enabling RNG early entropy source...
I (135) boot: DRAM: lma 0x00000020 vma 0x3fc958f8 len 0x2b34   (11060)
I (142) boot: IRAM: lma 0x00002b5c vma 0x40374000 len 0x118d4  (71892)
0x40374000: _WindowOverflow4 at /home/parallels/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

I (148) boot: padd: lma 0x00014448 vma 0x00000000 len 0xbbb0   (48048)
I (154) boot: IMAP: lma 0x00020000 vma 0x42000000 len 0x24510  (148752)
0x42000000: _stext at ??:?

I (161) boot: padd: lma 0x00044518 vma 0x00000000 len 0xbae0   (47840)
I (167) boot: DMAP: lma 0x00050000 vma 0x3c030000 len 0x6dd0   (28112)
I (173) boot: Image with 6 segments
I (176) boot: DROM segment: paddr=00050000h, vaddr=3c030000h, size=06DD0h ( 28112) map
I (184) boot: IROM segment: paddr=00020000h, vaddr=42000000h, size=2450Eh (148750) map
I (228) heap: ESP runtime heap init at 0x3fc9f358 size 296 kB.

I (228) spi_flash: flash io: dio
*** Booting Zephyr OS build v3.7.0-4339-g1ec5ce05f9f8 ***
[00:00:00.241,000] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.241,000] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:00.241,000] <inf> fs_nvs: data wra: 0, 0
[00:00:00.241,000] <inf> esp32_bt_adapter: BT controller compile version [963cad4]
[00:00:00.272,000] <err> os:  ** FATAL EXCEPTION
[00:00:00.272,000] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)
[00:00:00.272,000] <err> os:  **  PC 0x420012ff VADDR 0xf0b9d5c
0x420012ff: chunk_field at /home/parallels/zephyrproject/zephyr/lib/heap/heap.h:116

[00:00:00.272,000] <err> os:  **  PS 0x60e20
[00:00:00.272,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:14 CALLINC:2)
[00:00:00.272,000] <err> os:  **  A0 0x82001590  SP 0x3fc9ce60  A2 0x1ff  A3 0xf0b9d58
[00:00:00.272,000] <err> os:  **  A4 0xf0b9d5c  A5 0x20e  A6 0xfcc3  A7 0
[00:00:00.272,000] <err> os:  **  A8 0x7fff  A9 0x7fff A10 0x41c A11 0x3fc9d970
[00:00:00.272,000] <err> os:  ** A12 0x3fc9d972 A13 0x41c A14 0x3fc977bd A15 0x3fc9cf40
[00:00:00.272,000] <err> os:  ** LBEG 0x400570e8 LEND 0x400570f3 LCOUNT 0
[00:00:00.272,000] <err> os:  ** SAR 0x18
[00:00:00.272,000] <err> os:  **  THREADPTR 0x3fc99840
[00:00:00.272,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.272,000] <err> os: Current thread: 0x3fc990e8 (unknown)
[00:00:00.388,000] <err> os: Halting system

**Environment:**

 - OS: Ubuntu 24.04.1 LTS, 64-bit, ARM64
 - Toolchain: zephyr-sdk-0.16.8
 - Board: ESP32-S3-WROOM-1 (MCN8R8)
 - Connection: MicroB connected to "UART" port on the board.

**Additional context**
The minimal code to reproduce the error:

#include <zephyr/types.h>
#include <stddef.h>
#include <string.h>
#include <errno.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/byteorder.h>
#include <zephyr/kernel.h>

#include <zephyr/settings/settings.h>

#include <zephyr/bluetooth/bluetooth.h>
#include <zephyr/bluetooth/hci.h>
#include <zephyr/bluetooth/conn.h>
#include <zephyr/bluetooth/uuid.h>
#include <zephyr/bluetooth/gatt.h>

int main (void){
    // initialize BLE
    int err = bt_enable(NULL);

    return 0;
}

Zhi-Yih-Lim, 2024-11-18:
Built and flashed the "samples/bluetooth/peripheral" example from another PC with the same sdk version (zephyr-sdk-0.16.8) and same Ubuntu version (Ubuntu 24.04.1 LTS) but on "x86_64" instead of "ARM64", and the example ran without any errors.

sylvioalves, 2024-11-19:
@Zhi-Yih-Lim, `samples/bluetooth/peripheral` will work because it does not need NVS. `samples/bluetooth/peripheral_hids`  crashed because it does use NVS.

Current branch/release you have been using (v3.7.0) has a flash bug that was fixed in release v4.0.0 or v3.7-branch (which will be soon v3.7.1).

My suggestion for you is to switch to v4.0.0 if possible. If not, wait a few days for the v3.7.1.

Zhi-Yih-Lim, 2024-11-19:
Thank you @sylvioalves , the example works now. I followed your instructions to update Zephyr's version on my local machine and the error went away for the "samples/peripheral" example.   

However, there are still two lines of error during runtime when I build and flash the "samples/peripheral_hids" example onto the board. I have not looked into detail "samples/peripheral_hids" 's source code so the error could be some settings that I have to but did not set. I am not working with the "samples/peripheral_hids" example for now so the errors do not affect me. 

I will still include the log output below for your reference:

Logs and console output
------------------------
I (101) boot: ESP Simple boot
I (101) boot: compile time Nov 18 2024 22:22:44
W (101) boot: Unicore bootloader
I (102) spi_flash: detected chip: gd
I (103) spi_flash: flash io: dio
W (106) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (118) boot: chip revision: v0.2
I (121) boot.esp32s3: Boot SPI Speed : 40MHz
I (125) boot.esp32s3: SPI Mode       : DIO
I (129) boot.esp32s3: SPI Flash Size : 8MB
I (133) boot: Enabling RNG early entropy source...
I (138) boot: DRAM: lma 0x00000020 vma 0x3fc96290 len 0x2b34   (11060)
I (144) boot: IRAM: lma 0x00002b5c vma 0x40374000 len 0x1226c  (74348)
0x40374000: _WindowOverflow4 at /home/parallels/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

I (150) boot: padd: lma 0x00014dd8 vma 0x00000000 len 0xb220   (45600)
I (156) boot: IMAP: lma 0x00020000 vma 0x42000000 len 0x285c0  (165312)
0x42000000: _stext at ??:?

I (163) boot: padd: lma 0x000485c8 vma 0x00000000 len 0x7a30   (31280)
I (169) boot: DMAP: lma 0x00050000 vma 0x3c030000 len 0x743c   (29756)
I (175) boot: Image with 6 segments
I (178) boot: DROM segment: paddr=00050000h, vaddr=3c030000h, size=07440h ( 29760) map
I (186) boot: IROM segment: paddr=00020000h, vaddr=42000000h, size=285BEh (165310) map
I (230) heap_runtime: ESP heap runtime init at 0x3fca0df8 size 290 kB.
I (230) spi_flash: flash io: dio
*** Booting Zephyr OS build v4.0.0-405-g2efc8598e39f ***
[00:00:00.242,000] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.242,000] <inf> fs_nvs: alloc wra: 0, f70
[00:00:00.242,000] <inf> fs_nvs: data wra: 0, ac
[00:00:00.243,000] <inf> esp32_bt_adapter: BT controller compile version [f583012]
[00:00:00.371,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.371,000] <err> settings: set-value failure. key: bt/irk error(-2)
[00:00:00.372,000] <err> settings: set-value failure. key: bt/name error(-2)
[00:00:00.373,000] <inf> bt_hci_core: Identity: 24:58:7C:D5:EC:AC (public)
[00:00:00.373,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x02e5
[00:00:00.373,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Advertising successfully started
Bluetooth authentication callbacks registered.

closed 2024-11-23 by Zhi-Yih-Lim