zephyrproject-rtos/zephyr #81570: ESP32 fails on assert when WiFi enabled (3.7.x LTS)

real-tintin, 2024-11-19:
**Describe the bug**
In Zephyr 3.7.x (tested both 3.7.0 and 3.7.1-rc1), when enabling WiFi and asserts, the ESP32 won't start. This was previously reported and fixed in https://github.com/zephyrproject-rtos/zephyr/issues/37236.

When building and run the WiFi shell sample with asserts enabled it doesn't even log after boot.
`west build -p -b esp32_devkitc_wroom/esp32/procpu zephyr/samples/net/wifi/ -DCONFIG_ASSERT=y`

However, when I build the Hello World sample and enable WiFi:

```
CONFIG_ASSERT=y

CONFIG_WIFI=y
CONFIG_NETWORKING=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_DHCPV4=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_NET_MGMT_EVENT_INFO=y
CONFIG_NET_L2_ETHERNET=y
```
and 

```
&wifi {
	status = "okay";
};
```

and 

`west build -p -b esp32_devkitc_wroom/esp32/procpu zephyr/samples/hello_world`

I get:

```
ASSERTION FAIL [arch_irq_unlocked(key) || _kernel.cpus[0].current->base.thread_state & (((1UL << (0))) | ((1UL << (3))))] @ WEST_TOPDIR/zephyr/kernel/include/kswap.h:99
	Context switching while holding lock!

Backtrace:0x40084d2c:0x3ffb4220 0x400d3a75:0x3ffb4230 0x400871a4:0x3ffb4250 0x400877d2:0x3ffb4270 0x40085f69:0x3ffb42a0 0x400d5180:0x3ffb42d0 0x40090d05:0x3ffb42f0 0x400d3a67:0x3ffb4320 
0x40084d2c: xtensa_arch_except at /Users/oscargoldring/git/zephyr_os_github/zephyr/arch/xtensa/core/xtensa_asm2_util.S:207

0x400d3a75: assert_post_action at /Users/oscargoldring/git/zephyr_os_github/zephyr/lib/os/assert.c:43

0x400871a4: k_spin_release at /Users/oscargoldring/git/zephyr_os_github/zephyr/include/zephyr/spinlock.h:363
 (inlined by) do_swap at /Users/oscargoldring/git/zephyr_os_github/zephyr/kernel/include/kswap.h:116
 (inlined by) z_swap at /Users/oscargoldring/git/zephyr_os_github/zephyr/kernel/include/kswap.h:188

0x400877d2: z_pend_curr at /Users/oscargoldring/git/zephyr_os_github/zephyr/kernel/sched.c:696 (discriminator 2)

0x40085f69: z_impl_k_msgq_get at /Users/oscargoldring/git/zephyr_os_github/zephyr/kernel/msg_q.c:272

0x400d5180: k_msgq_get at /Users/oscargoldring/git/zephyr_os_github/build/zephyr/include/generated/zephyr/syscalls/kernel.h:1207
 (inlined by) queue_recv_wrapper at /Users/oscargoldring/git/zephyr_os_github/modules/hal/espressif/zephyr/esp32/src/wifi/esp_wifi_adapter.c:363

0x40090d05: ppTask at ??:?

0x400d3a67: z_thread_entry at /Users/oscargoldring/git/zephyr_os_github/zephyr/lib/os/thread_entry.c:48
```

which was also reported in the earlier mentioned issue. 

I did the same in 4.0.0., there it works.

**Expected behavior**
WiFi and assert works.

**Impact**
We can't bump to 3.7.x LTS.

**Environment (please complete the following information):**
 - Zephyr 3.7.0 and 3.7.1-rc1
 - Zephyr SDK: 16.9

sylvioalves, 2024-11-19:
@real-tintin thanks for reporting this. That was indeed a regression. Can you check #81584?

real-tintin, 2024-11-20:
Tested, works fine! Thanks for the quick fixing, much appreciated @sylvioalves

PR #81584 (2024-11-20): [v3.7-branch] west.yml: fix phy nested critical section
PHY section needs nested locking/unlocking control, which was wrongly removed from v3.7.x.

Fixes #81570
commit 6a50fd3c (2024-11-20): [v3.7-branch] west.yml: fix phy nested critical section

PHY section needs nested locking/unlocking control, which was
wrongly removed from v3.7.x.

Fixes #81570
closed 2024-11-20 by real-tintin