zephyrproject-rtos/zephyr #81572: gc9x01x_write does not write correct data if desc->width != desc->pitch

Casales, 2024-11-19:
**Bug description**
When writing a buffer to a gc9x01x display with a buffer width less than the displays width the display does not update correctly.

In the display driver for gc9x01x the buffer is split into rows and written in multiple transmits because the MIPI API requires width to be equal to pitch.

The mipi_desc.buff_size therefore should be set equal to the desc->width but it is set to desc->pitch.

**To Reproduce**
1. Take a board with a gc9x01x display
2. Make a buffer with a width smaller than the display width
3. Set the pixels in the buffer to show a clear pattern (for example a small image or a character)
4. Call display_write(..) with your buffer.
5. Display shows a weird pattern

**Expected behavior**
Display updates the correct pixels if write display_write(..) is called with a buffer with a smaller width than display width

**Impact**
This is annoying but can easily be fixed by changing line 533 in display_gc9x01x.c from:
`mipi_desc.buf_size = desc->pitch * data->bytes_per_pixel;`
 to: 
`mipi_desc.buf_size = desc->width* data->bytes_per_pixel;`

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain: Zephyr SDK
 - Version: 2.8.0

rruuaanng, 2024-11-19:
If we can. Maybe you should commit a PR for fix it :)

Casales, 2024-11-19:
it seems the same issue exists in mutiple other display drivers that swiched to the new mipi api. I have no other display to test right now but i'll mention some drivers that are probably affected so people can find this issue and the pullrequest (https://github.com/zephyrproject-rtos/zephyr/pull/81601)

- display_ili9xxx
- display_st7735r
- display_st7789v

closed 2024-11-23 by kartben