zephyrproject-rtos/zephyr #81582: Blobs: Add support to pull in NXP MCXW71 BLE LL and IEEE 802.15.4 PHY combo NBU binary blob

George-Stefan, 2024-11-19:
## Origin

The following binary blob is being added: `mcxw71/mcxw71_nbu_ble_15_4_dyn.sb3`

## Type

- [ ] Precompiled library
- [x] Firmware image

## Module

hal_nxp: https://github.com/zephyrproject-rtos/hal_nxp

## Purpose

Precompiled firmware (binary blob) supporting BLE LL and IEEE 802.15.4 PHY on NXP MCXW71 SoC.

## Pull Request

HAL: https://github.com/zephyrproject-rtos/hal_nxp/pull/470

## Dependencies

None

## License

https://github.com/zephyrproject-rtos/hal_nxp/blob/master/zephyr/blobs/license/LA_OPT_NXP_Software_License.txt

dleach02, 2024-12-04:
TSC approved 12/04/2024

andrei-menzopol, 2025-02-04:
https://github.com/zephyrproject-rtos/zephyr/pull/83195

- add mcxw ieee802154 and entropy drivers + all necessary changes

https://github.com/zephyrproject-rtos/hal_nxp/pull/491

- add sdk files and changes

closed 2025-02-14 by George-Stefan