zephyrproject-rtos/zephyr #81616: Page Fault in Zephyr OS HTTP Server Simulation After Running curl Command

learnerr2024, 2024-11-19:
I successfully built my Zephyr project using the following commands
west build -b qemu_x86  
west build -t run  

After running this command: curl -v http://192.0.2.1:8080 
I am getting this error:
``` exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build v4.0.0-rc1-22-g14478cafa6a4 ***
[00:00:00.020,000] <inf> net_config: Initializing network
[00:00:00.020,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.030,000] <inf> smart_bulb: Smart Bulb HTTP Service is Starting...
[00:00:00.030,000] <dbg> net_sock: zsock_socket_internal: (http_server_tid): socket: ctx=0x150620, fd=4
[00:00:00.030,000] <inf> smart_bulb: Service is running on port: 8080
[00:00:01.020,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.0.2.2 to 192.0.2.1
[00:00:03.140,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.0.2.2 to 192.0.2.1
[00:00:03.150,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.0.2.2 to 192.0.2.1
[00:00:03.150,000] <dbg> net_sock: zsock_accepted_cb: (rx_q[0]): parent=0x150620, ctx=0x1506f0, st=0
[00:00:03.150,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.0.2.2 to 192.0.2.1
[00:00:03.150,000] <dbg> net_sock: zsock_received_cb: (rx_q[0]): ctx=0x1506f0, pkt=0x16c3b4, st=0, user_data=0
[00:00:03.150,000] <dbg> net_sock: zsock_accept_ctx: (http_server_tid): accept: ctx=0x1506f0, fd=5
[00:00:03.160,000] <err> os: Page fault at address 0x13b768 (error code 0x3)
[00:00:03.160,000] <err> os: Access violation: supervisor thread not allowed to write
[00:00:03.160,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%sE: %p -> 0x%016llx: %s" argument:1
[00:00:03.160,000] <err> os: PTE: 0x13b000 -> 0x000000000013b000: US A XD
[00:00:03.160,000] <err> os: EAX: 0x0013b760, EBX: 0x00152dd0, ECX: 0x00000001, EDX: 0x00172428
[00:00:03.160,000] <err> os: ESI: 0x0013b760, EDI: 0x0000004e, EBP: 0x001724e8, ESP: 0x00172418
[00:00:03.160,000] <err> os: EFLAGS: 0x00000206 CS: 0x0008 CR3: 0x0014e000
[00:00:03.160,000] <err> os: EIP: 0x0012e0ad
[00:00:03.160,000] <err> os: call trace:
[00:00:03.160,000] <err> os:      0: 0x0012d317
[00:00:03.160,000] <err> os:      1: 0x00101d5b
[00:00:03.160,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:03.160,000] <err> os: Current thread: 0x152ce0 (http_server_tid)
[00:00:03.170,000] <err> os: Halting system 
```

Here are the files  main.c
 ``` #include <zephyr/kernel.h>
#include <zephyr/net/http/server.h>
#include <zephyr/net/http/service.h>
#include <zephyr/logging/log.h>
#include <string.h>
#include <zephyr/sys/util.h>
#include <zephyr/sys/mutex.h>
//#include <zephyr/sys/malloc.h>
LOG_MODULE_REGISTER(smart_bulbht, LOG_LEVEL_DBG);
/* Initial bulb state */
static char bulb_state[4] = "off"; // Simulate state ("on"/"off")
static struct k_mutex bulb_state_mutex; // Mutex for safe state access
/* Define the port */
static uint16_t http_service_port = 8080;
/* Helper function to allocate response */
static char* allocate_response(const char *msg) {
    size_t len = strlen(msg) + 1; // Include null terminator
    char *response = k_malloc(len);
    if (response == NULL) {
        LOG_ERR("Failed to allocate memory for response");
        return NULL;
    }
    strcpy(response, msg);
    return response;
}
/* GET / handler: Welcome message */
static int get_root_handler(struct http_client_ctx *client,
                            enum http_data_status status,
                            uint8_t *buffer, size_t len,
                            struct http_response_ctx *response_ctx,
                            void *user_data)
{
    LOG_INF("GET / called");
    const char *message = "Welcome to the Smart Bulb HTTP Server!";
    char *response = allocate_response(message);
    if (response == NULL) {
        return -ENOMEM;
    }
    response_ctx->body = response;
    response_ctx->body_len = strlen(response);
    response_ctx->final_chunk = true;
    int ret = 0;
    k_free(response);  // Explicitly free memory
    return ret;
}
/* GET /state handler: Returns the current state of the bulb */
static int get_state_handler(struct http_client_ctx *client,
                             enum http_data_status status,
                             uint8_t *buffer, size_t len,
                             struct http_response_ctx *response_ctx,
                             void *user_data)
{
    LOG_INF("GET /state called");
    char *response = k_malloc(4); // Allocate memory for state response
    if (response == NULL) {
        LOG_ERR("Failed to allocate memory for bulb state");
        return -ENOMEM;
    }
    k_mutex_lock(&bulb_state_mutex, K_FOREVER);
    strcpy(response, bulb_state); // Copy the current state
    k_mutex_unlock(&bulb_state_mutex);
   response_ctx->body = response;
    response_ctx->body_len = strlen(response);
    response_ctx->final_chunk = true;
    int ret = 0;
    k_free(response);  // Explicitly free memory
    return ret;
}
/* POST /state handler: Toggles the bulb state */
static int post_state_handler(struct http_client_ctx *client,
                              enum http_data_status status,
                              uint8_t *buffer, size_t len,
                              struct http_response_ctx *response_ctx,
                              void *user_data)
    LOG_INF("POST /state called");
    char *response = k_malloc(4); // Allocate memory for state response
    if (response == NULL) {
        LOG_ERR("Failed to allocate memory for bulb state");
        return -ENOMEM;
    }
    k_mutex_lock(&bulb_state_mutex, K_FOREVER);
    if (strcmp(bulb_state, "off") == 0) {
        strcpy(bulb_state, "on");
    } else {
        strcpy(bulb_state, "off");
    }
    strcpy(response, bulb_state); // Return the new state
    k_mutex_unlock(&bulb_state_mutex);
    LOG_INF("Bulb state changed to: %s", bulb_state);
    response_ctx->body = response;
    response_ctx->body_len = strlen(response);
    response_ctx->final_chunk = true;
    int ret = 0;
    k_free(response);  // Explicitly free memory
    return ret;
}
/* Define the HTTP service */
HTTP_SERVICE_DEFINE(test_http_service,
                    "0.0.0.0",
                    &http_service_port,
                    1,
                    10,
                    NULL);
/* Define Resource Details */
static const struct http_resource_detail_dynamic get_root_detail = {
    .common = {
        .type = HTTP_RESOURCE_TYPE_DYNAMIC,
        .bitmask_of_supported_http_methods = BIT(HTTP_GET),
    },
    .cb = get_root_handler,
};
static const struct http_resource_detail_dynamic get_state_detail = {
    .common = {
        .type = HTTP_RESOURCE_TYPE_DYNAMIC,
        .bitmask_of_supported_http_methods = BIT(HTTP_GET),
    },
    .cb = get_state_handler,
};
static const struct http_resource_detail_dynamic post_state_detail = {
    .common = {
        .type = HTTP_RESOURCE_TYPE_DYNAMIC,
        .bitmask_of_supported_http_methods = BIT(HTTP_POST),
    },
    .cb = post_state_handler,
};
/* Resource Definitions */
HTTP_RESOURCE_DEFINE(get_root_resource, test_http_service, "/", &get_root_detail);
HTTP_RESOURCE_DEFINE(get_state_resource, test_http_service, "/state", &get_state_detail);
HTTP_RESOURCE_DEFINE(post_state_resource, test_http_service, "/state", &post_state_detail);

int main(void)
{
    LOG_INF("Smart Bulb HTTP Service is Starting...");
    k_mutex_init(&bulb_state_mutex); // Initialize mutex
    int ret = http_server_start();
    if (ret < 0) {
        LOG_ERR("HTTP Service Initialization Error: %d", ret);
        return ret;
    }
   LOG_INF("Service is running on port: %d", http_service_port);
    while (1) {
        k_sleep(K_SECONDS(5));  // Keep the server alive
    }
    return 0;
} 
```

learnerr2024, 2024-11-19:
prj.conf
```
# General config
CONFIG_MAIN_STACK_SIZE=3072
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_SHELL=y
CONFIG_LOG=y
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y
CONFIG_ZVFS_OPEN_MAX=32
CONFIG_POSIX_API=y
CONFIG_FDTABLE=y
CONFIG_ZVFS_POLL_MAX=32
# Eventfd
CONFIG_EVENTFD=y
# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_SHELL=y
CONFIG_NET_STATISTICS=y
CONFIG_NET_STATISTICS_USER_API=y
CONFIG_NET_LOG=y
# HTTP parser
CONFIG_HTTP_PARSER_URL=y
CONFIG_HTTP_PARSER=y
CONFIG_HTTP_SERVER=y
# Network buffers
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=128
CONFIG_NET_BUF_TX_COUNT=128
CONFIG_NET_CONTEXT_NET_PKT_POOL=y
# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=32
CONFIG_NET_MAX_CONN=32
# Network address config
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV6=n
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
#CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
#CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_TCP_TIME_WAIT_DELAY=0
# Network debug config
CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y
CONFIG_NET_HTTP_LOG_LEVEL_DBG=y
CONFIG_NET_IPV4_LOG_LEVEL_DBG=y
````

jukkar, 2024-11-20:
Could you compile your app for `native_sim` and then run it under valgrind. It will tell immediately where the crash is.
So `valgrind build/zephyr/zephyr.exe`

learnerr2024, 2024-11-20:
output of valgrind build/zephyr/zephyr.exe

```   valgrind build_native_sim/zephyr/zephyr.exe
==14418== Memcheck, a memory error detector
==14418== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==14418== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==14418== Command: build_native_sim/zephyr/zephyr.exe
==14418==

valgrind:  Fatal error at startup: a function redirection
valgrind:  which is mandatory for this platform-tool combination
valgrind:  cannot be set up.  Details of the redirection are:
valgrind:  
valgrind:  A must-be-redirected function
valgrind:  whose name matches the pattern:      strlen
valgrind:  in an object with soname matching:   ld-linux.so.2
valgrind:  was not found whilst processing
valgrind:  symbols from the object with soname: ld-linux.so.2
valgrind:  
valgrind:  Possible fixes: (1, short term): install glibc's debuginfo
valgrind:  package on this machine.  (2, longer term): ask the packagers
valgrind:  for your Linux distribution to please in future ship a non-
valgrind:  stripped ld.so (or whatever the dynamic linker .so is called)
valgrind:  that exports the above-named function using the standard
valgrind:  calling conventions for this platform.  The package you need
valgrind:  to install for fix (1) is called
valgrind:  
valgrind:    On Debian, Ubuntu:                 libc6-dbg
valgrind:    On SuSE, openSuSE, Fedora, RHEL:   glibc-debuginfo
valgrind:  
valgrind:  Note that if you are debugging a 32 bit process on a
valgrind:  64 bit system, you will need a corresponding 32 bit debuginfo
valgrind:  package (e.g. libc6-dbg:i386).
valgrind:  
valgrind:  Cannot continue -- exiting now.  Sorry.
```

jukkar, 2024-11-20:
There does not seem to be any error in the latest log.

jukkar, 2024-11-21:
Have you created the zeth before running zephyr.exe? You need to run net-setup.sh script from net-tools repo. This is clearly described in documentation.

learnerr2024, 2024-11-21:
yes, i followed https://docs.zephyrproject.org/latest/connectivity/networking/qemu_setup.html#networking-with-qemu
before running zephyr.exe

jukkar, 2024-11-21:
You need to follow https://docs.zephyrproject.org/latest/connectivity/networking/native_sim_setup.html#networking-with-native-sim if you are using native-sim

learnerr2024, 2024-11-21:
No, I am using qemu_x86 board, and i followed the tutorial for networking https://docs.zephyrproject.org/latest/connectivity/networking/qemu_setup.html#networking-with-qemu
still facing the page fault error after using curl command.

jukkar, 2024-11-21:
I thought you used native-sim. Well, if using qemu then valgrind cannot be used.

learnerr2024, 2024-11-21:
can you please tell me how can i resolve page fault error while using qemu. I have already shared all the files of my project if anything else needed I will provide it.

learnerr2024, 2024-11-21:
Hello there!
any solution/suggestion for this error

jukkar, 2024-11-22:
I am sorry but I will not debug out of tree code, there just isn't enough time for that. If you find a bug in the existing network samples, then that is a different matter. If you need to use qemu, then you can attach remote debugger to it to find out the issues, there are documentation in zephyr and in internet how to do it. Normally you could use native-sim target too for the applications, unless you have a specific driver you must use and which only works in qemu. With native-sim, valgrind is a great asset to quickly spot issues in the code.

rlubos, 2024-11-22:
I don't see a linker section for the resources defined in your project, could be the case. It's described here:
https://docs.zephyrproject.org/latest/connectivity/networking/api/http_server.html#server-setup
And here how it's done in the sample code:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/http_server/sections-rom.ld
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/http_server/CMakeLists.txt#L34

Also, likely not the crash reason, but you shouldn't define two handlers for a single resource:
```
HTTP_RESOURCE_DEFINE(get_state_resource, test_http_service, "/state", &get_state_detail);
HTTP_RESOURCE_DEFINE(post_state_resource, test_http_service, "/state", &post_state_detail);
```
The server code matches resources based on the resource path only. You should define a single handler, with a bitmask of supported methods (`BIT(HTTP_GET) | BIT(HTTP_POST)`), and dispatch method handling in the registered callback.