zephyrproject-rtos/zephyr #81624: Integrate Realtek RTS5816 HAL module

RtkFP, 2024-11-20:
## Origin

https://github.com/RtkFP/rtkfp_lib

## Purpose

Provide HAL support for RTS5816 SoCs of Realtek Fingerprint family.

## Mode of integration

Mode of integration: Module
Module Name: hal_realtek

## Maintainership

@RtkFP

## Pull Request

N/A

## Description

The module provides driver abstraction layers to support Realtek RTS5816 SoCs. The module includes a library to support USB device function of RTS5816.

## Dependencies

This module does not have a direct dependency on Zephyr.

## Revision

4f560d70dd181b9ae073716289946539f7a2150c

## License

Apache-2.0

nordicjm, 2024-11-20:
> The module includes a library to support USB device function of RTS5816.

Not allowed in HALs or zephyr hosted repos

RtkFP, 2024-11-20:
> > The module includes a library to support USB device function of RTS5816.
> 
> Not allowed in HALs or zephyr hosted repos

Sorry for that I did't express it clearly. 
The module repo won't host the library blobs, the  blobs are fetched by west command. The library blob will be fetched from https://github.com/RtkFP/rtkfp_lib 
I need a module repo to record the blobs information. 
Thanks

nordicjm, 2024-11-20:
Can you edit your main post and replace it with the correct template and fill it in? https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/008_bin-blobs.md?plain=1

RtkFP, 2024-11-20:
> Can you edit your main post and replace it with the correct template and fill it in? https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/008_bin-blobs.md?plain=1

I have edit the main post, but I don't know how to write the **Module** section, because the module is not exists yet. 
What I write in the **Module** section is OK, or should I edit it again?
Thanks.

nashif, 2024-11-20:
so this is a request to include blobs in a module that still does not exist? Is this related to https://github.com/zephyrproject-rtos/zephyr/issues/78084?

RtkFP, 2024-11-21:
> so this is a request to include blobs in a module that still does not exist?

Yes.

> Is this related to #78084?

It is not realted to #78084. It is a different  from Realtek.
Maybe I can include blobs in the module created in #78084

RtkFP, 2024-11-25:
Sorry to bother.
Please tell me how you will deal with this request and the schedule, and are there any other things I should offer? 
Thanks.

thughes, 2024-12-02:
cc @niedzwiecki-dawid @duda-patryk

RtkFP, 2025-02-10:
Please help to check the issue. Or are there any suggestions about the issue?
Thanks.

closed 2025-04-26 by github-actions[bot]