zephyrproject-rtos/zephyr #81639: build of sample psa/its fails

benediktibk, 2024-11-20:
**Describe the bug**
Building the sample psa/its fails because a git repo is not available.

**To Reproduce**
west build --build-dir build/tfm --board mps3 zephyr/samples/psa/its/

**Expected behavior**
successful build

**Impact**
CI is broken:

- #81560
- #81212

**Logs and console output**

    [1/9] Creating directories for 'ethos_u_core_driver-populate'
    [1/9] Performing download step (git clone) for 'ethos_u_core_driver-populate'
    Cloning into 'ethos_u_core_driver-src'...
    fatal: unable to access 'https://review.mlplatform.org/ml/ethos-u/ethos-u-core-driver/': The requested URL returned error: 502
    Cloning into 'ethos_u_core_driver-src'...
    fatal: unable to access 'https://review.mlplatform.org/ml/ethos-u/ethos-u-core-driver/': The requested URL returned error: 502
    Cloning into 'ethos_u_core_driver-src'...
    fatal: unable to access 'https://review.mlplatform.org/ml/ethos-u/ethos-u-core-driver/': The requested URL returned error: 502
    -- Had to git clone more than once: 3 times.
    CMake Error at ethos_u_core_driver-subbuild/ethos_u_core_driver-populate-prefix/tmp/ethos_u_core_driver-populate-gitclone.cmake:39 (message):
      Failed to clone repository:
      'https://review.mlplatform.org/ml/ethos-u/ethos-u-core-driver'

**Environment (please complete the following information):**

Zephyr version 3364a35f05475a947dfd0bfc4f7b8984504caee4

benediktibk, 2024-11-20:
Also affected:

- #81633
- #81596

benediktibk, 2024-11-20:
@ithinuel do you have an idea on how to proceed with this issue?

benediktibk, 2024-11-20:
@laurenmurphyx64 @kristofer-jonsson-arm Could you maybe help us out here? This issue is blocking already a bunch of PRs.

jhedberg, 2024-11-20:
Seems like this might be a duplicate of #81656?

jhedberg, 2024-11-20:
> Seems like this might be a duplicate of #81656?

(or the other way around)

krijo210, 2024-11-20:
https://review.mlplatform.org is since yesterday experiencing an overload of github actions. Arm devops is to my understanding looking into the issue together with Linaro. Last I heard Linaor are restarting the service with more capacity.

benediktibk, 2024-11-20:
Sounds like this is in general a violation to pull in something externally:

> New board fetching code from external repositories in violation of zephyr rules and causing build issues in CI

https://github.com/zephyrproject-rtos/zephyr/issues/81656

Tarangraval21, 2024-11-20:
I guess issue is resolved now 
( [dts: arm: rpi_pico: remove #define from dts #81196](https://github.com/zephyrproject-rtos/zephyr/pull/81196) )

benediktibk, 2024-11-20:
Yes, @nashif has pushed 1c596b8b7771a65b46788038e66b93f7905d3ecb