zephyrproject-rtos/zephyr #81675: eth_nxp_enet should support set_config() function from ethernet api

theadib, 2024-11-20:
**Is your enhancement proposal related to a problem? Please describe.**

During runtime I want to change the MAC address using the net_if_set_link_addr() function.
The MAC address is moved to the driver structures so that newly created messages will contain that MAC address.
But the SOC MAC filter is not updated and so the device does not respond to any incoming messages.

During Init in /zephyr/drivers/ethernet/eth_nxp_enet_qos/nxp_enet_qos_mac.c the function eth_nxp_enet_qos_mac_init()  sets this SOC registers. But those are not touched by any other function.
That means one can not change from devicetree settings.

**Describe the solution you'd like**

The eth_nxp_enet driver must support the set_config api of the ethernet api.

**Describe alternatives you've considered**

**Additional context**

using FRDM_MCXN947 board.
discussed here in discord https://discord.com/channels/720317445772017664/883445396284772413/1308802268849766460

theadib, 2024-11-20:
PS my current work around is to manually update the SOC registers
```
        //Set MAC Address for frame filtering logic
        ENET0->MAC_ADDRESS0_HIGH =  mac[5] << 8 | 
                                    mac[4];
        ENET0->MAC_ADDRESS0_LOW =   mac[3] << 24 | 
                                    mac[2] << 16 | 
                                    mac[1] << 8 | 
                                    mac[0];
        memcpy(addr->addr, mac, 6);
        net_if_set_link_addr(iface, addr->addr, 6, NET_LINK_ETHERNET);
```

decsny, 2024-11-21:
@theadib have you considered submitting a PR for this, it looks like you figured out basically the code on your own, so the function implementation should be pretty straightforward, you could look at another ethernet driver for an example

theadib, 2024-11-26:
Hello @decsny here is my PR https://github.com/zephyrproject-rtos/zephyr/pull/82089 
Only solves that one issue with the MAC address.
not sure if that PR is formally correct.
I tried to be as close to https://docs.zephyrproject.org/latest/contribute/guidelines.html#
regards, Adib

mmahadevan108, 2024-12-05:
@theadib , thank you for the PR.

closed 2024-12-05 by mmahadevan108