zephyrproject-rtos/zephyr #81678: Cellular modem sample using Boron - Time outs

zpm1066, 2024-11-20:
**Describe the bug**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: macOS Sonoma 14.7.1
 - Toolchain: Zephyr SDK
 - Zephyr v3.7.0

bjarki-andreasen, 2024-11-22:
Could you try disabling the at shell? it seems the chat script fails after trying to open the CMUX DLCI channel used for it :)

zpm1066, 2024-11-25:
@bjarki-andreasen Thanks for the response.

After disabling AT shell, I still get errors as below.

It seems a few other folks have had the same  "Dropping byte 0x0d because 0x7e was expected" error using CMUX.

Curious to see if anyone else had success with getting Boron to connect successfully to a LTE-M network.

Do you have a Boron? If so, Have you tried the sample with a Boron device? 
Do you know if anyone else has had success with the Boron?

Any further suggestions would be welcomed. Thanks.

```
AT+CGDCONT=1,"IP","internet"
OK

AT+CFUN=1
OK

ATD*99***1#
<modem_ppp: modem_ppp_is_byte_expected: Dropping byte 0x0d because 0x7e was expected.>

```

```
00> rtt:~$ *** Booting Zephyr OS build v3.7.0-4713-g091c6664c51b ***
00> rtt:~$ Powering on modem
00> rtt:~$ Bring up network interface
00> rtt:~$ Waiting for L4 connected
00> rtt:~$ rtt:~$ [00:00:11.863,739] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_init_chat_script
00> rtt:~$ [00:00:11.863,769] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 0
00> rtt:~$ [00:00:11.863,800] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:11.963,958] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 1
00> rtt:~$ [00:00:11.963,989] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:12.064,147] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 2
00> rtt:~$ [00:00:12.064,178] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:12.164,337] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 3
00> rtt:~$ [00:00:12.164,367] <dbg> modem sending: AT
00> [00:00:21.863,861] <wrn> modem_chat: u_blox_sara_r4_init_chat_script: timed out
00> rtt:~$ rtt:~$ [00:00:33.364,166] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_init_chat_script
00> rtt:~$ [00:00:33.364,196] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 0
00> rtt:~$ [00:00:33.364,227] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.464,385] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 1
00> rtt:~$ [00:00:33.464,416] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.564,544] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 2
00> rtt:~$ [00:00:33.564,575] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.664,703] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 3
00> rtt:~$ [00:00:33.664,733] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.764,862] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 4
00> rtt:~$ [00:00:33.764,923] <dbg> modem_chat: modem_chat_script_next: sending: ATE0
00> rtt:~$ rtt:~$ [00:00:43.364,318] <wrn> modem_chat: u_blox_sara_r4_init_chat_script: timed out
00> rtt:~$ rtt:~$ [00:00:54.864,624] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_init_chat_script
00> rtt:~$ [00:00:54.864,654] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 0
00> rtt:~$ [00:00:54.864,685] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:54.889,343] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:54.889,373] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:54.964,843] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 1
00> rtt:~$ [00:00:54.964,874] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:54.989,013] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:54.989,044] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:55.065,002] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 2
00> rtt:~$ [00:00:55.065,063] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ rtt:~$ [00:00:55.089,141] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:55.089,172] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:55.165,161] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 3
00> rtt:~$ [00:00:55.165,191] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:55.189,270] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:55.189,300] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:55.265,319] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 4
[... 81 lines cut ...]
00> rtt:~$ rtt:~$ [00:01:01.947,265] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
00> rtt:~$ rtt:~$ [00:01:03.947,448] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
00> rtt:~$ [00:01:03.947,479] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
00> rtt:~$ [00:01:03.947,509] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
00> rtt:~$ [00:01:03.972,991] <wrn> modem_cmux: Frame FCS error
00> rtt:~$ [00:01:03.972,991] <wrn> modem_cmux: Dropped frame
00> rtt:~$ rtt:~$ [00:01:07.947,570] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
[... cut ...]
00> rtt:~$ rtt:~$ [00:01:19.948,181] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
00> rtt:~$ rtt:~$ [00:01:21.948,364] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
00> rtt:~$ [00:01:21.948,394] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
00> rtt:~$ [00:01:21.948,425] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
00> rtt:~$ rtt:~$ [00:01:25.948,486] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
00> rtt:~$ rtt:~$ 
```

zpm1066, 2024-11-25:
@WilliamGFish 
I'm back again to see if I can get a Boron working with Zephyr for LTE-M connectivity. The cellular modem sample runs into the issue(s) documented above.

Any suggestions what may be the problem? Thanks.

zpm1066, 2024-11-25:
@bjarki-andreasen  Thanks.
I'm not that familiar with the modem_chat_script, just recently started to review the "modem_celluar" source.

What's the command to update the u_blox_sara_r4_init_chat_script_cmds script to wait for "CONNECTED 1500000\r\n"? Thanks.

bjarki-andreasen, 2024-11-25:
Heres another script waiting on connect, adjust to fit your needs :)

https://github.com/zephyrproject-rtos/zephyr/blob/305ceba9204d9a7ee4bae0a6103ef4cbd9a1c54e/drivers/modem/modem_cellular.c#L2306-L2312

WilliamGFish, 2024-11-25:
I'll have a look when I'm back in the office... Often it's timing with initialization or the custom firmware that particle used.Sent via mobile device, apologies for brevity and/or typos.

zpm1066, 2024-11-25:
@bjarki-andreasen 
I added "connect_match"and  "waiting on connect" for the Sara R4 modem. I tried a few different delays but the script still times out.

```
00> rtt:~$ [00:00:55.855,926] <dbg> modem_chat: modem_chat_log_received_command: OK
00> rtt:~$ [00:00:55.855,957] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_dial_chat_script: step: 1
00> rtt:~$ [00:00:55.855,987] <dbg> modem_chat: modem_chat_script_next: sending: AT+CGDCONT=1,"IP","internet"
00> rtt:~$ [00:00:55.905,914] <dbg> modem_chat: modem_chat_log_received_command: OK
00> rtt:~$ [00:00:55.905,944] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_dial_chat_script: step: 2
00> rtt:~$ [00:00:55.906,005] <dbg> modem_chat: modem_chat_script_next: sending: AT+CFUN=1
00> rtt:~$ rtt:~$ [00:00:55.937,194] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:56.906,158] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_dial_chat_script: step: 3
00> rtt:~$ [00:00:56.906,188] <dbg> modem_chat: modem_chat_script_next: sending: ATD*99***1#
00> rtt:~$ rtt:~$ [00:01:05.828,552] <wrn> modem_chat: u_blox_sara_r4_dial_chat_script: timed out
00> rtt:~$ rtt:~$ L4 was not connected in time

```

WilliamGFish, 2024-11-28:
OK, not completely to grips with how this all works but, you may need to adjust the timeout in this:
```
MODEM_CHAT_SCRIPT_DEFINE(u_blox_sara_r4_dial_chat_script, u_blox_sara_r4_dial_chat_script_cmds,
                   dial_abort_matches, modem_cellular_chat_callback_handler, **20**);
```

I think there will be issues the first time you register onto an NB-IoT or LTE network as this could take 5-10 mins. It might be worth running a basic Sara modem app to get initially connected.

Addiotnally, I ran a couple of other commands just to see if the network was ok:
```
MODEM_CHAT_SCRIPT_CMD_RESP_MULT("AT+CGACT=0,1", allow_match),
MODEM_CHAT_SCRIPT_CMD_RESP("AT+CGDCONT=1,\"IP\","
            "\""CONFIG_MODEM_CELLULAR_APN"\"",
            ok_match),
MODEM_CHAT_SCRIPT_CMD_RESP("AT+CFUN=1", ok_match),
MODEM_CHAT_SCRIPT_CMD_RESP("AT+URAT?", ok_match),
MODEM_CHAT_SCRIPT_CMD_RESP("AT+CESQ", ok_match),
MODEM_CHAT_SCRIPT_CMD_RESP("ATD*99***1#", connect_match));
```

> 00:00:55.832,061] \033[0m<dbg> modem_chat: modem_chat_script_next: sending: AT+URAT?\033[0m
> [00:00:55.857,666] \033[0m<dbg> modem_chat: modem_chat_on_unknown_command_received: +URAT: 7,8\033[0m
> [00:00:55.857,727] \033[0m<dbg> modem_chat: modem_chat_log_received_command: OK\033[0m
> [00:00:55.857,727] \033[0m<dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_dial_chat_script: step: 4\033[0m
> [00:00:55.857,757] \033[0m<dbg> modem_chat: modem_chat_script_next: sending: AT+CESQ\033[0m
> [00:00:55.882,293] \033[0m<dbg> modem_chat: modem_chat_on_unknown_command_received: +CESQ: 99,99,255,255,255,255\033[0m

As you can see I do not have a working network where I am so can help much more.

The other aspect I have noticed is there are a number of hardware aspects that are not included in the start-up sequence which will have a bearing on the initialisation stability.

zpm1066, 2024-11-28:
@WilliamGFish Thanks. What’s a reasonable timeout worked for you?  I’ll run some tests this evening.

`> MODEM_CHAT_SCRIPT_DEFINE(u_blox_sara_r4_dial_chat_script, u_blox_sara_r4_dial_chat_script_cmds,                    dial_abort_matches, modem_cellular_chat_callback_handler, **20**);`

klarasrl, 2024-11-29:
Hello !
Not sure if that helps but I'm currently struggling with the same issue, with an nRF5340dk communicating through UART with a SARA-R422.

I'd already modify the modem_celular.c file, to make sure the command response to `ATD*99***1#` was a connect_match, and increased the timeout of the periodic_chat_script but it doesn't change anything.
It seems to me that this part of the script sometimes runs succesfully 2 or 3 times (sometimes only once, since the last answer received from the module is the connect match), then times out, and the modem_cmux module is outputting a Frame FCS error after that. Then the only way to get a new response from the Sara is to restart the whole software.

```

[00:00:13.260,040] <dbg> modem_chat: modem_chat_log_received_command: CONNECT  150000000
[00:00:13.260,070] <dbg> modem_chat: modem_chat_script_stop: u_blox_sara_r4_dial_chat_script: complete
[00:00:13.260,192] <dbg> modem_cellular: modem_cellular_log_event: event script success
[00:00:13.260,314] <dbg> modem_cellular: modem_cellular_log_state_changed: switch from run dial script to await registered
[00:00:13.283,447] <dbg> modem_cmux: modem_cmux_log_frame: rcvd ch:0 cr:0 pf:0 type:UIH dlen:4
uart:~$ [00:00:13.283,477] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     e3 05 07 8d                                      |....
[00:00:13.283,508] <dbg> modem_cmux: modem_cmux_log_received_command: ea:1,cr:1,type:MSC
[00:00:13.283,538] <dbg> modem_cmux: modem_cmux_log_received_command: data:
                                     07 8d                                            |..
[00:00:13.283,569] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:0 cr:0 pf:0 type:UIH dlen:4
[00:00:13.283,599] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     e1 05 07 8d                                      |....
[00:00:13.283,630] <dbg> modem_cmux: modem_cmux_log_transmit_command: ea:1,cr:0,type:MSC
[00:00:13.283,660] <dbg> modem_cmux: modem_cmux_log_transmit_command: data:
                                     07 8d                                            |..
[00:00:15.260,345] <dbg> modem_cellular: modem_cellular_log_event: event timeout
[00:00:15.260,406] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
[00:00:15.260,437] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
[00:00:15.260,467] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
[00:00:15.260,559] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:2 cr:1 pf:0 type:UIH dlen:8
[00:00:15.260,589] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     41 54 2b 43 52 45 47 3f                          |AT+CREG?
[00:00:13.283,477] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     e3 05 07 8d                                      |....
[00:00:13.283,508] <dbg> modem_cmux: modem_cmux_log_received_command: ea:1,cr:1,type:MSC
[00:00:13.283,538] <dbg> modem_cmux: modem_cmux_log_received_command: data:
                                     07 8d                                            |..
[00:00:13.283,569] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:0 cr:0 pf:0 type:UIH dlen:4
[00:00:13.283,599] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     e1 05 07 8d                                      |....
[00:00:13.283,630] <dbg> modem_cmux: modem_cmux_log_transmit_command: ea:1,cr:0,type:MSC
[00:00:13.283,660] <dbg> modem_cmux: modem_cmux_log_transmit_command: data:
                                     07 8d                                            |..
[00:00:15.260,345] <dbg> modem_cellular: modem_cellular_log_event: event timeout
[00:00:15.260,406] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
[00:00:15.260,437] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
[00:00:15.260,467] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
[00:00:15.260,559] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:2 cr:1 pf:0 type:UIH dlen:8
[00:00:15.260,589] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     41 54 2b 43 52 45 47 3f                          |AT+CREG?
uart:~$ [00:00:15.260,650] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:2 cr:1 pf:0 type:UIH dlen:1
[00:00:15.260,681] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     0d                                               |.
[00:00:15.260,650] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:2 cr:1 pf:0 type:UIH dlen:1
[00:00:15.260,681] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     0d                                               |.
uart:~$ [00:00:20.260,620] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
[00:00:20.260,711] <dbg> modem_cellular: modem_cellular_log_event: event script failed
[00:00:20.260,620] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
[00:00:20.260,711] <dbg> modem_cellular: modem_cellular_log_event: event script failed
uart:~$ [00:00:22.260,864] <dbg> modem_cellular: modem_cellular_log_event: event timeout
[00:00:22.260,955] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
[00:00:22.260,955] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
[... cut ...]
[00:00:22.261,108] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     41 54 2b 43 52 45 47 3f                          |AT+CREG?
[00:00:22.261,169] <dbg> modem_cmux: modem_cmux_log_frame: tx ch:2 cr:1 pf:0 type:UIH dlen:1
[00:00:22.261,199] <dbg> modem_cmux: modem_cmux_log_frame: data:
                                     0d                                               |.
[00:00:22.289,154] <wrn> modem_cmux: Frame FCS error
[00:00:22.289,154] <wrn> modem_cmux: Dropped frame
```

zpm1066, 2024-11-29:
> I'd already modified the modem_celular.c file, to make sure the command response to `ATD*99***1#` was a connect_match, and increased the timeout of the periodic_chat_script but it doesn't change anything.

I’ve changed a few timeouts but I’m still struggling with making any progress.

What timeout did you use for the periodic_chat_script? 
Do you mind sharing the modem chat script that you’re using so that I can compare it against mine. Hopefully can then see what I’m doing incorrectly. Thank you.

klarasrl, 2024-11-29:
Only things I've changed in the `modem_cellular.c` file are : 

```
#if DT_HAS_COMPAT_STATUS_OKAY(swir_hl7800) || DT_HAS_COMPAT_STATUS_OKAY(sqn_gm02s) || DT_HAS_COMPAT_STATUS_OKAY(u_blox_sara_r4)
MODEM_CHAT_MATCH_DEFINE(connect_match, "CONNECT", "", NULL);
#endif
```

```
MODEM_CHAT_SCRIPT_CMDS_DEFINE(u_blox_sara_r4_dial_chat_script_cmds,
			      MODEM_CHAT_SCRIPT_CMD_RESP_MULT("AT+CGACT=0,1", allow_match),
			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CGDCONT=1,\"IP\","
							 "\""CONFIG_MODEM_CELLULAR_APN"\"",
							 ok_match),
			      MODEM_CHAT_SCRIPT_CMD_RESP("AT+CFUN=1", ok_match),
			      MODEM_CHAT_SCRIPT_CMD_RESP("ATD*99***1#", connect_match));
```

```
MODEM_CHAT_SCRIPT_DEFINE(u_blox_sara_r4_periodic_chat_script,
			 u_blox_sara_r4_periodic_chat_script_cmds, abort_matches,
			 modem_cellular_chat_callback_handler, 10);
```

zpm1066, 2024-11-29:
@klarasrl Thanks.

As you can see below, I'm running Zephyr v3.7.0.  Which version of Zephyr are you running?

I now, for the first time, do get "CONNECT  150000000" but afterwards get the "Frame FCS error" and "Dropped frame", same as you.

@bjarki-andreasen 
Any suggestions how we can get around the dropped frame errors? Thanks.

```
00> rtt:~$ [00:00:55.990,661] <dbg> modem_chat: modem_chat_script_next: sending: ATD*99***1#
00> rtt:~$ [00:00:56.018,707] <dbg> modem_chat: modem_chat_log_received_command: CONNECT  150000000
00> rtt:~$ [00:00:56.018,737] <dbg> modem_chat: modem_chat_script_stop: u_blox_sara_r4_dial_chat_script: complete

```
...
...

```
00> rtt:~$ [00:01:10.019,287] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
00> rtt:~$ [00:01:10.019,317] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
00> rtt:~$ [00:01:10.043,609] <wrn> modem_cmux: Frame FCS error
00> rtt:~$ [00:01:10.043,609] <wrn> modem_cmux: Dropped frame
```

Here is the full modem output:

```
00> rtt:~$ *** Booting Zephyr OS build v3.7.0-4713-g091c6664c51b ***
00> rtt:~$ Powering on modem
00> rtt:~$ Bring up network interface
00> rtt:~$ Waiting for L4 connected
00> rtt:~$ rtt:~$ [00:00:11.857,452] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_init_chat_script
00> rtt:~$ [00:00:11.857,482] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 0
00> rtt:~$ [00:00:11.857,543] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:11.957,672] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 1
00> rtt:~$ [00:00:11.957,702] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:12.057,830] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 2
00> rtt:~$ [00:00:12.057,861] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:12.157,989] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 3
00> rtt:~$ [00:00:12.158,020] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:12.258,148] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 4
00> rtt:~$ [00:00:12.258,209] <dbg> modem_chat: modem_chat_script_next: sending: ATE0
00> rtt:~$ rtt:~$ [00:00:21.857,604] <wrn> modem_chat: u_blox_sara_r4_init_chat_script: timed out
00> rtt:~$ rtt:~$ [00:00:33.357,910] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_init_chat_script
00> rtt:~$ [00:00:33.357,971] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 0
00> rtt:~$ [00:00:33.358,001] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.381,866] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:33.381,896] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:33.458,129] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 1
00> rtt:~$ [00:00:33.458,160] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.481,140] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:33.481,170] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:33.558,288] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 2
00> rtt:~$ [00:00:33.558,349] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ rtt:~$ [00:00:33.581,298] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:33.581,359] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:33.658,447] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 3
00> rtt:~$ [00:00:33.658,477] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:33.758,605] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 4
00> rtt:~$ [00:00:33.758,636] <dbg> modem_chat: modem_chat_script_next: sending: ATE0
00> rtt:~$ rtt:~$ [00:00:43.358,062] <wrn> modem_chat: u_blox_sara_r4_init_chat_script: timed out
00> rtt:~$ rtt:~$ [00:00:54.858,428] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_init_chat_script
00> rtt:~$ [00:00:54.858,489] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 0
00> rtt:~$ [00:00:54.858,520] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:54.882,324] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
00> rtt:~$ [00:00:54.882,354] <dbg> modem_chat: modem_chat_on_unknown_command_received: OK
00> rtt:~$ [00:00:54.958,648] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 1
00> rtt:~$ [00:00:54.958,679] <dbg> modem_chat: modem_chat_script_next: sending: AT
00> rtt:~$ [00:00:54.981,109] <dbg> modem_chat: modem_chat_on_unknown_command_received: AT
[... 66 lines cut ...]
00> rtt:~$ [00:00:56.018,707] <dbg> modem_chat: modem_chat_log_received_command: CONNECT  150000000
00> rtt:~$ [00:00:56.018,737] <dbg> modem_chat: modem_chat_script_stop: u_blox_sara_r4_dial_chat_script: complete
00> rtt:~$ [00:00:56.737,457] <dbg> modem_chat: modem_chat_log_received_command: +CREG:  2
00> rtt:~$ [00:00:56.737,548] <dbg> modem_chat: modem_chat_log_received_command: +CGREG:  2
00> rtt:~$ [00:00:56.737,731] <dbg> modem_chat: modem_chat_log_received_command: +CEREG:  2
[... cut ...]
00> rtt:~$ rtt:~$ [00:01:32.019,683] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
00> rtt:~$ rtt:~$ [00:01:34.019,866] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
00> rtt:~$ [00:01:34.019,897] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
00> rtt:~$ [00:01:34.019,927] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
00> rtt:~$ [00:01:34.117,950] <wrn> modem_cmux: Frame FCS error

```

klarasrl, 2024-12-02:
I'm using ncs `v2.7.0` so i guess it's Zephyr `v3.6.99`.

zpm1066, 2024-12-02:
@klarasrl Thanks. I’ll Zephyr v3.6 to see if it makes any difference. Have you had a successful LTE connection up with the SARA modem at all?

weinholt, 2024-12-03:
@zpm1066 I haven't used a SARA R4 modem with recent Zephyr versions, but it has worked before. I used it without CMUX.

Your logs at one point say "+CME ERROR: 13", which is a SIM failure, but that could be unrelated.

Getting these modems to work reliably is more difficult than it should be. Here are some things to consider (from slightly hazy memory):

- (Some/all?) R4 modems have a poorly documented feature where they break the PPP connection if they need to connect to the U-blox LwM2M server. You can disable the built-in LwM2M client with `AT+ULWM2M=1` and see if that helps. Normally it should only hijack the PDP context once a day, but you never know. There can be good reasons to want the client enabled, so stopping it is not for everyone, and IIRC there are firmware versions without this command. I also think that Particle didn't expose the USB pins for the modem so you might not be able to upgrade the modem firmware.
- If you don't clear the FPLMN list then you might end up with a SIM that doesn't want to register to any of the networks in range. Networks are added to the list when the modem is denied registration and it's not that difficult to accidentally have that happen.
- As someone else mentioned, you need to be patient when first connecting to a network. Basically if the SIM doesn't see its home network then it takes quite a while for it to give up the search and select a network for roaming. You can make that faster with appropriate `AT+URAT=` and `AT+UBANDMASK=` commands, but the first connection will be slower than subsequent ones.
- Don't issue `AT+COPS=` unnecessarily as it causes the modem to forget some state and it does the whole search process all over again.
- If PSM is activated then IIRC the serial line to the modem stops working when PSM kicks in. PSM is negotiated between the modem and the network, so if you see that the serial line stops working once you're connected, then this might be the cause. No idea how this interacts with CMUX, but probably poorly. You can query PSM with `AT+CPSM?` and disable it with `AT+CFUN=0`, `AT+CPSM=0`, `AT+CFUN=15`.

I hope that something here helps. :)

zpm1066, 2024-12-04:
Hello @weinholt , Thank you for your input. It's very much appreciated.

I looked into your suggestions but unfortunately I'm still  getting modem "not registered & still searching", frame FCS errors and dropped frames.

```
00> rtt:~$ [00:01:17.477,294] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_dial_chat_script: step: 5
00> rtt:~$ [00:01:17.477,325] <dbg> modem_chat: modem_chat_script_next: sending: ATD*99***1#
00> rtt:~$ rtt:~$ [00:01:17.504,821] <dbg> modem_chat: modem_chat_log_received_command: CONNECT  150000000
00> rtt:~$ [00:01:17.504,852] <dbg> modem_chat: modem_chat_script_stop: u_blox_sara_r4_dial_chat_script: complete
00> rtt:~$ [00:01:17.924,774] <dbg> modem_chat: modem_chat_log_received_command: +CREG:  2
00> rtt:~$ [00:01:17.924,865] <dbg> modem_chat: modem_chat_log_received_command: +CGREG:  2
00> rtt:~$ [00:01:17.925,048] <dbg> modem_chat: modem_chat_log_received_command: +CEREG:  2
00> rtt:~$ rtt:~$ [00:01:19.505,065] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
00> rtt:~$ [00:01:19.505,096] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
00> rtt:~$ [00:01:19.505,126] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
00> rtt:~$ rtt:~$ [00:01:29.505,187] <wrn> modem_chat: u_blox_sara_r4_periodic_chat_script: timed out
00> rtt:~$ rtt:~$ [00:01:31.505,371] <dbg> modem_chat: modem_chat_script_start: running script: u_blox_sara_r4_periodic_chat_script
00> rtt:~$ [00:01:31.505,401] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_periodic_chat_script: step: 0
00> rtt:~$ [00:01:31.505,432] <dbg> modem_chat: modem_chat_script_next: sending: AT+CREG?
00> rtt:~$ [00:01:31.530,578] <wrn> modem_cmux: Frame FCS error
00> rtt:~$ [00:01:31.530,578] <wrn> modem_cmux: Dropped frame
```

The AT command "AT+ULWM2M=1" didn't work. 

Regarding FPMLN, I got the following. So, the modem hasn't been denied registration and nothing in the forbidden list. 

```
00> rtt:~$ [00:00:55.402,404] <dbg> modem_chat: modem_chat_script_next: sending: AT+CRSM=176,28539,0,0,12
00> rtt:~$ [00:00:55.429,199] <dbg> modem_chat: modem_chat_on_unknown_command_received: +CRSM: 144,0,"FFFFFFFFFFFFFFFFFFFFFFFF"
00> rtt:~$ [00:00:55.429,260] <dbg> modem_chat: modem_chat_log_received_command: OK
```

Network searches yielded below. I tried giving priority to LTE-M (7).

```
00> rtt:~$ [00:00:55.517,486] <dbg> modem_chat: modem_chat_script_next: sending: AT+URAT=?
00> rtt:~$ rtt:~$ rtt:~$ [00:00:55.542,205] <dbg> modem_chat: modem_chat_on_unknown_command_received: +URAT: (7-9),(7-9)(7-9)
00> rtt:~$ [00:00:55.542,266] <dbg> modem_chat: modem_chat_log_received_command: OK

00> rtt:~$ [00:00:55.553,222] <dbg> modem_chat: modem_chat_script_next: sending: AT+UBANDMASK?
00> rtt:~$ [00:00:55.578,521] <dbg> modem_chat: modem_chat_on_unknown_command_received: +UBANDMASK: 0,2074,1,2074
00> rtt:~$ [00:00:55.578,582] <dbg> modem_chat: modem_chat_log_received_command: OK
```

There's no PSM activated.

MNO profile yielded the following:

```
00> rtt:~$ [00:00:55.523,620] <dbg> modem_chat: modem_chat_script_next: u_blox_sara_r4_init_chat_script: step: 16
00> rtt:~$ [00:00:55.523,651] <dbg> modem_chat: modem_chat_script_next: sending: AT+UMNOPROF=?
00> rtt:~$ rtt:~$ [00:00:55.557,067] <dbg> modem_chat: modem_chat_on_unknown_command_received: +UMNOPROF: 
00> rtt:~$ [00:00:55.557,189] <dbg> modem_chat: modem_chat_on_unknown_command_received: 0: SW default
00> rtt:~$ [00:00:55.557,342] <dbg> modem_chat: modem_chat_on_unknown_command_received: 1: SIM ICCID select
00> rtt:~$ [00:00:55.557,403] <dbg> modem_chat: modem_chat_on_unknown_command_received: 2: ATT
00> rtt:~$ [00:00:55.557,495] <dbg> modem_chat: modem_chat_on_unknown_command_received: 6: CT
00> rtt:~$ [00:00:55.557,586] <dbg> modem_chat: modem_chat_on_unknown_command_received: 4: Telstra
00> rtt:~$ [00:00:55.557,647] <dbg> modem_chat: modem_chat_on_unknown_command_received: 5: TMO
00> rtt:~$ [00:00:55.557,769] <dbg> modem_chat: modem_chat_on_unknown_command_received: 3: Verizon
00> rtt:~$ [00:00:55.557,800] <dbg> modem_chat: modem_chat_log_received_command: OK

```

I'll take another look at few other things today. Although I have working nRF91 modems, I certainly would like to get these Particle Borons running Zephyr for LTE-M connectivity. I've been using Zephyr 3.7.0 and 4.0 but will try 3.6, though I don't expect much difference.

zpm1066, 2024-12-04:
@WilliamGFish Thanks. Unfortunately, not much progress with the frame FCS errors and dropped frames but plan to try out a few things later today. Appreciate your assistance.

zpm1066, 2024-12-04:
@klarasrl , @WilliamGFish 
I forgot to check earlier. Are you using the Boron builtin eSIM or a physical SIM inserted in the Boron SIM holder?
Does it make any difference? Thanks.

closed 2025-02-17 by github-actions[bot]

zpm1066, 2025-02-17:
I'd like to re-open this issue, as I plan to work on it this week. Thanks.

zpm1066, 2025-07-15:
Hi @athankok  Unfortunately, I haven't pursued it since Feb. I ended up re-flashing the Boron with Particle Device OS and deploying it for another project. However, I'll take another crack at it, as you have made some good progress and are able to connect to the LTE-M network. 

 I assume you're using a Boron with the `ublox SARA R4 `modem and the internal eSIM. Correct? 

Do you mind sharing your full init sequence? I can take out one of the spare Boron and try again. Thanks.

Have you tried adding "MODEM_CHAT_SCRIPT_CMD_RESP_NONE("AT+URAT=7,0", 100) to the init script for fallback to GSM?

`ublox-sara-r4-at.pdf`  has the following syntax for +URAT.

[image]

athankok, 2025-07-16:
It is not the PSM I have confirmed that it is disabled.

weinholt, 2025-07-16:
Last idea from me: are you supplying enough power to the Boron?

athankok, 2025-07-16:
Sorry, my bad for not clearly mentioning it. As I said earlier, my setup is similar to the one mentioned by @klarasrl - nRF5340 communicating through UART with a SARA-R422.
Power is not an issue - I verified it with an oscilloscope. Handling of PWR_ON and RESET_N pins is not a problem - I verified them with an oscilloscope as well.
I am suspecting that the issue comes from the dial script. Your suggestion gave me the idea that in my init script, I am not setting cfun=0. So, is there any chance that the modem has an already active PDP before the execution of MODEM_CHAT_SCRIPT_CMD_RESP_NONE("ATD*99***1#", 0); and that's why it's a coin flip whether it's unstable or not