zephyrproject-rtos/zephyr #81687: [2.7-branch] ci: backports failing on clang workflow

cfriedt, 2024-11-21:
**Describe the bug**
At some point there was a change that broke the clang workflow for `2.7-branch`. It has not been a high priority. However, there are backports that cannot be merged until CI is green.

From what I can tell, the clang workflow simply times-out whenever it is run.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? CI
 - What have you tried to diagnose or workaround this issue? 
 - Is this a regression? Yes, but it was not a code commit that caused it. Something changed with the platform where the clang workflow runs

**To Reproduce**
Observe failing jobs in #80615 #74140 #74122

**Expected behavior**
CI should be green for the above backport PRs.

**Impact**
We shouldn't have CI failing. It's "kind of a big deal", but since LTSv2 is approaching EOL, it is not a high priority.

We should fix it before cutting the final LTSv2 release.

**Logs and console output**
See failing jobs in #80615 #74140 #74122

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 6477a75db3b1944c83eaf2e95170268a985547ff

**Additional context**

cfriedt, 2024-11-21:
@stephanosio or @nashif - can you take a look? I haven't been able to fix it in the bit of time that I've had to look at it.

nashif, 2024-12-03:
looks like we have old runner names in there, i.e.

zephyr-runner-linux-x64-4xlarge instead of zephyr-runner-v2-linux-x64-4xlarge

nashif, 2024-12-12:
should be fixed now

cfriedt, 2025-01-21:
This is still affecting the following two PRs after a rebase:

https://github.com/zephyrproject-rtos/zephyr/pull/74122
https://github.com/zephyrproject-rtos/zephyr/pull/82392

nashif, 2025-01-22:
> This is still affecting the following two PRs after a rebase:
> 
> #74122 #82392

both seem to be passing?

cfriedt, 2025-01-22:
> > This is still affecting the following two PRs after a rebase:
> > 
> > #74122 #82392
> 
> both seem to be passing?

My bad - I spoke too soon. Thought they would time out because it took a long time to get picked up.

Sorry

PR #82752 (2024-12-11): [2.7] ci: update gh action runners
Use  zephyr-runner-v2-linux-x64-4xlarge as a runner.

Fixes #81687
closed 2025-01-22 by cfriedt