zephyrproject-rtos/zephyr #81704: Incorrect Interrupt Numbers for UART4, UART5, and UART6 in STM32G0 Device Tree Files

adelfak, 2024-11-21:
Description: There is an issue in the Zephyr Device Tree configuration for STM32G0 series UART peripherals. Specifically, the interrupt numbers for UART4, UART5, and UART6 are assigned the same interrupt number (IRQ 29). This results in conflicts when using these UARTs simultaneously.

Current Issue:
USART4 in STM32G071 is incorrectly assigned IRQ 29 in STM32g071.dtsi.
USART5 and USART6 in STM32G0B1 are also assigned IRQ 29 in STM32g0b1.dtsi.
This overlap leads to issues when multiple UART peripherals are used together, causing the system to not differentiate between interrupts properly, and potentially leading to incorrect interrupt handling or system crashes.

Correct Interrupt Numbers: According to the STM32G0 reference manual, the correct interrupt numbers should be:
USART4: IRQ 30
USART5: IRQ 31
USART6: IRQ 32

Proposed Fix:
Update the interrupts values for USART4, USART5, and USART6 in the respective .dtsi files to reflect the correct IRQ assignments:
In STM32g071.dtsi, update USART4 to IRQ 30.
In STM32g0b1.dtsi, update USART5 to IRQ 31, and USART6 to IRQ 32.

Example of the Corrected DTS Configuration:
usart4: serial@40004c00 {
	compatible = "st,stm32-usart", "st,stm32-uart";
	reg = <0x40004c00 0x400>;
	clocks = <&rcc STM32_CLOCK(APB1, 19U)>;
	resets = <&rctl STM32_RESET(APB1L, 19U)>;
	interrupts = <30 0>; /* Correct IRQ for USART4 */
	status = "disabled";
};

usart5: serial@40005000 {
	compatible = "st,stm32-usart", "st,stm32-uart";
	reg = <0x40005000 0x400>;
	clocks = <&rcc STM32_CLOCK(APB1, 8U)>;
	resets = <&rctl STM32_RESET(APB1L, 8U)>;
	interrupts = <31 0>; /* Correct IRQ for USART5 */
	status = "disabled";
};

usart6: serial@40013c00 {
	compatible = "st,stm32-usart", "st,stm32-uart";
	reg = <0x40013c00 0x400>;
	clocks = <&rcc STM32_CLOCK(APB1, 9U)>;
	resets = <&rctl STM32_RESET(APB1L, 9U)>;
	interrupts = <32 0>; /* Correct IRQ for USART6 */
	status = "disabled";
};

Conclusion: This update will prevent interrupt conflicts when using UART4, UART5, and UART6 together, ensuring that each peripheral has its own unique interrupt handler.

Steps to Reproduce:
Use STM32G0B1 and STM32G071 with UART4, UART5, and UART6 enabled.
Observe interrupt conflicts and failures due to the same interrupt number (29) being used for all three UARTs.

FRASTM, 2024-11-21:
hi @adelfak 
on my side, I do not see any error in the interrupt position of USART 3/4/5/6  in the device tree of the stm32G0x0 or stm32G0x1 

referring to the RefMan [RM0454 STM32G0x0 advanced Arm®-based 32-bit MCUs](https://www.st.com/resource/en/reference_manual/rm0454-stm32g0x0-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) or to the [RM0444
STM32G0x1 advanced Arm®-based 32-bit MCUs](https://www.st.com/resource/en/reference_manual/rm0444-stm32g0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)  the IRQ 29 is common ( shared) for usart 3/4/5/6

adelfak, 2024-11-23:
thank for your responce
when adding these lines to nucleo_g0b1re.dts :

&usart3 {
	pinctrl-0 = <&usart3_tx_pb10 &usart3_rx_pb11>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};

&usart4 {
	pinctrl-0 = <&usart4_tx_pa0 &usart4_rx_pa1>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};

&usart5 {
	pinctrl-0 = <&usart5_tx_pd3 &usart5_rx_pd2>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};

&usart6 {
	pinctrl-0 = <&usart6_tx_pb8 &usart6_rx_pb9>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};

to enable all 6 UARTs to my baord.

I have this message error:
(( .../zephyr/zephyr_pre0.elf --intlist-section .intList --intlist-section intList --sw-isr-table --vector-table"
gen_isr_tables.py: error: multiple registrations at table_index 29 for irq 29 (0x1d)
Existing handler 0x8003d9f, new handler 0x8003d9f Has IRQ_CONNECT or IRQ_DIRECT_CONNECT accidentally been invoked on the same irq multiple times?)).

what is the reason?

djiatsaf-st, 2024-11-25:
hello @adelfak, 

These 4 usart instances have the same irq.

This PR https://github.com/zephyrproject-rtos/zephyr/pull/81877 will help you to use all these nodes without change their irq position.

With all usart nodes enabled in nucleo_g0b1re.dts, you need to set the config `CONFIG_SHARED_IRQ_MAX_NUM_CLIENTS` to `4 ` (number of devices with same irq) or more in the kconfig file of your application  `prj.conf` because the default value is only 2. 

So you will be able to build your project without this error.